sustainable-computing-io/kepler #606: Kubelet update is blocked by network IO?

SamYuan1990, 2023-03-28:
When I try to add benchmark for `updateKubeletMetrics` I found it seems a blocked network IO?

Here is the tracing stack:

https://github.com/sustainable-computing-io/kepler/blob/3f117b3c8b198806a7f26cc0e46bb18830bb0789/pkg/collector/container_cgroup_collector.go#L45

```
func GetContainerMetrics() (containerCPU, containerMem map[string]float64, nodeCPU, nodeMem float64, retErr error) {
	return podLister.ListMetrics()
}
```

```
// ListMetrics accesses Kubelet's metrics and obtain pods and node metrics
func (k *KubeletPodLister) ListMetrics() (containerCPU, containerMem map[string]float64, nodeCPU, nodeMem float64, retErr error) {
	resp, err := httpGet(metricsURL)
...
```

hence each time for `updateKubeletMetrics`, before it goes to loop containers/pods, there is a IO at networking to get and convert kubelet. 
Usually we thought IO is blocking, as the major time cost, and meaning less for benchmarking based on container.
Suggestion as decouple this network IO from code stack.

and btw, why we rebuild constants below for each container?
```
	cpuMetricName := collector_metric.AvailableKubeletMetrics[0]
	memMetricName := collector_metric.AvailableKubeletMetrics[1]
```

marceloamaral, 2023-03-28:
Ok, this is a good opportunity to discuss the need for both kubelet and cgroup metrics. 
In my opinion, it seems that the kubelet metrics may duplicate the cgroup metrics, so we only need the cgroup metrics.

@sunya-ch @rootfs ↑↑

SamYuan1990, 2023-03-28:
> Ok, this is a good opportunity to discuss the need for both kubelet and cgroup metrics. In my opinion, it seems that the kubelet metrics may duplicate the cgroup metrics, so we only need the cgroup metrics.
> 
> @sunya-ch @rootfs ↑↑

I suppose we need a scope of both kubelet features and cgroup features. Can anyone help?
CC: @jichenjc

SamYuan1990, 2023-03-28:
link with https://github.com/sustainable-computing-io/kepler/issues/605

rootfs, 2023-03-28:
The following is my mental map. For cgroup v1, kubelet metrics is the way to go.

Models | Features | Use Case
-- | -- | --
CounterOnly | HW Perf Counters | Baremetal servers
CGroupOnly | Cgroup stats (CPU, memory, IO) | OpenShift on VMs
KubeletOnly | Kubelet stats from cAdvisor | None CGroup V2 Kubernetes on VM

jichenjc, 2023-03-29:
>>>CGroup V1 Kubernetes on VM

cgroup v2 is the future and given kubelet might not give so accurate data, should we still need kubelet ?
if we do, I think the current code is fine ?

marceloamaral, 2023-03-29:
Thanks @rootfs, it makes sense.

So it will also be related to the discussion in #605 an #558 of using an external lib to collect cgroup metrics.
The external lib should support both v1 and v2, then we will not need the kubelet metrics anymore

SamYuan1990, 2023-03-29:
@marceloamaral , @rootfs , @jichenjc 
please correct me for any misunderstanding:
- we consume cpu and memory usage from kubelet, which can be replaced by cgroup implementation.
- we should consider the implementation works for either cgroup v1 or cgroup v2.
- for cgroup, we should read cpu, memory and IO from cgroup.

SamYuan1990, 2023-03-29:
> cgroups is a Linux kernel-specific feature, so you won't find it on your Mac.
https://unix.stackexchange.com/questions/558799/how-to-install-cgroups-on-mac-os

em.... how should I test it on mac?

SamYuan1990, 2023-03-29:
```go
func IgnoreNotExist(err error) error {
	if os.IsNotExist(err) {
		return nil
	}
	return err
}
// Cgroupv1
		//test
		shares := uint64(100)
		control, err := cgroup1.New(cgroup1.StaticPath("/test"), &specs.LinuxResources{
			CPU: &specs.LinuxCPU{
				Shares: &shares,
			},
		})
		Expect(err).NotTo(HaveOccurred())
		defer control.Delete()
		// create
		control2, err := cgroup1.Load(cgroup1.StaticPath("/test"))
		defer control2.Delete()
		// metric is cgroup interface
		metric, err := control2.Stat(IgnoreNotExist)
		Expect(err).NotTo(HaveOccurred())
		// CPU
		fmt.Println(metric.CPU.GetUsage().String())
		// Memory
		fmt.Println(metric.Memory.GetUsage().String())
		// Blkio IO?
		fmt.Println(metric.Blkio.String())

		// cgroup v2
		res := cgroup2.Resources{}
		m, err := cgroup2.NewSystemd("/", "my-cgroup-abc.slice", -1, &res)
		Expect(err).NotTo(HaveOccurred())
		defer m.DeleteSystemd()
		// create
		m2, err := cgroup2.LoadSystemd("/", "my-cgroup-abc.slice")
		Expect(err).NotTo(HaveOccurred())
		defer m2.DeleteSystemd()
		// metricv2 is cgroup interface
		metricv2, err := m2.Stat()
		Expect(err).NotTo(HaveOccurred())
		// CPU
		fmt.Println(metricv2.CPU.String())
		// Memory
		fmt.Println(metricv2.Memory.String())
		// IO
		fmt.Println(metricv2.Io.String())
```
@marceloamaral , @jichenjc , @rootfs 
I don't have linux OS as I use my personal mac as desktop for development works.
Could you please help verify if code blocks above meet our goals?
or any guidance to use `pkg/cgroup/test/hierachywithqos/cpu/kubepods.slice/kubepods-best-effort.slice`?

SamYuan1990, 2023-03-29:
ref https://github.com/containerd/cgroups/issues/55

sunya-ch, 2023-03-29:
I think we do not need the kubelet metric anymore because now we already have the IRQ metrics for base case (including v1 I suppose). Maybe it's time to retire the `(k *KubeletPodLister) ListMetrics()`.

SamYuan1990, 2023-03-29:
> I think we do not need the kubelet metric anymore because now we already have the IRQ metrics for base case (including v1 I suppose). Maybe it's time to retire the `(k *KubeletPodLister) ListMetrics()`.

thx for your quick reply @sunya-ch , well, let's further check with history...
1. we do agree that we can replace kubelet https://github.com/sustainable-computing-io/kepler/issues/606#issuecomment-1487886211 
2. as kubelet is from cgroup as well, and #605 , cgroup impl has some bad code smell hence we are try to use any other package or impl as an alternative.
3. https://github.com/sustainable-computing-io/kepler/issues/606#issuecomment-1487142569 could you please check with @rootfs 's comments and let's further make a final decision which feature implemented by which approach?
- it seems we have cgroup, IRQ, kubelet ... lots of impl for same feature(get CPU data?)

marceloamaral, 2023-03-29:
> - we consume cpu and memory usage from kubelet, which can be replaced by cgroup implementation.
> - we should consider the implementation works for either cgroup v1 or cgroup v2.
> - for cgroup, we should read cpu, memory and IO from cgroup.

Correct. But we might also collect network metrics in the future.

> I don't have linux OS as I use my personal mac as desktop for development works.

@SamYuan1990 If you don't mind I can work on this...

> Could you please help verify if code blocks above meet our goals?

I tested a modified version of the code that works well for both cgroup v1 and v2.

cgroup V1:
```
CPU
total:26062510758 kernel:9420000000 user:16710000000 per_cpu:630714713 per_cpu:381714362 per_cpu:352684506 
Memory
limit:9223372036854771712 usage:88735744 max:92286976
IO
io_service_bytes_recursive:{op:"Read" device:"/dev/sda" major:8}
```

cgroup V2:
```
CPU
usage_usec:4354475015  user_usec:4134824093  system_usec:219650921
Memory
anon:675196928  file:560631808  kernel_stack:638976  slab:17439832  sock:4096  file_mapped:122204160  anon_thp:44040192  inactive_anon:929349632  active_anon:4096  inactive_file:560017408  active_file:614400  slab_reclaimable:16991136  slab_unreclaimable:448696  pgfault:14868894  pgmajfault:1  pgactivate:3419  thp_fault_alloc:1793  thp_collapse_alloc:179  usage:1511608320  usage_limit:18446744073709551615  swap_limit:18446744073709551615
IO
usage:{major:8  rbytes:40960  wbytes:1857515520  rios:10  wios:39270}  usage:{major:253  minor:1  rbytes:40960  wbytes:1857515520  rios:10  wios:30646}
```

SamYuan1990, 2023-03-29:
> Correct. But we might also collect network metrics in the future.
>
>
> @SamYuan1990 If you don't mind I can work on this...
>
>
> I tested a modified version of the code that works well for both cgroup v1 [...]

@marceloamaral , please go ahead/take over here, due to OS limitation, I can't move forward for now.

marceloamaral, 2023-03-31:
@SamYuan1990 I created the PR #609 implementing the lib

closed 2023-08-28 by stale[bot]