sustainable-computing-io/kepler #610: Timely occur invalid memory address or nil pointer in updateAcceleratorMetrics

sunya-ch, 2023-04-04:
**Describe the bug**
A clear and concise description of what the bug is.

There is timely occurring bug during collecting GPU metric. It seems some of the following component is not properly initialized by `createContainersMetricsIfNotExist`.

https://github.com/sustainable-computing-io/kepler/blob/4bdddb1e2b5829f142f8fb2beb7db54c8aee0538/pkg/collector/container_accelerator_collector.go#L57

https://github.com/sustainable-computing-io/kepler/blob/4bdddb1e2b5829f142f8fb2beb7db54c8aee0538/pkg/collector/container_accelerator_collector.go#L51

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Run Kepler on node with GPU enable for a while. We still cannot identify the trigger point.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

```bash
found 8 gpu devices
Using nvml to obtain gpu power
I0330 01:56:03.586089       1 exporter.go:150] Kepler running on version: 2b59dbd
I0330 01:56:03.586260       1 config.go:169] using gCgroup ID in the BPF program: true
I0330 01:56:03.586322       1 config.go:170] kernel version: 4.18
I0330 01:56:03.646782       1 rapl_msr_util.go:151] input/output error
I0330 01:56:03.646852       1 power.go:64] Not able to obtain power, use estimate method
I0330 01:56:03.695517       1 exporter.go:170] Initializing the GPU collector
perf_event_open: No such file or directory
I0330 01:56:04.678864       1 bcc_attacher.go:98] failed to attach perf event cpu_cycles_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0330 01:56:04.678916       1 bcc_attacher.go:98] failed to attach perf event cpu_ref_cycles_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0330 01:56:04.678956       1 bcc_attacher.go:98] failed to attach perf event cpu_instr_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0330 01:56:04.678998       1 bcc_attacher.go:98] failed to attach perf event cache_miss_hc_reader: failed to open bpf perf event: no such file or directory
I0330 01:56:04.679009       1 bcc_attacher.go:135] Successfully load eBPF module with option: [-DNUM_CPUS=80 -DSET_GROUP_ID]
I0330 01:56:04.738285       1 exporter.go:212] Started Kepler in 1.152227472s
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0xbb7648]

goroutine 48 [running]:
github.com/sustainable-computing-io/kepler/pkg/collector.(*Collector).updateAcceleratorMetrics(0xc0002aeb40)
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/collector/container_accelerator_collector.go:57 +0x448
github.com/sustainable-computing-io/kepler/pkg/collector.(*Collector).Update(0xc0002aeb40)
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/collector/metric_collector.go:120 +0xb6
github.com/sustainable-computing-io/kepler/pkg/manager.(*CollectorManager).Start.func1()
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/manager/manager.go:64 +0x7b
created by github.com/sustainable-computing-io/kepler/pkg/manager.(*CollectorManager).Start
	/opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/manager/manager.go:56 +0x6
```

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marceloamaral, 2023-04-04:
Interesting, I never seen this problem before.

Does this machine has GPUs?

Can you check which part has the nil pointer?
-  c.ContainersMetrics[containerID]
- CounterStats[config.GPUSMUtilization]
- processUtilization.SmUtil

rootfs, 2023-04-04:
it happened probably due to potential GPU failure

rootfs, 2023-04-04:
@marceloamaral @sunya-ch

rootfs, 2023-04-07:
closed by #618

closed 2023-04-07 by rootfs