sustainable-computing-io/kepler #624: bpf: use batch bpf map reset

rootfs, 2023-04-12:
**Is your feature request related to a problem? Please describe.**
When running on a rt kernel, deleting map elements individually causes high cpu usage:
```console
# go tool pprof _output/bin/linux_amd64/kepler  /tmp/k.c.2
File: kepler
Build ID: 30b660e068fddc061eccde71b5bfb749fe1cb749
Type: cpu
Time: Apr 12, 2023 at 3:29pm (EDT)
Duration: 60.13s, Total samples = 59.28s (98.58%)
Entering interactive mode (type "help" for commands, "o" for options)
(pprof) top20 -cum (all)
Active filters:
   focus=(all)
Showing nodes accounting for 58.23s, 98.23% of 59.28s total
Dropped 55 nodes (cum <= 0.30s)
      flat  flat%   sum%        cum   cum%
    58.02s 97.87% 97.87%     58.53s 98.73%  runtime.cgocall
         0     0% 97.87%     58.20s 98.18%  main.main
         0     0% 97.87%     58.20s 98.18%  runtime.main
         0     0% 97.87%     58.18s 98.14%  github.com/iovisor/gobpf/bcc.(*Table).DeleteAll
         0     0% 97.87%     58.18s 98.14%  github.com/sustainable-computing-io/kepler/pkg/collector.(*Collector).Initialize
         0     0% 97.87%     58.18s 98.14%  github.com/sustainable-computing-io/kepler/pkg/collector.(*Collector).resetBPFTables (inline)
         0     0% 97.87%     58.18s 98.14%  github.com/sustainable-computing-io/kepler/pkg/manager.(*CollectorManager).Start
         0     0% 97.87%     49.43s 83.38%  github.com/iovisor/gobpf/bcc.(*Table).DeleteAll.func4
         0     0% 97.87%     49.14s 82.89%  github.com/iovisor/gobpf/bcc._Cfunc_bpf_get_next_key
         0     0% 97.87%      8.75s 14.76%  github.com/iovisor/gobpf/bcc.(*Table).DeleteAll.func5
         0     0% 97.87%      8.49s 14.32%  github.com/iovisor/gobpf/bcc._C2func_bpf_delete_elem
         0     0% 97.87%      0.90s  1.52%  github.com/iovisor/gobpf/bcc._Cfunc_bpf_module_create_c_from_string
         0     0% 97.87%      0.90s  1.52%  github.com/iovisor/gobpf/bcc.compile
         0     0% 97.87%      0.90s  1.52%  github.com/iovisor/gobpf/bcc.newModule
         0     0% 97.87%      0.90s  1.52%  github.com/iovisor/gobpf/bcc.newModule.func3
     0.03s 0.051% 97.93%      0.55s  0.93%  runtime.cgoCheckPointer
     0.05s 0.084% 98.01%      0.52s  0.88%  runtime.cgoCheckArg
     0.04s 0.067% 98.08%      0.44s  0.74%  runtime.exitsyscall
     0.09s  0.15% 98.23%      0.33s  0.56%  runtime.cgoCheckUnknownPointer
```

Close reading iovisor `DeleteAll`, the element is deleted sequentially:
```go
// DeleteAll deletes all entries from the table
func (table *Table) DeleteAll() error {
        mod := table.module.p
        fd := C.bpf_table_fd_id(mod, table.id)

        keySize := C.bpf_table_key_size_id(mod, table.id)
        key := make([]byte, keySize)
        keyP := unsafe.Pointer(&key[0])
        for res := C.bpf_get_first_key(fd, keyP, keySize); res == 0; res = C.bpf_get_next_key(fd, keyP, keyP) {
                r, err := C.bpf_delete_elem(fd, keyP)
                if r != 0 {
                        return fmt.Errorf("Table.DeleteAll: unable to delete element: %v", err)
                }
        }
        return nil
}
```

**Describe the solution you'd like**
We should try the ebpf [batch API bpf_map_delete_batch](https://lore.kernel.org/bpf/20211225203717.35718-1-grantseltzer@gmail.com/). Note this API is not in the golang binding, we have to create go binding first.

Thank @novacain1 for sharing the info

rootfs, 2023-04-12:
@sustainable-computing-io/maintainer

jichenjc, 2023-04-13:
>>>Note this API is not in the golang binding, we have to create go binding first.

still learning this ... what's the binding means in bcc ? didn't find through google (only give me gin binding..)

marceloamaral, 2023-04-13:
Actually there is a better way to that. I forgot to submit a PR for this.

We must iterate all elements in the map to read it anyway. Therefore, we can delete the element after reading it.
I will submit a PR soon.

```golang
for it := c.bpfHCMeter.Table.Iter(); it.Next(); {
    key := it.Key()
    data := it.Leaf()
    var ct ProcessBPFMetrics
    if err := binary.Read(bytes.NewBuffer(data), binary.LittleEndian, &ct); err != nil {
	    klog.Infof("failed to decode received data: %v", err)
	    continue // this only happens if there is a problem in the bpf code
    }
    if err := c.bpfHCMeter.Table.Delete(key); err != nil {
	    klog.Infof("could not delete bpf table elemets, err: %v", err)
    }
}
```

marceloamaral, 2023-04-13:
>what's the binding means in bcc ?

@jichenjc the BCC Golang library we use does not contains all BCC functions. This means we need to create a function that invokes the C code for a specific BCC function that is missing. As far as I understand, binding refers to the act of exposing/implementing the BCC function in the golang side.

rootfs, 2023-04-13:
@marceloamaral deleting one element one by one doesn't scale in real time kernel. As explained [here](https://lwn.net/Articles/812503/), the locking and context switching through sys_bpf() are quite restrictive in rt kernels. Batching deleting doesn't have these issues.

marceloamaral, 2023-06-13:
Closing since the PR #626 fix this.

closed 2023-06-13 by marceloamaral