sustainable-computing-io/kepler #627: Logs indicate issues with Kepler helm chart installation on a Kind cluster

sprathod369, 2023-04-13:
**Describe the bug**
Just started exploring Kepler that seems to be an exciting initiative in the Green Software area. 
I'm running a Kind cluster (our node - three workers, one controller) on an on-prem Ubuntu 20x VM. Kubernetes v1.26.3
Have installed Kepler using the latest available Helm Chart. The Kepler exporter pods are running but the logs indicate errors as part of the setup.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Kepler using the latest available Helm Chart and using the default values.yaml
`# helm install kepler . --values values.yaml --create-namespace -n kepler`
```
NAME: kepler
LAST DEPLOYED: Mon Apr 10 16:54:47 2023
NAMESPACE: kepler
STATUS: deployed
REVISION: 1
TEST SUITE: None
NOTES:
Get the application URL by running these commands:
  export POD_NAME=$(kubectl get pods --namespace kepler -l "app.kubernetes.io/name=kepler-exporter,app.kubernetes.io/instance=kepler" -o jsonpath="{.items[0].metadata.name}")
  export CONTAINER_PORT=$(kubectl get pod --namespace kepler $POD_NAME -o jsonpath="{.spec.containers[0].ports[0].containerPort}")
  echo "Visit http://127.0.0.1:8080 to use your application"
  kubectl --namespace kepler port-forward $POD_NAME 8080:$CONTAINER_PORT

```
2. Validate the Helm deployment and pod status
`# helm ls -n kepler`

[image]

`# kubectl get all -n kepler`
The kepler-exporter service and the 3 pods running
[image]

3. Review the logs of the running kepler exporter pods that has errors
`# kubectl logs pod/kepler-exporter-g5ftw -n kepler`

```
~/Kepler# kubectl logs pod/kepler-exporter-g5ftw -n kepler
I0411 08:35:26.401399       1 gpu.go:43] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0411 08:35:26.403749       1 exporter.go:149] Kepler running on version: 5b9c91a
I0411 08:35:26.403794       1 config.go:169] using gCgroup ID in the BPF program: true
I0411 08:35:26.403885       1 config.go:170] kernel version: 5.15
I0411 08:35:26.403963       1 rapl_msr_util.go:122] failed to open path /dev/cpu/0/msr: no such file or directory
I0411 08:35:26.405297       1 power.go:64] Not able to obtain power, use estimate method
I0411 08:35:26.550052       1 exporter.go:167] Initializing the GPU collector
xz: (stdin): Read error: No such device
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
chdir(/lib/modules/5.15.0-69-generic/build): No such file or directory
I0411 08:35:38.046491       1 bcc_attacher.go:64] failed to attach the bpf program: <nil>
I0411 08:35:38.046548       1 bcc_attacher.go:124] failed to attach perf module with options [-DNUM_CPUS=4]: failed to attach the bpf program: <nil>, not able to load eBPF modules
I0411 08:35:38.046608       1 exporter.go:184] failed to start : failed to attach bpf assets: failed to attach the bpf program: <nil>
I0411 08:35:38.046743       1 exporter.go:209] Started Kepler in 11.643043983s
```

`# kubectl logs pod/kepler-exporter-g7d6r -n kepler`

```
~/Kepler# kubectl logs pod/kepler-exporter-g7d6r -n kepler
I0411 08:35:23.814418       1 gpu.go:43] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0411 08:35:23.817037       1 exporter.go:149] Kepler running on version: 5b9c91a
I0411 08:35:23.817219       1 config.go:169] using gCgroup ID in the BPF program: true
I0411 08:35:23.817463       1 config.go:170] kernel version: 5.15
I0411 08:35:23.817689       1 rapl_msr_util.go:122] failed to open path /dev/cpu/0/msr: no such file or directory
I0411 08:35:23.817995       1 power.go:64] Not able to obtain power, use estimate method
I0411 08:35:23.881693       1 exporter.go:167] Initializing the GPU collector
perf_event_open: No such file or directory
I0411 08:35:39.480026       1 bcc_attacher.go:98] failed to attach perf event cpu_cycles_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: Invalid argument
I0411 08:35:39.480139       1 bcc_attacher.go:98] failed to attach perf event cpu_ref_cycles_hc_reader: failed to open bpf perf event: invalid argument
perf_event_open: No such file or directory
I0411 08:35:39.480456       1 bcc_attacher.go:98] failed to attach perf event cpu_instr_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0411 08:35:39.480608       1 bcc_attacher.go:98] failed to attach perf event cache_miss_hc_reader: failed to open bpf perf event: no such file or directory
I0411 08:35:39.480633       1 bcc_attacher.go:135] Successfully load eBPF module with option: [-DNUM_CPUS=4]
I0411 08:35:39.498904       1 exporter.go:209] Started Kepler in 15.681913307s
```

`# kubectl logs pod/kepler-exporter-l9fsn -n kepler`

```
~/Kepler# kubectl logs pod/kepler-exporter-l9fsn -n kepler
I0411 11:15:33.138778       1 gpu.go:43] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0411 11:15:33.143044       1 exporter.go:149] Kepler running on version: 5b9c91a
I0411 11:15:33.143159       1 config.go:169] using gCgroup ID in the BPF program: true
I0411 11:15:33.143388       1 config.go:170] kernel version: 5.15
I0411 11:15:33.143607       1 rapl_msr_util.go:122] failed to open path /dev/cpu/0/msr: no such file or directory
I0411 11:15:33.143983       1 power.go:64] Not able to obtain power, use estimate method
I0411 11:15:33.624353       1 exporter.go:167] Initializing the GPU collector
perf_event_open: No such file or directory
I0411 11:15:42.034885       1 bcc_attacher.go:98] failed to attach perf event cpu_instr_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0411 11:15:42.035050       1 bcc_attacher.go:98] failed to attach perf event cache_miss_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0411 11:15:42.035185       1 bcc_attacher.go:98] failed to attach perf event cpu_cycles_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: Invalid argument
I0411 11:15:42.035343       1 bcc_attacher.go:98] failed to attach perf event cpu_ref_cycles_hc_reader: failed to open bpf perf event: invalid argument
I0411 11:15:42.035376       1 bcc_attacher.go:135] Successfully load eBPF module with option: [-DNUM_CPUS=4]
I0411 11:15:42.099191       1 exporter.go:209] Started Kepler in 8.956189783s

```

**Expected behavior**
The running kepler-exporter pods should be capturing the metrics without any errors in the logs as showcased in the Kepler deployment and demo link - https://www.youtube.com/watch?v=P5weULiBl60.

jichenjc, 2023-04-13:
```
perf_event_open: No such file or directory
I0411 11:15:42.034885       1 bcc_attacher.go:98] failed to attach perf event cpu_instr_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0411 11:15:42.035050       1 bcc_attacher.go:98] failed to attach perf event cache_miss_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0411 11:15:42.035185       1 bcc_attacher.go:98] failed to attach perf event cpu_cycles_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: Invalid argument
I0411 11:15:42.035343       1 bcc_attacher.go:98] failed to attach perf event cpu_ref_cycles_hc_reader: failed to open bpf perf event: invalid argument
```

if you are talking about those logs , it's fine because in kind cluster you are not able to obtain because it's baremetal only enabled
did you see metrics in `curl` command?

rootfs, 2023-04-13:
`failed to attach perf event` is expected on VMs. If the HW are not presented to the VMs, then ebpf program will report this message. This helps us to identify the runtime environment. If better wording helps, @sprathod369 please submit a PR

sprathod369, 2023-04-14:
Thanks @jichenjc for the inputs - the below are the metrics I see in my browser for the Energy Stats Exporter (kepler-exporter service) running on port 9102. 
I'm wondering if a Kind based cluster good enough to understand, explore and supported by Kepler? Though I've multiple pods running as part of the Temporal workload in this cluster, the metrics do not capture any pod_* metrics as I see in the video running curl command. Looking at the below output, I'm certain that if I set up Prometheus and Grafana, all the metrics counter may end up being 0 or null. It seems something is amiss in my setup to get the required energy metrics on my local cluster.

```
# HELP go_gc_duration_seconds A summary of the pause duration of garbage collection cycles.
# TYPE go_gc_duration_seconds summary
go_gc_duration_seconds{quantile="0"} 5.5501e-05
go_gc_duration_seconds{quantile="0.25"} 0.000120208
go_gc_duration_seconds{quantile="0.5"} 0.000142096
go_gc_duration_seconds{quantile="0.75"} 0.000197469
go_gc_duration_seconds{quantile="1"} 0.000445609
go_gc_duration_seconds_sum 0.002138696
go_gc_duration_seconds_count 13
# HELP go_goroutines Number of goroutines that currently exist.
# TYPE go_goroutines gauge
go_goroutines 10
# HELP go_info Information about the Go environment.
# TYPE go_info gauge
go_info{version="go1.18.1"} 1
# HELP go_memstats_alloc_bytes Number of bytes allocated and still in use.
# TYPE go_memstats_alloc_bytes gauge
go_memstats_alloc_bytes 1.0898e+06
# HELP go_memstats_alloc_bytes_total Total number of bytes allocated, even if freed.
# TYPE go_memstats_alloc_bytes_total counter
go_memstats_alloc_bytes_total 9.367976e+06
# HELP go_memstats_buck_hash_sys_bytes Number of bytes used by the profiling bucket hash table.
# TYPE go_memstats_buck_hash_sys_bytes gauge
go_memstats_buck_hash_sys_bytes 1.449563e+06
# HELP go_memstats_frees_total Total number of frees.
# TYPE go_memstats_frees_total counter
go_memstats_frees_total 22771
# HELP go_memstats_gc_sys_bytes Number of bytes used for garbage collection system metadata.
# TYPE go_memstats_gc_sys_bytes gauge
go_memstats_gc_sys_bytes 4.78384e+06
# HELP go_memstats_heap_alloc_bytes Number of heap bytes allocated and still in use.
# TYPE go_memstats_heap_alloc_bytes gauge
go_memstats_heap_alloc_bytes 1.0898e+06
# HELP go_memstats_heap_idle_bytes Number of heap bytes waiting to be used.
# TYPE go_memstats_heap_idle_bytes gauge
go_memstats_heap_idle_bytes 9.863168e+06
# HELP go_memstats_heap_inuse_bytes Number of heap bytes that are in use.
# TYPE go_memstats_heap_inuse_bytes gauge
go_memstats_heap_inuse_bytes 2.260992e+06
# HELP go_memstats_heap_objects Number of allocated objects.
# TYPE go_memstats_heap_objects gauge
go_memstats_heap_objects 11223
# HELP go_memstats_heap_released_bytes Number of heap bytes released to OS.
# TYPE go_memstats_heap_released_bytes gauge
go_memstats_heap_released_bytes 9.715712e+06
# HELP go_memstats_heap_sys_bytes Number of heap bytes obtained from system.
# TYPE go_memstats_heap_sys_bytes gauge
go_memstats_heap_sys_bytes 1.212416e+07
# HELP go_memstats_last_gc_time_seconds Number of seconds since 1970 of last garbage collection.
# TYPE go_memstats_last_gc_time_seconds gauge
go_memstats_last_gc_time_seconds 1.6814450112840195e+09
# HELP go_memstats_lookups_total Total number of pointer lookups.
# TYPE go_memstats_lookups_total counter
go_memstats_lookups_total 0
# HELP go_memstats_mallocs_total Total number of mallocs.
# TYPE go_memstats_mallocs_total counter
go_memstats_mallocs_total 33994
# HELP go_memstats_mcache_inuse_bytes Number of bytes in use by mcache structures.
# TYPE go_memstats_mcache_inuse_bytes gauge
go_memstats_mcache_inuse_bytes 4800
# HELP go_memstats_mcache_sys_bytes Number of bytes used for mcache structures obtained from system.
# TYPE go_memstats_mcache_sys_bytes gauge
go_memstats_mcache_sys_bytes 15600
# HELP go_memstats_mspan_inuse_bytes Number of bytes in use by mspan structures.
# TYPE go_memstats_mspan_inuse_bytes gauge
go_memstats_mspan_inuse_bytes 78336
[... 43 lines cut ...]
process_max_fds 1.048576e+06
# HELP process_open_fds Number of open file descriptors.
# TYPE process_open_fds gauge
process_open_fds 34
# HELP process_resident_memory_bytes Resident memory size in bytes.
# TYPE process_resident_memory_bytes gauge
process_resident_memory_bytes 1.2771328e+07
# HELP process_start_time_seconds Start time of the process since unix epoch in seconds.
# TYPE process_start_time_seconds gauge
process_start_time_seconds 1.68137636424e+09
# HELP process_virtual_memory_bytes Virtual memory size in bytes.
# TYPE process_virtual_memory_bytes gauge
process_virtual_memory_bytes 2.0066304e+07
# HELP process_virtual_memory_max_bytes Maximum amount of virtual memory available in bytes.
# TYPE process_virtual_memory_max_bytes gauge
process_virtual_memory_max_bytes 1.8446744073709552e+19
# HELP promhttp_metric_handler_requests_in_flight Current number of scrapes being served.
# TYPE promhttp_metric_handler_requests_in_flight gauge
promhttp_metric_handler_requests_in_flight 1
# HELP promhttp_metric_handler_requests_total Total number of scrapes by HTTP status code.
# TYPE promhttp_metric_handler_requests_total counter
promhttp_metric_handler_requests_total{code="200"} 11
promhttp_metric_handler_requests_total{code="500"} 0
promhttp_metric_handler_requests_total{code="503"} 0
```

sprathod369, 2023-04-14:
@rootfs - Thanks!! Do we have a better detailed or alternative documentation to Kepler installation than [this](https://sustainable-computing.io/installation/kepler/) ? I'm still struggling and trying to figure out a successful way to build the manifest and deploy from source code as an alternative to Helm. 
Is Kind based Kubernetes cluster good enough to understand, explore and supported for Kepler use cases?

sprathod369, 2023-04-15:
@rootfs - Yes, that's what I suspected after going through the Troubleshooting pointers mentioned in the doc. The kernel headers are probably missing on the nodes (docker containers) in my Kind cluster. I'll fix that and test. Thanks!!

sprathod369, 2023-04-19:
@rootfs - Created a new single node Kind cluster based on configuration you suggested [here](https://github.com/sustainable-computing-io/local-dev-cluster/blob/main/kind/manifests/kind.yml) to make sure extraMounts: cover linux header and BCC. Logged in to the kind docker node to confirm that my kind cluster node has the kernel headers
```
root:~# docker exec -it f4e1307faf50 bash
root@kind-control-plane:/# ls -l /usr/src/linux-headers-$(uname -r)
total 1804
lrwxrwxrwx 1 root root      49 Mar 20 15:32 Documentation -> ../linux-hwe-5.15-headers-5.15.0-69/Documentation
lrwxrwxrwx 1 root root      42 Mar 20 15:32 Kbuild -> ../linux-hwe-5.15-headers-5.15.0-69/Kbuild
lrwxrwxrwx 1 root root      43 Mar 20 15:32 Kconfig -> ../linux-hwe-5.15-headers-5.15.0-69/Kconfig
lrwxrwxrwx 1 root root      44 Mar 20 15:32 Makefile -> ../linux-hwe-5.15-headers-5.15.0-69/Makefile
-rw-r--r-- 1 root root 1814929 Mar 20 15:32 Module.symvers
drwxr-xr-x 3 root root    4096 Mar 31 00:59 arch
lrwxrwxrwx 1 root root      41 Mar 20 15:32 block -> ../linux-hwe-5.15-headers-5.15.0-69/block
lrwxrwxrwx 1 root root      41 Mar 20 15:32 certs -> ../linux-hwe-5.15-headers-5.15.0-69/certs
lrwxrwxrwx 1 root root      42 Mar 20 15:32 crypto -> ../linux-hwe-5.15-headers-5.15.0-69/crypto
lrwxrwxrwx 1 root root      43 Mar 20 15:32 drivers -> ../linux-hwe-5.15-headers-5.15.0-69/drivers
lrwxrwxrwx 1 root root      38 Mar 20 15:32 fs -> ../linux-hwe-5.15-headers-5.15.0-69/fs
drwxr-xr-x 4 root root    4096 Mar 31 00:59 include
lrwxrwxrwx 1 root root      40 Mar 20 15:32 init -> ../linux-hwe-5.15-headers-5.15.0-69/init
lrwxrwxrwx 1 root root      44 Mar 20 15:32 io_uring -> ../linux-hwe-5.15-headers-5.15.0-69/io_uring
lrwxrwxrwx 1 root root      39 Mar 20 15:32 ipc -> ../linux-hwe-5.15-headers-5.15.0-69/ipc
drwxr-xr-x 2 root root    4096 Mar 31 00:59 kernel
lrwxrwxrwx 1 root root      39 Mar 20 15:32 lib -> ../linux-hwe-5.15-headers-5.15.0-69/lib
lrwxrwxrwx 1 root root      38 Mar 20 15:32 mm -> ../linux-hwe-5.15-headers-5.15.0-69/mm
lrwxrwxrwx 1 root root      39 Mar 20 15:32 net -> ../linux-hwe-5.15-headers-5.15.0-69/net
lrwxrwxrwx 1 root root      43 Mar 20 15:32 samples -> ../linux-hwe-5.15-headers-5.15.0-69/samples
drwxr-xr-x 7 root root   12288 Mar 31 00:59 scripts
lrwxrwxrwx 1 root root      44 Mar 20 15:32 security -> ../linux-hwe-5.15-headers-5.15.0-69/security
lrwxrwxrwx 1 root root      41 Mar 20 15:32 sound -> ../linux-hwe-5.15-headers-5.15.0-69/sound
drwxr-xr-x 4 root root    4096 Mar 31 00:59 tools
lrwxrwxrwx 1 root root      42 Mar 20 15:32 ubuntu -> ../linux-hwe-5.15-headers-5.15.0-69/ubuntu
lrwxrwxrwx 1 root root      39 Mar 20 15:32 usr -> ../linux-hwe-5.15-headers-5.15.0-69/usr
lrwxrwxrwx 1 root root      40 Mar 20 15:32 virt -> ../linux-hwe-5.15-headers-5.15.0-69/virt

```

Manually build the manifest using `make build-manifest OPTS="CI_DEPLOY"`, and then applied the generated deployment.yaml
`kubectl apply -f _output/generated-manifest/deployment.yaml `

Below are the logs for the kepler-exporter but still something is amiss based on comparing it with the demo video
```
~# kubectl logs pod/kepler-exporter-4g8kc -n kepler
I0419 03:45:32.846266       1 gpu.go:43] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0419 03:45:32.849142       1 exporter.go:149] Kepler running on version: d1bfdc4
I0419 03:45:32.849191       1 config.go:169] using gCgroup ID in the BPF program: true
I0419 03:45:32.849357       1 config.go:170] kernel version: 5.15
I0419 03:45:32.849525       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0419 03:45:32.849743       1 power.go:64] Not able to obtain power, use estimate method
I0419 03:45:33.624165       1 exporter.go:167] Initializing the GPU collector
perf_event_open: No such file or directory
I0419 03:45:35.573617       1 bcc_attacher.go:98] failed to attach perf event cpu_cycles_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: Invalid argument
I0419 03:45:35.573811       1 bcc_attacher.go:98] failed to attach perf event cpu_ref_cycles_hc_reader: failed to open bpf perf event: invalid argument
perf_event_open: No such file or directory
I0419 03:45:35.573976       1 bcc_attacher.go:98] failed to attach perf event cpu_instr_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0419 03:45:35.574108       1 bcc_attacher.go:98] failed to attach perf event cache_miss_hc_reader: failed to open bpf perf event: no such file or directory
I0419 03:45:35.574167       1 bcc_attacher.go:135] Successfully load eBPF module with option: [-DNUM_CPUS=4]
I0419 03:45:35.596341       1 exporter.go:209] Started Kepler in 2.747228921s
```
Even when I run `curl http://localhost:9102/metrics` I still do not see any pod_* stats / counters (like pod_gpu_energy, pod_cpu_energy, pod_dram_energy etc.) even though I have the sample bookinfo app workloads deployed and running on this single node kind cluster. 
What am I missing?

marceloamaral, 2023-04-19:
The prometheus metrics does not have the `pod` string in the name.
We follow the general prometheus metrics name recommendation and report the metrics per container.

Can you try to see if you have the metrics:
`kepler_container_joules_total`
`kepler_container_package_joules_total`

sprathod369, 2023-04-20:
Thanks @marceloamaral for the response. Yes, I can see the below metrics in the kepler exporter 
```
kepler_container_joules_total
kepler_container_package_joules_total
```

Actually I was referring to the [Kepler demo video](https://www.youtube.com/watch?v=P5weULiBl60) that showcases the **pod_* stats / metrics** (that are also seen in Prometheus and Grafana). Please see the screenshot below
[image]

For me, the `kepler_pod_energy_stat` metrics are showing zero in the kepler exporter even though I've workloads deployed and running (bookinfo and yelp app) on this single node KIND cluster.  After setting up the monitoring stack based on in the [kepler installation guide](https://sustainable-computing.io/installation/kepler/), when I visualize the power consumption metrics made available by the Kepler Exporter by importing the pre-generated Kepler Dashboard into Grafana, all metrics are zero so I was wondering if anything is missing in my setup on a KIND cluster.

[image]

SamYuan1990, 2023-04-22:
aha，
```
Kepler Installation
Requirements
Kernel 4.18+
Access to a Kubernetes cluster
kubectl v1.21.0+
```
https://sustainable-computing.io/installation/kepler/#requirements
we'd better enhance this part of document with bcc?

rootfs, 2023-04-23:
@sprathod369 cgroup v1 should work now. There are some changes in metrics names. kepler_pod_energy_stat is no longer supported. Can you check from your `curl http://localhost:9102/metrics` output and see if `kepler_container_joules_total` is there and what values does it have?

sprathod369, 2023-04-24:
Here's what I did to test the latest changes 
1) Deleted the old KIND cluster and created a new single node KIND cluster - based on extraMounts configuration [here](https://github.com/sustainable-computing-io/local-dev-cluster/blob/main/kind/manifests/kind.yml). Ran the below checks inside the Kind node (container) to validate
`ls -l /usr/src/linux-headers-$(uname -r) `-  Yes, Linux Headers are installed
`stat -fc %T /sys/fs/cgroup/` - the output is tmpfs so Yes, the cluster node is still using cgroup v1
2) Deployed sample apps like bookinfo and yelb as part of the cluster workload
3) Using the latest Kepler code-base, build the manifest using `make build-manifest OPTS="CI_DEPLOY"`, and then applied the generated deployment.yaml `kubectl apply -f _output/generated-manifest/deployment.yaml`
4) Also deployed the entire kube-prometheus monitoring stack

@rootfs , @SamYuan1990 - Thanks!! The metrics for kepler_container_joules_total and kepler_container_package_joules_total are attached. If they are good, how do we ensure they reflect in Prometheus and also the pre-generated [Kepler Dashboard](https://github.com/sustainable-computing-io/kepler/blob/main/grafana-dashboards/Kepler-Exporter.json) into Grafana as mentioned in the docs?
**Observation** - Pod line items with mode="dynamic" have non-zero values and the same pod entries with mode="idle" seem to have 0 value in the attached metrics file.

[Kepler-Exporter_kepler_container_joules_total.txt](https://github.com/sustainable-computing-io/kepler/files/11308128/Kepler-Exporter_kepler_container_joules_total.txt)

[Kepler-Exporter_kepler_container_package_joules_total.txt](https://github.com/sustainable-computing-io/kepler/files/11308130/Kepler-Exporter_kepler_container_package_joules_total.txt)

marceloamaral, 2023-04-24:
@sprathod369 this is the expected behaviour.

Since you are running in a VM, with no real-time power metrics, Kepler is using the power estimation mode.
```bash
I0411 08:35:23.817995       1 power.go:64] Not able to obtain power, use estimate method
```

Currently, the power estimation algorithm only calculates the dynamic power. This is because the idle power needs to be distributed among all VMs running on the node. The amount of idle power consumed by a virtual machine (VM) depends on the number of concurrently running VMs. However, it is not possible to determine the number of VMs running on the system from within the VM. Then, we only estimate the dynamic power.

rootfs, 2023-04-24:
@sprathod369 `kepler_container_joules_total` look good. The reason your dashboard doesn't show any plotting could have quite some possibilities:
1) can you check if the kepler `ServiceMonitor` is present?  `kubectl get servicemonitor -n kepler` will do the trick
2) can you query kepler metrics (e.g. `kepler_container_joules_total` from Prometheus UI, to make sure Prometheus actually scrapes the kepler endpoint?
3) In grafana, can you click into the panel and run the query manually?

cc @jiangphcn for helm chart connection

marceloamaral, 2023-04-25:
Within the Prometheus UI, there is a tab labeled "targets," which allows users to verify whether Prometheus is actively collecting metrics from an endpoint or not.

sprathod369, 2023-04-25:
Thanks @marceloamaral , @rootfs for the support. 

Yes, seems `kepler ServiceMonitor` was absent based on running `kubectl get servicemonitor -n kepler` as I only deployed the generated "`deployment.yaml`" to my cluster in the kepler namespace and the default `kube-prometheus` project as mentioned in the installation guide

This is what I have in my `_output` folder after building all the manifests using the CI_DEPLOY option

```
~/Kepler/kepler/_output/generated-manifest# ls
base  cluster-prereqs  dashboard  deployment.yaml  exporter  model-server  rbac

~/Kepler/kepler/_output/generated-manifest# ls exporter/
exporter.yaml  kustomization.yaml  kustomizeconfig.yaml  openshift_scc.yaml  patch  prometheus_service_monitor.yaml
```

I tried deploying the generated prometheus_service_monitor.yaml but I do not see any metrics if I search using "kepler" or even "energy" in the Prometheus metrics explorer so Prometheus is not actively collecting any kepler related metrics.

marceloamaral, 2023-04-25:
Different Prometheus deployments might have different labels to discover the servicemonitor.
We use the label `app.kubernetes.io/component: exporter` but you might need to use another one.

To verify which label you need check the `serviceMonitorSelector` configuration of your prometheus object.
```bash
 kubectl get prometheus -A
 kubectl get prometheus -n $namespace $name -o yaml
```

Btw, we probably need to include this in our Documentation....

sprathod369, 2023-04-25:
@marceloamaral - I'm on a learning curve with Prometheus and Grafana so will look at the labels, configuration etc. to troubleshoot the Grafana dashboard and Kepler metrics scrapping by Prometheus. Appreciate your inputs!!

Closing this issue as the Kepler metrics issue seems to be now resolved based on fix for cgroup v1  and @rootfs feedback / review of the data in Kepler Exporter logs. Thanks!!