sustainable-computing-io/kepler #630: duplicate implementation in code?

SamYuan1990, 2023-04-14:
https://github.com/sustainable-computing-io/kepler/blob/a6dce3b27561c0fb24c9bd33c254682813441424/pkg/cgroup/resolve_container.go#L264-L294

When we invoke this function, in most case we already have the cgroup_id there as input.
Hence why we re search the cgroup_id in code?

https://github.com/containerd/cgroups/blob/main/cgroup2/paths_test.go
https://github.com/containerd/cgroups/blob/main/cgroup1/paths_test.go

or we can use some verify function to ensure the cgroup exist?

SamYuan1990, 2023-04-14:
I am not sure if this one related with #594 , but our current implementation of this part may need a online/offline review?

marceloamaral, 2023-04-18:
You mean duplication in line 277? https://github.com/sustainable-computing-io/kepler/blob/a6dce3b27561c0fb24c9bd33c254682813441424/pkg/cgroup/resolve_container.go#L277

As far as I understood the `getPathFromcGroupID` function checks every path in the cgroup and retrieves the cgroup ID for each path. If the cgroup ID of a path matches the ID provided to the function, the function returns that path.

SamYuan1990, 2023-04-22:
> getPathFromcGroupID

as far as I remember @marceloamaral  told me offline? 
well I hope we can refactor by coming implement or add comments for the block I posted.
from line 264 we have a cgroup ID as input, and seems we double checked it at line 277. which looks wired, as we got cgroupid but don't trusted it....
if we have to check cgroup id, I hope we can refine line 264? or .... add some comments to avoid confusing.

marceloamaral, 2023-04-25:
Line 265 utilizes the `cgroupID` input variable to retrieve the path from the cache. And this variable is only used for that.

If the path is not found in the cache, the code proceeds to repopulate the entire cache, overwriting all entries in the cache with the cgroup ID of each path. It should be noted that this does not double-check the cgroup ID and instead reads the cgroup ID of every path, including all existing cgroup IDs in the system, which can be a costly operation.

However, it's worth mentioning that the new library, `pkg/cgroup/cgroup_stats_linux.go`, might implement this process much more efficiently.

closed 2023-08-28 by stale[bot]