sustainable-computing-io/kepler #638: command cannot be displayed normally

LeiZhou-97, 2023-04-20:
**Describe the bug**
 
Right now, there is a command length limit (len=10). If the length of command larger than 10, cuts it.
`kepler_container_package_joules_total{command="redis-serv", container_name="redis", container_namespace="default", mode="dynamic", pod_name="ei-redis-deployment-59b8dd9476-lv89k"} `

**Expected behavior**

command="redis-server"

Could we remove the length limitation, split the command string directly, and use the first string after splitting?

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rootfs, 2023-04-23:
@LeiZhou-97 unfortunately, the command line length is fixed in kernel. Getting the entire cmdline needs to parse the /proc/pid/cmdline. This could add overheads in reading files as well as in exporting metrics.

LeiZhou-97, 2023-04-24:
@rootfs got it, thanks for the explanation!

closed 2023-04-24 by LeiZhou-97