sustainable-computing-io/kepler #640: Container image release/versioning

yellowhat, 2023-04-22:
First of all. Thank you very much for this amazing project.

**Is your feature request related to a problem? Please describe.**

It is very difficult to incrementally update `kepler`:

- github releases do not have a corresponding container image tag, i.e there is no `v0.4.10`
- `latest` container tag points to a non github release, currently:
  - `latest` container tag is 3 days old while the latest github release is 1 month old
  - `release-0.4` container tag is 4 months old,  while the `release-0.4` release is 1 month old

therefore it is not guarantee to always use a "released" container image.

**Describe the solution you'd like**

- Create container image with `rpm_version` and `rpm_release` tag.
- Only the `release` job can create container image with `latest` tag
- The `image` job use commit sha tag and/or branch name, not `latest`

**Describe alternatives you've considered**
-

**Additional context**
-

jichenjc, 2023-04-24:
>>>latest container tag is 3 days old while the latest github release is 1 month old

latest is always last merged PR built image .. so I see no issue on this 

>>>release-0.4 container tag is 4 months old, while the release-0.4 release is 1 month old

0.5 is being planned, not sure we want to create 0.4.x ... @SamYuan1990 @rootfs FYI

yellowhat, 2023-04-24:
I am sorry, but this is not what I am pointing out.

let me clarify:

* the `latest` tag should point to the latest **release** (something known that should work) not a PR; a PR will have its own tag (ie `<branch>`)
* I was not pointing out how long it takes to release a new version, but the fact that there is a discrepancy between what `release-0.4` container tag is and what `release-0.4` github release is by looking at the creation date.
* If I would like to rollback to a previous release (ie 0.4.10 > 0.4.9) there is no container image after the creation of a new release.

Does this clarify my points?

Thanks

SamYuan1990, 2023-05-16:
it seems just updated with latest but not release-0.4 https://github.com/sustainable-computing-io/kepler/actions/runs/4437344205/jobs/7786948391#step:6:549 @rootfs

rootfs, 2023-05-17:
@SamYuan1990 can you make PR tagged build images? That'll be very helpful to share and validate fixes before merge.

SamYuan1990, 2023-05-17:
> @SamYuan1990 can you make PR tagged build images? That'll be very helpful to share and validate fixes before merge.

Let me think and try.
for PR level, 
- I don't want to build pre PR in automatic hence there will be lots of them and some thing as nit change( e.x. log typo) seems useless.
- but I do agree for specific PR, we build it in manually for sharing.

yellowhat, 2023-06-02:
#719 is a good progress.
But having a release name `release-0.5` is not very useful.
As patch release (ie 0.5.1) will overwrite it.

closed 2023-07-01 by yellowhat