sustainable-computing-io/kepler #645: Provide network/storage/accelerator power attribution

rootfs, 2023-04-24:


jichenjc, 2023-04-28:
what will be add in addition to our existing model (e.g the network card, the SAN or the GPU/DPU ) etc??

marceloamaral, 2023-05-01:
There are two approaches that we can take to tackle this issue.

First, we can conduct a fine-grain estimation of the power consumption of each process for a given device (i.e., CPU, DRAM, and GPU). Alternatively, we can estimate the total power consumption of the node using multiple metrics, leading to a coarse-grain estimation (i.e., Platform power).

The Kepler code already has the coarse-grain (i.e., Platform power) and fine-grain (i.e., CPU, DRAM, and GPU) estimations.

Therefore, If we can collect network and storage power consumption data, we can develop a power model to estimate the power consumption per process.

Furthermore, we can improve the current Platform power model by including more metrics that better reflect the utilization of other resources. However, we must ensure that the collected Platform power includes the power consumption of the target resource.

jichenjc, 2023-05-02:
>>> network and storage power consumption data

I assume the network = NIC /DPU, yes, this seems ok, so either coarse-grain or fine-grain should be able to provide such data
but storage usually SAN/SDS will be used which is distributed, so maybe not that easy..

jiere, 2023-05-09:
Due to accelerator power attribution topic, I think there are two kinds of definition of the word "accelerator":
1. Stand-alone HW accelerator, such as GPU. DPU also can be regarded as such accelerator since it offloads specific computing from CPU.
2. CPU built-in accelerator, such as Intel Xeon SP's bunch of in-die/in-package accelerators.
The work mechanism should be greatly different between the two. So let's think about how to organize the current code hierarchy of pkg/power and define appropriate framework to support different kinds of accelerators.

marceloamaral, 2023-05-09:
Sounds good.

But note that CPU built-in accelerator on Intel Processors is part of the RAPL UNCORE power.

jiere, 2023-05-09:
> But note that CPU built-in accelerator on Intel Processors is part of the RAPL UNCORE power.
I am sorry to say that the "uncore" type RAPL only belongs to client CPUs, such as Intel Core, Atom, etc. But for server CPUs such as Xeon SP, there is no "uncore" definition under RAPL sysfs... instead, there are only pkg+dram

rootfs, 2023-05-10:
05/08 meeting:
- Intel investigates accelerators (in-die and out-of-die). How to measure in-die accelerator: this can be done in kepler through RAPL pkg domain. The usage of the accelerator is cpu usage based. Out-of-die is measured separately.

jichenjc, 2023-05-10:
IBM Z also has a set of CPU built-in accelerator  like zip/crypto etc purpose so we can joint build the model with x86 world

rootfs, 2023-05-30:
May 30 meeting: using perf interface to collect stats from accelerator AMX, this will be made generic so that Kepler can use it for other type of accelerators. The overhead of continuous monitoring needs to be addressed, either by trading off overhead vs accuracy, or based on test results. Sampling can help reduce the monitoring the overhead.

closed 2023-08-28 by stale[bot]