sustainable-computing-io/kepler #654: Update base image

rootfs, 2023-04-26:
**Is your feature request related to a problem? Please describe.**
The base image is ubi8_6. The [cuda ubi9](https://hub.docker.com/layers/nvidia/cuda/12.1.0-base-ubi9/images/sha256-07b8daa23565abfa69e713dd14e0c734da8bc0eaa9caca0090d474b9872fc915?context=explore) image has been around for a while. In addition, ubi9 has s390 support

**Describe the solution you'd like**
Support both ubi8 and 9

@marceloamaral @jichenjc @SamYuan1990

jichenjc, 2023-04-28:
>>>I hope an ubi based image and install GPU by deb package.

UBI is RHEL based and DEB is Canonical based?

bradmccoydev, 2023-05-03:
Please find attached a security report with the CVE's for the current container and also a note of in what version they are fixed. https://artifacthub.io/packages/helm/kepler/kepler?modal=security-report

SamYuan1990, 2023-05-06:
> > > > I hope an ubi based image and install GPU by deb package.
> 
> UBI is RHEL based and DEB is Canonical based?

I mean the base image can be UBI instead of currently GPU based. and we can install GPU driver via DEB package.

SamYuan1990, 2023-05-06:
> Please find attached a security report with the CVE's for the current container and also a note of in what version they are fixed. https://artifacthub.io/packages/helm/kepler/kepler?modal=security-report

@rootfs I suppose we need to rebuild tag on-demand or the job build the image in a schedule as monthly? so that some CVE can be fixed through a monthly build.

closed 2023-08-28 by stale[bot]