sustainable-computing-io/kepler #660: Making sense of Kepler Grafana dashboard - Carbon Emission Metrics

sprathod369, 2023-04-28:
**Background:**
I'm running my Kepler setup on a single node KIND cluster that is runing on an Ubuntu 20x VM. Based on reading the docs, I think my approach is that of a "Minimum Deployment". I've also deployed the whole monitoring stack and after some troubleshooting (thanks to inputs from @marceloamaral ), my Prometheus setup is successfully and now actively collecting `kepler_*_joules_*` metrics from my Kepler exporter. 

**Making sense of the Kepler Grafana Dashboard**
My question is on the [Grafana Dashboard](https://github.com/sustainable-computing-io/kepler/blob/main/grafana-dashboards/Kepler-Exporter.json) json that shows the Carbon Emission. 
1) Do we have some documentation on this to better understand how these values are calculated and what they mean / indicate? (I did browse through to understand some of the expression mentioned in the Kepler-Exporter.json)
2) Can someone help me explain in layman terms what these CO2 emissions (Coal, Petroleum, Natural Gas) value indicate vis-a-vis the `kepler_*_joules_*` metrics coming from Prometheus?
3) How can I test the Carbon emission behavior further - is it by adding more intensive workloads or reducing workloads?
4) How can I test and check that by doing A B C I can reduce the CO2 emission (especially Coal and Petroleum values that do not move much over different timeframes) showing on the Grafana dashboard?

I was looking at some of the expression mentioned in the Kepler-Exporter.json file but if someone can throw more light it may help 
me map Kepler metrics to what story and insights are shaping up on Grafana dashboard? 
Below are my screenshots with a couple of sample apps running in their own namespace in my cluster)

[image]
[image]

rootfs, 2023-04-28:
@sprathod369 there is discussion in [this](https://github.com/sustainable-computing-io/kepler/discussions/600) thread and you can find some real time co2 intensity data sources there that can be used alongside of kepler metrics to calculate workload co2 footprint.

sprathod369, 2023-05-02:
Thanks @rootfs - I'll follow up the discussion you suggested. Looks interesting!!

@marceloamaral - Going through the [Grafana Dashboard](https://github.com/sustainable-computing-io/kepler/blob/main/grafana-dashboards/Kepler-Exporter.json) code that is available in the Kepler repo, my understanding is that the Carbon Emissions metrics that are projected on the dashboard basically using a formula that takes into account `kepler_container_package_joules_total` and `kepler_container_dram_joules_total` values from kepler exporter that is then multiplied by a FIXED Coeff value(Coal or Petroleum or Natural Gas) - is my understanding correct?

The "standard" carbon factors" that we consider in the example dashboard has an underlying assumption that power source powering this Kubernetes cluster is sourced from "fossil fuel" and not any renewable source (like wind, solar, hydrogen etc) - is my thinking correct? 
Thank you!!

marceloamaral, 2023-05-02:
@sprathod369 that is correct. But as @rootfs commented before, there is an effort to use real time carbon metrics as detailed in #600. But of course it will change depending on the data center power source.

closed 2023-05-09 by sprathod369