sustainable-computing-io/kepler #663: Remove Code and Data duplication

marceloamaral, 2023-05-01:
Currently, the Kepler code uses two data structures: `ContainerMetrics` and `ProcessMetrics`, which contain similar information. However, `ContainerMetrics` contains an aggregation of `ProcessMetrics`, resulting not only in duplicated data but also duplicated code for energy consumption calculations and duplicated Prometheus metrics for `kepler_process...` and `kepler_container...`.

Additionally, as far as I understood before, the original idea behind exposing process metrics was to allow Kepler to run standalone without the need to report container metrics. However, the current code uses the container power to estimate the process power in the ratio power model. This estimation should instead utilize the node power instead of the container power.

To enhance the code and improve consistency, we should simplify the code to have only one map, such as `ProcessMetrics` only, where the map key will correspond to the level of aggregation configured in Kepler (`Container ID`, `Pod ID`, or `Process ID`). This approach enables better scalability by reporting lower metrics as the level of aggregation increases.

When per-process metrics are enabled, there is no need to aggregate metrics per container in the map. Instead, process metrics can be exposed, including labels for process name/command, container name and ID, and pod name. This requires only the `kepler_process...` metrics. When Kepler aggregates data per container, the process-related labels could be empty if per-process metrics is not enabled, whereas if Kepler reports metrics per process, the container labels may exist or not if if Kepler is running in Kubernetes or standalone.

By adopting this approach, we can achieve a more efficient, flexible, and consistent way of exposing metrics. Users can aggregate per-container and per-pod metrics using the process metrics in a Prometheus query. 

Additional information on this topic can be found in issue #583.

marceloamaral, 2023-05-01:
/cc @rootfs @jichenjc @sunya-ch @SamYuan1990 ^^

SamYuan1990, 2023-05-01:
I am good with this proposal.

rootfs, 2023-05-01:
sounds good!

jichenjc, 2023-05-02:
>>>where the map key will correspond to the level of aggregation configured in Kepler (Container ID, Pod ID, or Process ID)

agree this is the right approach and yes, it's flexible and easy to handle

jichenjc, 2023-05-30:
some design as reference and discussion:

```
type ProcessMetrics struct {
        PID          uint64
        Command      string
        CounterStats map[string]*types.UInt64Stat
        // ebpf metrics
        CPUTime             *types.UInt64Stat
        SoftIRQCount        []types.UInt64Stat      

        DynEnergyxxx    ==> a list of Dyn energy 
        IdleEnergyxxx     ==> a list of Idle energy
```

container metrics (in addition to ProcessMetrics)
```
        CGroupPID     uint64
        PIDS          []uint64       ==>  PIDs of the current container 
        ContainerName string
        PodName       string 
        Namespace     string
        ContainerID   string         

        CurrProcesses int         => how many active process of current pod 

        CgroupStatHandler cgroup.CCgroupStatHandler
        CgroupStatMap     map[string]*types.UInt64StatCollection
        // TODO: kubelet stat metrics is deprecated since it duplicates the cgroup metrics. We will remove it soon.
        KubeletStats map[string]*types.UInt64Stat

```

so we should consider consolidate them into one map the map key will be `processID` while the map struct will be TBD
 N process : 1 container  and    M container : 1 pod , so the map will be K process, we can aggregate them into container level (provide backward compatible) and currently kepler only provide kepler_process_xx and kepler_container_xxx, so no pod info

jichenjc, 2023-06-02:
>>>The use of processID or containerID should be defined if process metrics are enabled or not..

>>>where the map key will correspond to the level of aggregation configured in Kepler (Container ID, Pod ID, or Process ID)

you mean we create a `processmetrics` map variable and the depends on whether PROCESS enabled or not , 
use something like processmetrics[processID ] or processmetrics[containerID]? 
the contents will be something like above struct content?

>>>When per-process metrics are enabled, there is no need to aggregate metrics per container in the map. Instead, process metrics can be exposed, including labels for process name/command, container name and ID, and pod name.

so something like kepler_process_xxx will be the only report metrics (no kepler_container_xxx) ?

>>>When Kepler aggregates data per container, the process-related labels could be empty if per-process metrics is not enabled, whereas if Kepler reports metrics per process, the container labels may exist or not if if Kepler is running in Kubernetes or standalone.

how about this ? whether we need report kepler_container_xxx but the kepler_process-xxx is not reported ?

marceloamaral, 2023-06-02:
> use something like processmetrics[processID ] or processmetrics[containerID]?

Yes!

If process metrics are enabled do we need to report container metrics? What is the use case?

On the other hand, If process metrics are not enable, it is easy, we just report `kepler_container_*`

jichenjc, 2023-06-02:
>>>If process metrics are enabled do we need to report container metrics? What is the use case?

 e.g do we need consider running container + process workload ? at current design 
node either run container or traditional workload? or will this be something that need customer handle? 

something like: 
if customer has container workload only, they can get container metrics
but if they need process and container, they need aggregate from kepler_process_xx ?

this changed user expectation..

marceloamaral, 2023-09-21:
The PR #931 introduced even more duplication.

marceloamaral, 2023-11-21:
I am working on this.

closed 2024-01-15 by marceloamaral