sustainable-computing-io/kepler #671: Latency increase on kernel rt

alysondeives, 2023-05-05:
**Describe the bug**
Using the cyclictest tool from [rt-tests](https://git.kernel.org/pub/scm/utils/rt-tests/rt-tests.git/tree/src/cyclictest?h=stable/v1.0) show an increase in latency in comparison when kepler is not running

**To Reproduce**
Steps to reproduce the behavior:
1. Isolate cores. For instance, my /boot/efi/EFI/BOOT/boot.env contains: nohz_full=2-11,32-41 isolcpus=2-11,32-41 rcu_nocbs=2-63 kthread_cpus=0-1 irqaffinity=12-31,42-63
2. Run cyclictest without kepler running
3. Run cyclictest with kepler running
4. Kepler results has increased values in latency.

**Expected behavior**
I think Kepler should not affect latency this much.

**Desktop (please complete the following information):**
 - OS: Debian 11 with linux-yocto rt kernel 5.10

**Additional context**

Results without kepler running:

cyclic test latencies:
Min Latencies: 02693 02742 02682 02621 02662 02658 02677 02671
Avg Latencies: 03068 03176 02892 02857 02865 02889 02885 02895
Max Latencies: 04994 05403 04615 04232 04502 04329 04385 05497 05497

kernel tracing snippet:
<idle>-0       [034] d...2.. 108530.157786: sched_switch: prev_comm=swapper/34 prev_pid=0 prev_prio=120 prev_state=R ==> next_comm=cyclictest next_pid=635853 next_prio=4
<...>-635853  [034] ....1.. 108530.157787: sys_clock_nanosleep -> 0x0
<...>-635853  [034] ....1.. 108530.157788: sys_clock_nanosleep(which_clock: 1, flags: 1, rqtp: 7fef5851c910, rmtp: 0)
<...>-635853  [034] ....... 108530.157788: hrtimer_init: hrtimer=0000000013fb1e26 clockid=CLOCK_MONOTONIC mode=ABS
<...>-635853  [034] d...1.. 108530.157788: hrtimer_start: hrtimer=0000000013fb1e26 function=hrtimer_wakeup expires=108530160820232 softexpires=108530160820232 mode=0x8
<...>-635853  [034] d...1.. 108530.157788: rcu_utilization: Start context switch
<...>-635853  [034] d...1.. 108530.157788: rcu_utilization: End context switch
<...>-635853  [034] d...2.. 108530.157789: sched_switch: prev_comm=cyclictest prev_pid=635853 prev_prio=4 prev_state=S ==> next_comm=swapper/34 next_pid=0 next_prio=120

 

Results with kepler running:

cyclic test latencies:
Min Latencies: 04356 04224 04278 04203 04203 04325 04147 04200
Avg Latencies: 05276 04606 04736 04918 05536 04730 04568 04698
Max Latencies: 07668 06817 06735 07178 08312 06960 06443 06414 08312

kernel tracing snippet:
<idle>-0       [034] d...2.. 106838.048757: sched_switch: prev_comm=swapper/34 prev_pid=0 prev_prio=120 prev_state=R ==> next_comm=cyclictest next_pid=496624 next_prio=4
**<...>-496624  [034] d...311 106838.048759: rdpmc: 40000001, value 80c754c6d255**
**<...>-496624  [034] d...311 106838.048759: rdpmc: 40000002, value 80995aaa3541**
**<...>-496624  [034] d...311 106838.048759: rdpmc: 40000000, value 803dd8124938**
**<...>-496624  [034] d...311 106838.048759: rdpmc: 0, value 8000006d4b65**
<...>-496624  [034] ....1.. 106838.048760: sys_clock_nanosleep -> 0x0
<...>-496624  [034] ....1.. 106838.048761: sys_clock_nanosleep(which_clock: 1, flags: 1, rqtp: 7fdbeb566910, rmtp: 0)
<...>-496624  [034] ....... 106838.048761: hrtimer_init: hrtimer=000000004791ce7a clockid=CLOCK_MONOTONIC mode=ABS
<...>-496624  [034] d...1.. 106838.048761: hrtimer_start: hrtimer=000000004791ce7a function=hrtimer_wakeup expires=106838051041009 softexpires=106838051041009 mode=0x8
<...>-496624  [034] d...1.. 106838.048761: rcu_utilization: Start context switch
<...>-496624  [034] d...1.. 106838.048761: rcu_utilization: End context switch
<...>-496624  [034] d...2.. 106838.048762: sched_switch: prev_comm=cyclictest prev_pid=496624 prev_prio=4 prev_state=S ==> next_comm=swapper/34 next_pid=0 next_prio=120

The kernel tracing snippet when kepler is running shows 4 rdpmc calls, which seems to be related to the hardware counters read by kepler (cpu cycles, cpu ref cycles, cpu instructions, and cache miss)

is there any way to reduce this latency when reading hardware counters?
I just need to measure energy consumption from Intel CPUs, so just the cpu instructions hw counter is necessary, right?
Is it possible to adjust kepler code so we are able to specify which hw counter to read?

rootfs, 2023-05-05:
@alysondeives thanks for reporting!

You can try the follow hacks.
- use conditional compile tag (such as [this one](https://github.com/sustainable-computing-io/kepler/blob/main/bpfassets/perf_event/perf_event.c#L195) ) to disable other perf counters than cpu instructions. 
- only choose the needed counters in [this map](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/bpfassets/attacher/bcc_attacher.go#L56)

Let me know if you have any success.

rootfs, 2023-05-05:
Turning off perf counter reading reduces ebpf runtime as well

Using `bpftool prog show`, I got these results:

### With perf counter probing, the average ebpf runtime is 1521 ns (i.e. 2339982948/1538185)
```
1558: kprobe  name finish_task_swi  tag ffe5f649866774cd  gpl run_time_ns 2339982948 run_cnt 1538185
        loaded_at 2023-05-04T21:22:34-0400  uid 0
        xlated 2896B  jited 1620B  memlock 4096B  map_ids 2208,2207,2206,2205,2204,2203,2202,2201,2200,2199,2209
        btf_id 203
        pids kepler(3106479)
```
### turning off perf counter reading, the average runtime is 775ns (i.e. 699780591/902460)
```
1655: kprobe  name finish_task_swi  tag efa67d0639adece5  gpl run_time_ns 699780591 run_cnt 902460
        loaded_at 2023-05-05T12:12:13-0400  uid 0
        xlated 832B  jited 478B  memlock 4096B  map_ids 2576,2577
        btf_id 253
        pids kepler(3503624)
```

alysondeives, 2023-05-11:
Hi @rootfs, unfortunately the changes in #665 did not solve the latency increase. 
One thing that reduced the latency was to not attach the ebpf assets. (However it is still higher  when comparison when kepler is not running.
But I'm not sure if disabling the ebpf code, kepler can still calculate the metrics using only cgroup metrics.

marceloamaral, 2023-05-12:
@alysondeives, in specific scenarios like this, we could have the option to disable BPF.

Cgroup metrics also introduce some overhead, so we need to consider the following options:
-Disable BPF entirely and keep the cgroup metrics.
-Only disable the BPF performance counters and disable cgroup metrics entirely.

It would be really helpful if you could test both scenarios and provide feedback.

If we decide to disable BPF completely, the PR #635 is a step towards enabling Kepler to run without BPF. 
Note that the currently Kepler code relies on BPF to discover new containers, but PR #635 introduces logic to rely on the apiserver for discovering new containers as well. However, please note that this PR does not immediately allow us to disable BPF; we will need another PR to enable that.

rootfs, 2023-05-15:
There are a few things we need to consider. 
- ebpf overhead can be managed. Currently ebpf probes *all* context switch. This frequency can be reduced to e.g. every 10 context switches, without losing fidelity.  
- reading cgroup metrics doesn't scale well. I see 30% CPU spent in cgroup stat when running kepler in rt kernels.
- cgroup metrics don't tell much about system processes that are running outside of cgroups.

alysondeives, 2023-05-16:
@marceloamaral I would like to evaluate disabling ebpf entirely and use cgroup metrics instead. I'm trying this approach using this commit (https://github.com/alysondeives/kepler/commit/8949412d3c35c5a7110444b9dc7c8f266e0b80e6), however I'm still getting segfault when process metrics is enabled. PR #635 would solve it?

@rootfs How can I reduce the frequency of ebpf probes? Can it be parameterized via environment vars?

rootfs, 2023-05-16:
@alysondeives 
> How can I reduce the frequency of ebpf probes? Can it be parameterized via environment vars?

My thought is that we can add a count down mechanism (implemented as a hash map or global variable) in eBPF, so we can control the frequency of `kprobe__finish_task_switch` function get called: the function is bypassed until count down expires. The count down counter can be passed as env var.

Note that, we are using hash map in the eBPF code, this is not as efficient as global variables. We should investigate this as well.

marceloamaral, 2023-05-18:
Let's try to have sampling rate configuration as tracing tools.

https://www.jaegertracing.io/docs/1.45/sampling/#:~:text=All%20operations%20for%20service%20bar,0.5%20defined%20by%20the%20default_strategy%20.

- **Probabilistic** (sampler.type=probabilistic) sampler makes a random sampling decision with the probability of sampling equal to the value of sampler.param property. For example, with sampler.param=0.1 approximately 1 in 10 traces will be sampled.
- **Rate Limiting** (sampler.type=ratelimiting) sampler uses a leaky bucket rate limiter to ensure that traces are sampled with a certain constant rate. For example, when sampler.param=2.0 it will sample requests with the rate of 2 traces per second.

But let's stat with the probabilistic approach since it will be easier to implement.

rootfs, 2023-07-24:
to be fixed in #823

rootfs, 2023-09-20:
with ebpf sampling merged, this should be resolved. closing this PR until further update.

closed 2023-09-20 by rootfs