sustainable-computing-io/kepler #677: release-0.3 Kepler Error Log

LAI-chuchi, 2023-05-08:
**Describe the bug**
When deploy the Kepler from the branch release-0.3 of Source Code. And the error occured.

My OS: Ubuntu 20.04.4 LTS (GNU/Linux 5.4.0-121-generic x86_64)

**To Reproduce**
 

`git clone -b v0.3 https://github.com/sustainable-computing-io/kepler`
`cd ~/kepler/`
`kubectl create -f manifests/kubernetes/deployment.yaml`
`kubectl logs  -n kepler  kepler-exporter-xxxxx `

I0508 14:18:03.258437       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0508 14:18:03.261762       1 exporter.go:148] Kepler running on version: 09b28a7
I0508 14:18:03.261779       1 config.go:172] using gCgroup ID in the BPF program: true
I0508 14:18:03.261830       1 config.go:174] kernel version:
I0508 14:18:03.261844       1 exporter.go:161] EnabledBPFBatchDelete: true
I0508 14:18:03.261964       1 power.go:77] Not able to obtain power, use estimate method
I0508 14:18:03.899190       1 exporter.go:174] Initializing the GPU collector
modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.4.0-121-generic
chdir(/lib/modules/5.4.0-121-generic/build): No such file or directory
I0508 14:18:03.905765       1 bcc_attacher.go:70] failed to attach the bpf program: <nil>
I0508 14:18:03.905782       1 bcc_attacher.go:130] failed to attach perf module with options [-DNUM_CPUS=4]: failed to attach the bpf program: <nil>, not able to load eBPF modules
I0508 14:18:03.905845       1 exporter.go:191] failed to start : failed to attach bpf assets: failed to attach the bpf program: <nil>
I0508 14:18:03.905990       1 exporter.go:218] Started Kepler in 644.273794ms

**Expected behavior**
http://localhost:9102/metrics      should have metrics

rootfs, 2023-05-08:
@LAI-chuchi this log message indicates the kernel headers are missing. Can you install the headers and run kepler again?

```
modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.4.0-121-generic
chdir(/lib/modules/5.4.0-121-generic/build): No such file or directory
```

LAI-chuchi, 2023-05-12:
@rootfs After I reinstall the Kernel headers and run kepler again. And an error related to BPF occurred.  How can I solve it?

```
I0512 08:30:36.585627       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0512 08:30:36.588348       1 exporter.go:148] Kepler running on version: 25cf0b3
I0512 08:30:36.588363       1 config.go:172] using gCgroup ID in the BPF program: true
I0512 08:30:36.588387       1 config.go:174] kernel version: 5.4
I0512 08:30:36.588404       1 exporter.go:161] EnabledBPFBatchDelete: true
I0512 08:30:36.588487       1 power.go:77] Not able to obtain power, use estimate method
I0512 08:30:37.006245       1 exporter.go:174] Initializing the GPU collector
I0512 08:30:38.322258       1 bcc_attacher.go:153] Successfully load eBPF module with option: [-DNUM_CPUS=4]
I0512 08:30:38.339042       1 exporter.go:218] Started Kepler in 1.750703453s
I0512 08:30:41.365645       1 container_hc_collector.go:172] resetting EnabledBPFBatchDelete to false
```

marceloamaral, 2023-05-12:
@LAI-chuchi, which error? `resetting EnabledBPFBatchDelete to false`?

BPFBatchDelete is only related to performance and can only be used in Kernels that support it.

marceloamaral, 2023-05-12:
@LAI-chuchi, yes!

Maybe we should improve this log message by providing a more informative explanation.

LAI-chuchi, 2023-05-12:
@marceloamaral Sorry, I  I want to ask you another question. If my link(http://localhost:9102/metrics) doesn't capture all of pods, how can I check the error?

marceloamaral, 2023-05-15:
@LAI-chuchi, can you see some metrics but some pods are missing. Is that your problem?

Also, which Kubernetes Cluster are you using?
We just had a merged PR (#684) that fixed the containedID translation in minikube clusters.

LAI-chuchi, 2023-05-15:
@marceloamaral, Yes, this is my problem.

I think the source of the problem is caused by me using branch release-0.3 or I am using 1.16.0 version of Kubernetes?What do you think?

The website mentioned the installation requirements: kubectl v1.21.0 . Is this restriction limited to Build manifests when deploy the Kepler?

marceloamaral, 2023-05-15:
Can you try to build and deploy the Kepler from the github main branch?

LAI-chuchi, 2023-05-15:
@marceloamaral Sure, But when I Build manifests, the error will occurs. Is it my Kubernetes version 1.16.0 enough?

**Command**
` make build-manifest`

**error log**
make: command: Command not found
make: command: Command not found
mkdir -p bin
go: creating new go.mod: module tmp
Downloading sigs.k8s.io/kustomize/kustomize/v4@v4.5.2
can't load package: package sigs.k8s.io/kustomize/kustomize/v4@v4.5.2: can only use path@version syntax with 'go get'
make: *** [Makefile:244: kustomize] Error 1

closed 2023-08-28 by stale[bot]