sustainable-computing-io/kepler #679: Segmentation violation when ENABLE_PROCESS_METRICS is True and EXPOSE_HW_COUNTER_METRICS is False

alysondeives, 2023-05-11:
**Describe the bug**

The following SEGV error occurs when ENABLE_PROCESS_METRICS is set to True and EXPOSE_HW_COUNTER_METRICS is set to False:

```
2023-05-11T14:50:07.247197018Z stderr F I0511 14:50:07.247156       1 node_energy_collector.go:57] System energy collection is supported
2023-05-11T14:50:07.290440491Z stderr F panic: runtime error: invalid memory address or nil pointer dereference
2023-05-11T14:50:07.29046993Z stderr F [signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0xc71f39]
2023-05-11T14:50:07.290472916Z stderr F 
2023-05-11T14:50:07.290475758Z stderr F goroutine 91 [running]:
2023-05-11T14:50:07.290478791Z stderr F github.com/sustainable-computing-io/kepler/pkg/model/estimator/local.UpdateProcessComponentEnergyByRatioPowerModel(0xc00039fb00, 0xc0005ea000, {0xe78e33, 0x8}, {0xc00003a360, 0x9}, 0x756c6f762f736e69?)
2023-05-11T14:50:07.290546201Z stderr F         /opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/model/estimator/local/ratio_process.go:45 +0x1b9
2023-05-11T14:50:07.290550838Z stderr F created by github.com/sustainable-computing-io/kepler/pkg/model.UpdateProcessEnergy
2023-05-11T14:50:07.29055398Z stderr F  /opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/model/process_power.go:78 +0x27a
```

jichenjc, 2023-05-12:
I didn't see this in latest image ...
```
# kubectl logs ds/kepler-exporter -n kepler  | head
....
I0512 06:21:56.175756       1 config.go:115] ENABLE_PROCESS_METRICS: true
I0512 06:21:56.175759       1 config.go:116] EXPOSE_HW_COUNTER_METRICS: false
```

jichenjc, 2023-05-12:
Reviewed the code location and I guess it's related to HW counter not enabled when access the array variable..
submit a PR for this

PR #682 (2023-05-16): fix: avoid use metrics if HW counter not open
Fixes: #679
closed 2023-05-16 by rootfs