sustainable-computing-io/kepler #685: Add SBOM to release

bradmccoydev, 2023-05-14:
**Is your feature request related to a problem? Please describe.**
Currently, we do not have an SBOM attached to the release of Kepler. Having one attached will help us identify what components we have if there was a vulnerability in one and also help adopters with securing their software supply chain.

A software bill of materials (SBOM).  is a list of all the open-source and third-party components present in a codebase. An SBOM also lists the licenses that govern those components, the versions of the components used in the codebase, and their patch status, which allows security teams to quickly identify any associated security or license risks.

**Describe the solution you'd like**
We can easily put an additional step on the release.yml to attach the SBOM to the release, not changing or breaking anything.

I'm happy to do this work.

jichenjc, 2023-05-15:
@bradmccoydev thanks , assigned to you just now

@marceloamaral @rootfs FYI a good improvement to kepler

marceloamaral, 2023-05-15:
@bradmccoydev and @jichenjc sounds like a good plan!

It is also important to establish some recommendations for defining the best way to handle code updates.

SamYuan1990, 2023-05-16:
Cool!
is there any SBOM which able to support both docker/container and binary?

bradmccoydev, 2023-05-21:
Hello, I have finished my testing on this and it worked fine. I have made the PR

closed 2023-05-21 by rootfs