sustainable-computing-io/kepler #689: Expose fine-grained metrics

fshiga-01, 2023-05-15:
**Is your feature request related to a problem? Please describe.**
Currently Kepler exposes energy consumption metrics and also metrics related to cpu utilization, bpf and cgroups. Giving the option to publish more fine-grained node metrics as CPU operating frequencies, CPU temperature, CPU time spent in a c-state, power consumption of processor package, among others would give more information for monitoring systems to act.

**Describe the solution you'd like**
Create new metrics for data coming from msr, rapl, cpufreq and uncore frequency. Most of these data sources is already being used by Kepler so the idea is read new data from them, create the metrics and add flags to enable/disabling them.

**Describe alternatives you've considered**
-

**Additional context**
I would like to contribute to the project by working on this feature. There is already a metric for the current frequency of CPU (kepler_node_cpu_scaling_frequency_hertz) and I'm thinking in use it as a model to include the others metrics.

Studying the code I saw that if RAPL is enabled the MSR is not read (https://github.com/sustainable-computing-io/kepler/blob/v0.4.11/pkg/power/components/power.go#L51). To add other metrics it would be necessary to read from MSR and also include the reading of intel-uncore-frequency data.

Looking at the issues there are other related requests (#669, #68)

rootfs, 2023-05-15:
Thanks for the info. Some of the metrics are available in node_exporter too. If you can come up with a proposal of the metrics that help understand the cpu energy and performance, that'll be great.

fshiga-01, 2023-05-15:
Thank you @rootfs. I will create a PR giving more details about the metrics.

fshiga-01, 2023-05-16:
Hi @rootfs,

Based on the metrics provided by [Linux rapl metricset](https://www.elastic.co/guide/en/beats/metricbeat/8.7/metricbeat-metricset-linux-rapl.html), [collected turbostat plugin](https://collectd.org/wiki/index.php/Plugin:turbostat) and [Intel PowerStat Input Plugin](https://github.com/influxdata/telegraf/blob/master/plugins/inputs/intel_powerstat) my initial idea is to add the following information:

- package_power_consumption_watts: Current processor package power consumption in Watts (rapl).
- package_power_dram_consumption_watts: Current processor package DRAM subsystem power consumption in Watts (rapl).
- package_tdp_watts: Thermal Design Power (TDP) for processor package in Watts (rapl).
- package_cpu_base_frequency_mhz - cpu base frequency for the processor package in Mhz.
- package_uncore_frequency_mhz: Current uncore frequency for the processor package in Mhz (intel_uncore_frequency).
- cpu_mininum_frequency_mhz: Minimum cpu frequency setting in MHz.
- cpu_maximum_frequency_mhz: Maximum cpu frequency setting in MHz.
- cpu_scaling_frequency_hertz: Current average cpu frequency in Hertz (cpufreq) - Already implemented.
- cpu_busy_frequency_mhz: Current cpu busy frequency in MHz (msr).
- cpu_c0_state_percent: Percent of time that cpu spent in C0 state (msr).
- cpu_c1_state_percent: Percent of time that cpu spent in C1 state (msr).
- cpu_c6_state_percent: Percent of time that cpu spent in C6 state (msr).
- cpu_temperature_celsius: Current temperature of cpu in Celsius (msr).

For settings information, as max and min frequency, I am thinking to include them in kepler_node_nodeInfo (kepler_node_info).

rootfs, 2023-05-16:
@fshiga-01 looks good. For node level metrics, please use `kepler_node_` prefix, and container level with `kepler_container_`

closed 2023-08-28 by stale[bot]