sustainable-computing-io/kepler #703: not report newly deployed pod metrics

jichenjc, 2023-05-22:
**Describe the bug**
A clear and concise description of what the bug is.

reported this before, but seems still exist with latest code 

created pod and wait for 15 min
```
# kubectl get po -n default
NAME                               READY   STATUS    RESTARTS   AGE
nginx-deployment-cbdccf466-glkdv   1/1     Running   0          15m
nginx-deployment-cbdccf466-xv8zz   1/1     Running   0          15m
```

but the pod is not reported in metrics at all..
```
# curl localhost:9102/metrics | grep kepler_container_package_joules_total | grep nginx
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  107k    0  107k    0     0  54.5M      0 --:--:-- --:--:-- --:--:--  105M
(base) [root@lozlp51 ~]#
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marceloamaral, 2023-05-22:
There is a bug, that if a container is starting, the pod might not have the container ID yet. But if  a pod is wrongly identified as system process, it will never be fixed later. The apiserver PR (#635) fix this.

jichenjc, 2023-07-03:
not sure someone already verified this ? if anyone did and it works fine , I think we can close this 
if not , I will try to do some test on it?

jichenjc, 2023-07-06:
did a quick test with new deployed pod , I can see those reported after a few seconds, so closing this issue
thanks @marceloamaral

closed 2023-07-06 by jichenjc