sustainable-computing-io/kepler #712: [RFC] Platform validation in Kepler.

jiere, 2023-05-28:
**Is your feature request related to a problem? Please describe.**
Currently, Kepler's test cases focus on code logic verification, lack of actual platform validation.
Kepler's data files contain static information and lack of test cases and test report as the evidence that whether the specific CPU model is well supported in Kepler, whether the data/metrics of specific power source is correctly exported, whether the power measurement accuracy is acceptable, etc. 

**Describe the solution you'd like**
We would suggest to add platform validation test cases and define content template of the test report. The specific tasks could be broken down as follows:

- [ ] Platform validation guide document.
- [ ] Platform validation test cases design and implementation.
- [ ] Platform validation test report format definition, report generation and release process definition.

**Describe alternatives you've considered**
N/A.

**Additional context**
The current Kepler test framework is based on ginkgo, we will do more investigation to decide whether use Golang(align with ginkgo) or bash shell to implement the automation test logic. 

We are  proposing to contribute in this area in community.

There is an enhancement request PR [here](https://github.com/sustainable-computing-io/kepler/pull/711).

kenplusplus, 2023-05-28:
I also hope we can create a standard process or tool for early-silicon processor, so it can make Kepler can better work with the latest processor or platform quickly.

rootfs, 2023-05-28:
For baremetal machines Kepler uses ratio power attribution, so it is a lesser issue than running Kepler on VMs where ML models based estimate is used.

This platform validation will cover both baremetal and VM environments.

rootfs, 2023-05-28:
https://github.com/sustainable-computing-io/kepler-model-server/issues/100

closed 2023-08-28 by stale[bot]

jiere, 2023-08-28:
@rootfs , please help reopen this issue, since it has been chosen as part of 0.6.0 release features.
We will submit PR and update this feature status soon.