sustainable-computing-io/kepler #713: Calendar invite for biWeekly in the Readme is not working

nttDamien, 2023-05-29:
**Describe the bug**
Imported the iCal in outlook to attend the biweekly, but no meeting appeared...

**To Reproduce**
Steps to reproduce the behavior:
1. open outlook online,
2. Drap and drop the .ics file
3. check different Tuesdays for the meeting

**Expected behavior**
a biweekly meeting on Tuesday appears with zoom call id to join

**Desktop (please complete the following information):**
 - OS Windows 11, Chrome 113.0.5672.127 

**Additional context**
Also tried on Android: "unable to launch event" error
Looked in the iCal, there is some info (NY time), but it seems to be missing the zoom call details at least.

rootfs, 2023-05-30:
The previous community meeting calendar ended. We will move to new schedules starting in the middle of June. Stay tuned!

rootfs, 2023-06-02:
@nttDamien the new meeting calendar is posted 
https://github.com/sustainable-computing-io/kepler/pull/720

closed 2023-06-02 by rootfs

rootfs, 2023-06-07:
The zoom meeting is migrating to CNCF zoom. Please check the info [here ](https://github.com/sustainable-computing-io/community/blob/main/community-event.md)

@nttDamien

nttDamien, 2023-06-08:
@rootfs thanks a lot for pinging me on this. I'll update !