sustainable-computing-io/kepler #727: failed to attach BPF with IRQ tracepoint

sunya-ch, 2023-06-07:
**Describe the bug**
A clear and concise description of what the bug is.

I got the following error which causes the BPF attachment failure when deploying kepler on minikube on my machine.

```
/virtual/main.c:239:1: warning: declaration of 'struct tracepoint__irq__softirq_entry' will not be visible outside of this function [-Wvisibility]
TRACEPOINT_PROBE(irq, softirq_entry)
^
/virtual/include/bcc/helpers.h:1317:46: note: expanded from macro 'TRACEPOINT_PROBE'
int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)
                                             ^
<scratch space>:111:1: note: expanded from here
tracepoint__irq__softirq_entry
^
/virtual/main.c:246:17: error: incomplete definition of type 'struct tracepoint__irq__softirq_entry'
        if (args->vec < 10) {
            ~~~~^
/virtual/main.c:239:1: note: forward declaration of 'struct tracepoint__irq__softirq_entry'
TRACEPOINT_PROBE(irq, softirq_entry)
^
/virtual/include/bcc/helpers.h:1317:46: note: expanded from macro 'TRACEPOINT_PROBE'
int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)
                                             ^
<scratch space>:111:1: note: expanded from here
tracepoint__irq__softirq_entry
^
/virtual/main.c:247:41: error: incomplete definition of type 'struct tracepoint__irq__softirq_entry'
            process_metrics->vec_nr[args->vec] ++;
                                    ~~~~^
/virtual/main.c:239:1: note: forward declaration of 'struct tracepoint__irq__softirq_entry'
TRACEPOINT_PROBE(irq, softirq_entry)
^
/virtual/include/bcc/helpers.h:1317:46: note: expanded from macro 'TRACEPOINT_PROBE'
int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)
                                             ^
<scratch space>:111:1: note: expanded from here
tracepoint__irq__softirq_entry
^
1 warning and 2 errors generated.
```

Currently, I avoid this issue by defining SET_IRQ to make it optional. 
```golang
> bcc_attacher.go
func AttachBPFAssets() (*BpfModuleTables, error) {
...
	if config.ExposeIRQCounterMetrics {
		options = append(options, "-DSET_IRQ")
	}

> perf_event_bindata.go
#ifdef SET_IRQ
// per https://www.kernel.org/doc/html/latest/core-api/tracepoint.html#c.trace_softirq_entry
TRACEPOINT_PROBE(irq, softirq_entry)
{
    ...
}
#endif
``` 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jichenjc, 2023-06-15:
should we adopt the workaround as less likely we can change minikube itself ?

sunya-ch, 2023-06-21:
I found that the tracepoint is actually there.

```
docker@minikube-docker:~$ grep softirq_entry /proc/kallsyms
0000000000000000 T __traceiter_softirq_entry
0000000000000000 T __SCT__tp_func_softirq_entry
0000000000000000 d __tpstrtab_softirq_entry
0000000000000000 d event_softirq_entry
0000000000000000 D __SCK__tp_func_softirq_entry
0000000000000000 D __tracepoint_softirq_entry
0000000000000000 d __bpf_trace_tp_map_softirq_entry
0000000000000000 d __event_softirq_entry
```

Similarly to this issue https://github.com/iovisor/bcc/issues/2201, I think the problem may come from the access permission of minkube user permission.

sunya-ch, 2023-06-22:
I think this issue should be solved by enabling permission access.

closed 2023-06-22 by sunya-ch