sustainable-computing-io/kepler #738: e2e metric test cases were not executed in CI integration test.

jiere, 2023-06-20:
**Describe the bug**
Current CI jobs only execute 3 cases in e2e integration test.

**To Reproduce**
Steps to reproduce the behavior:
1. make cluster-up && make cluster-sync.
2. Follow test case [here](https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/integration_test.yml#L38-L48)
3. Check on console log, shows like this:
Ran 3 of 3 Specs in 0.005 seconds
SUCCESS! -- 3 Passed | 0 Failed | 0 Pending | 0 Skipped
PASS
4. cd e2e && ginkgo -v to check the verbose output of Ginkgo, like this:
$ ginkgo
Running Suite: E2eTest Suite - /home/jie/dev/kepler/e2e
=======================================================
Random Seed: 1687229277

Will run 3 of 3 specs
Handling connection for 9102
•Handling connection for 9102
•Handling connection for 9102
•

Ran 3 of 3 Specs in 0.006 seconds
SUCCESS! -- 3 Passed | 0 Failed | 0 Pending | 0 Skipped
PASS

Ginkgo ran 1 suite in 401.680312ms
Test Suite Passed
jie@jie-nuc:~/dev/kepler/e2e$ ginkgo -v
Running Suite: E2eTest Suite - /home/jie/dev/kepler/e2e
=======================================================
Random Seed: 1687229283

Will run 3 of 3 specs
------------------------------
[BeforeSuite]
/home/jie/dev/kepler/e2e/e2e_suite_test.go:42
------------------------------
[BeforeSuite] PASSED [0.000 seconds]
[BeforeSuite]
/home/jie/dev/kepler/e2e/e2e_suite_test.go:42
------------------------------
request check should pass Test with endpoints with requests
  default endpoint
  /home/jie/dev/kepler/e2e/restapicheck_test.go:38
Handling connection for 9102
•
------------------------------
request check should pass Test with endpoints with requests
  default healthz
  /home/jie/dev/kepler/e2e/restapicheck_test.go:39
Handling connection for 9102
•
------------------------------
request check should pass Test with endpoints with requests
  default metrics
  /home/jie/dev/kepler/e2e/restapicheck_test.go:40
Handling connection for 9102
•
------------------------------
[AfterSuite]
/home/jie/dev/kepler/e2e/e2e_suite_test.go:61
------------------------------
[AfterSuite] PASSED [0.000 seconds]
[AfterSuite]
/home/jie/dev/kepler/e2e/e2e_suite_test.go:61
------------------------------

Ran 3 of 3 Specs in 0.005 seconds
SUCCESS! -- 3 Passed | 0 Failed | 0 Pending | 0 Skipped
PASS

Ginkgo ran 1 suite in 398.12812ms
Test Suite Passed

We could see that the 3 Specs all belong to restapicheck_test.go.

**Expected behavior**
The cases in e2e_metric_test.go were bypassed, this should not be expected.

**Additional context**
The RCA is identified as the go:build annotation [there](https://github.com/sustainable-computing-io/kepler/blob/main/e2e/e2e_metric_test.go#L1-L2).
Will raise a fix PR soon, please help review.

jichenjc, 2023-06-30:
looks like this is fixed , should we close this ？

jiere, 2023-06-30:
yes, let me close then.
BTW, all test cases skipped due to Github-hosted runners are VMs, no metrics available from kepler-exporter... this is known issue.

closed 2023-06-30 by jiere