sustainable-computing-io/kepler #740: Unexpected console error/warning in Kepler integration test.

jiere, 2023-06-20:
**Describe the bug**
Several potential issues found in integration test. See unexpected errors or warnings on console:
1. See error msg in "make cluster-sync" console logs:
...
GO111MODULE=off go get -d github.com/go-bindata/go-bindata/...
./hack/bindata.sh
./hack/bindata.sh: line 5: /home/jie/go/bin/go-bindata: No such file or directory
...

2. Occasionally see console warning msg after Kepler deployed:
=> There is 1 zombie process.
$ ps -ef |grep defunct
root     1652813 1652790  0 07:58 ?        00:00:00 [cpuid] <defunct>

$ ps -ef |grep defunct 1652790
root     1652790 1549838  0 07:58 ?        00:00:00 /usr/bin/kepler -v=1
root     1652813 1652790  0 07:58 ?        00:00:00 [cpuid] <defunct>

3. See warning msg in "make cluster-clean" console logs:
hack/common.sh: line 63: [: =: unary operator expected

4. "kubectl port-forward" process occasionally silently quit when running integration test for kepler metrics check.

**To Reproduce**
Steps to reproduce the behavior:
For issue1 above, no special step, just run "make cluster-sync".
For issue2 above, after run "make cluster-sync", when kepler is successfully deployed in cluster, could occasionally see the issue.
For issue3 above, just run "make cluster-clean" to undeploy kepler.
For issue4 above, it is occasionally hit, when hit the issue, integration test case failed and check "ps -ef |grep kubectl" the background process of "kubectl port-forward --address localhost $(kubectl -n kepler get pods -oname) 9102:9102 -n kepler -v7 &" has quitted silently.

**Additional context**
For issue1 above, it seems that using "go get -u" instead of "go get -d" could resolve the issue, my question is: do we need to fix this currently? Is the hack/bindata.sh usage obsoleted or not?

For issue2&3 above, has RCAed, will raise PR soon.

For issue4 above, when hit the issue, checked the syslog, it seems OOM related. I would suggest to use svc "nodePort" type patch instead of current port-forwarding mechanism to expose metrics for integration test, after test, could patch back to "ClusterIP" type, if we do not want to expose kepler-exporter svc port.

Welcome community's comments.

rootfs, 2023-06-20:
for 1), `go get -d` is no longer a documented option, `-u` is valid. The change in #741 makes sense. For 4), I think it `kubectl port-forward` should be watched and warned if it ends unexpectedly.

marceloamaral, 2023-06-20:
Closing as #741 was merged.

closed 2023-06-20 by marceloamaral

jiere, 2023-06-29:
>  For 4), I think it `kubectl port-forward` should be watched and warned if it ends unexpectedly.
Sorry for starting this topic again. @rootfs , @SamYuan1990 , how could we handle such "kubectl port-forward" command unexpected failure issue, especially in CI actions? See my latest fork repo's failure job [here](https://github.com/jiere/kepler/actions/runs/5407725623/jobs/9830770272) 
I also googled such [issue](https://github.com/kubernetes/kubectl/issues/1169), seemed to be easily hit and no clear solutions...
I know there might be security concern for expose service port through "NodePort" type instead of "ClusterIP" type, but how about to apply it in CI workflows and restore to "ClusterIP" type service after the job?

marceloamaral, 2023-06-29:
@jiere the PR #754 should fix the port-forward problem.

SamYuan1990, 2023-06-29:
@marceloamaral I discussed with @jiere offline. the issue is happened on his self hosted github action agent.
https://github.com/jiere/kepler/actions/runs/5407725623/jobs/9830770272#step:1:5
@jiere , please do not... or at least highlight the issue is happened on your self host github action agent next time.

As self host github agent has different settings as network firewall rules with github action agent pools.
Hence, by default, I would like to summary suggest as
1. Some suggestions from @jiere is good founding. 
2. Some are limited with github action agent as ubuntu.
3. Some are specific to the self host instance. In this case, as we can't touch the instance, we can help but... seems just suggestions?.
4. For specific settings we can change at CI-path.yaml for CI usage.

As long as our pipeline still running on github action, we should keep CI at 1st.

jiere, 2023-06-30:
Thanks @marceloamaral and @SamYuan1990 's comments above. Yes, the current issue I meet is happening on my self-hosted runner since I am debugging platform-validation cases.

@marceloamaral , after applied your fix onto my test workflow [file](https://github.com/jiere/kepler/blob/platform-validation/.github/workflows/platform_validation.yml), and rerun my job, the job still failed, see logs [here](https://github.com/jiere/kepler/actions/runs/5418947667/jobs/9851558553), so it might be my runner's local issue, maybe firewall, maybe others...

Anyway, thanks for all the discussion threads, and let me do further debug on my side then, I will also try other ways to let it go.

BTW, another interesting finding during my debugging, I added one line [here](https://github.com/jiere/kepler/blob/platform-validation/.github/workflows/platform_validation.yml#L41) and set the runner to be `ubuntu-20.04` just as community CI does currently. @SamYuan1990 , I found that the Github hosted runners by default installed `kind` latest version(0.20.0 currently, see [log](https://github.com/jiere/kepler/actions/runs/5418849757/jobs/9851361336#step:7:17)), this explains why community's CI jobs never fail [here](https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/integration_test.yml#L42). 
So in the near future, we can cleanup those kind installation jobs in our current workflows. There is no need to so such [definition](https://github.com/sustainable-computing-io/local-dev-cluster/blob/main/kind/kind.sh#L53) then :-D