sustainable-computing-io/kepler #758: Kepler on kind not showing metrics

gpgn, 2023-07-02:
### What happened?

Hi, Im trying to deploy Kepler on a local (bare metal) `kind` cluster, but although the Pod is Running, I do not see any metrics in Prometheus/Grafana. Pod logs:

```sh
I0702 12:26:50.194944       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0702 12:26:50.198016       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0702 12:26:50.203160       1 exporter.go:151] Kepler running on version: 364c44f
I0702 12:26:50.203174       1 config.go:212] using gCgroup ID in the BPF program: true
I0702 12:26:50.203192       1 config.go:214] kernel version: 6.2
I0702 12:26:50.203226       1 config.go:174] kernel source dir is set to /usr/share/kepler/kernel_sources
I0702 12:26:50.203261       1 exporter.go:171] EnabledBPFBatchDelete: true
I0702 12:26:50.203307       1 power.go:53] use sysfs to obtain power
I0702 12:26:50.443161       1 exporter.go:184] Initializing the GPU collector
I0702 12:26:50.443402       1 watcher.go:67] Using in cluster k8s config
cannot attach kprobe, probe entry may not exist
I0702 12:26:51.830614       1 bcc_attacher.go:171] Successfully load eBPF module with option: [-DMAP_SIZE=10240 -DNUM_CPUS=12 -DSET_GROUP_ID]
I0702 12:26:51.839296       1 exporter.go:228] Started Kepler in 1.636145659s
```

My steps to deploy:

```sh
# create cluster
kind create cluster --config=local-cluster-config.yaml`
# build and deploy kepler
git clone --depth 1 git@github.com:sustainable-computing-io/kepler.git ./infra/kepler
cd ./infra/kepler
make build-manifest OPTS="CI_DEPLOY"
kubectl apply -f _output/generated-manifest/deployment.yaml
```

`local-cluster-config.yaml`:
```yaml
kind: Cluster
apiVersion: kind.x-k8s.io/v1alpha4
name: cluster-sustainable-computing-sandbox
networking:
  ipFamily: dual
  kubeProxyMode: "ipvs"
nodes:
- role: control-plane
  image: kindest/node:v1.27.3@sha256:3966ac761ae0136263ffdb6cfd4db23ef8a83cba8a463690e98317add2c9ba72
  kubeadmConfigPatches:
  - |
    kind: ClusterConfiguration
    apiServer:
        extraArgs:
          enable-admission-plugins: NodeRestriction,MutatingAdmissionWebhook,ValidatingAdmissionWebhook 
  extraPortMappings:
  - containerPort: 80
    hostPort: 80
  extraMounts:
  - hostPath: /proc
    containerPath: /proc-host
    readOnly: true
    selinuxRelabel: false
  - hostPath: /usr/src
    containerPath: /usr/src
    readOnly: true
    selinuxRelabel: false
- role: worker
  image: kindest/node:v1.27.3@sha256:3966ac761ae0136263ffdb6cfd4db23ef8a83cba8a463690e98317add2c9ba72
  extraMounts:
  - hostPath: /proc
    containerPath: /proc-host
    readOnly: true
    selinuxRelabel: false
  - hostPath: /usr/src
    containerPath: /usr/src
[... 255 lines cut ...]

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-07-02:
would you please increase the kepler pod verbosity?
```bash
kubectl edit -n kepler daemonset kepler-exporter
```

and change `/usr/bin/kepler -v=1` to `/usr/bin/kepler -v=5`. Then please get and share the kepler logs:
```
kubectl logs -n kepler daemonset/kepler-exporter
```

gpgn, 2023-07-02:
Sure thing:
```
I0702 14:02:37.139708       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0702 14:02:37.144561       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0702 14:02:37.154175       1 exporter.go:151] Kepler running on version: 364c44f
I0702 14:02:37.154197       1 config.go:212] using gCgroup ID in the BPF program: true
I0702 14:02:37.154226       1 config.go:214] kernel version: 6.2
I0702 14:02:37.154270       1 config.go:174] kernel source dir is set to /usr/share/kepler/kernel_sources
I0702 14:02:37.154325       1 exporter.go:171] EnabledBPFBatchDelete: true
I0702 14:02:37.154333       1 config.go:125] ENABLE_EBPF_CGROUPID: true
I0702 14:02:37.154337       1 config.go:126] ENABLE_GPU: true
I0702 14:02:37.154340       1 config.go:127] ENABLE_PROCESS_METRICS: false
I0702 14:02:37.154380       1 config.go:128] EXPOSE_HW_COUNTER_METRICS: true
I0702 14:02:37.154388       1 config.go:129] EXPOSE_CGROUP_METRICS: true
I0702 14:02:37.154392       1 config.go:130] EXPOSE_KUBELET_METRICS: true
I0702 14:02:37.154395       1 config.go:131] EXPOSE_IRQ_COUNTER_METRICS: true
I0702 14:02:37.154484       1 power.go:53] use sysfs to obtain power
I0702 14:02:37.154492       1 bcc_attacher.go:183] hardeware counter metrics config true
I0702 14:02:37.154497       1 bcc_attacher.go:201] irq counter metrics config true
I0702 14:02:44.860314       1 utils.go:56] Available ebpf metrics: [cpu_time irq_net_tx irq_net_rx irq_block]
I0702 14:02:44.860341       1 utils.go:57] Available counter metrics: [cpu_cycles cpu_ref_cycles cpu_instr cache_miss]
I0702 14:02:44.860346       1 utils.go:58] Available cgroup metrics from cgroup: [cgroupfs_memory_usage_bytes cgroupfs_kernel_memory_usage_bytes cgroupfs_tcp_memory_usage_bytes cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us cgroupfs_ioread_bytes cgroupfs_iowrite_bytes block_devices_used]
I0702 14:02:44.860352       1 utils.go:59] Available cgroup metrics from kubelet: [container_cpu_usage_seconds_total container_memory_working_set_bytes]
I0702 14:02:44.860402       1 model.go:87] Model Config NODE_TOTAL: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}
I0702 14:02:44.860586       1 lr.go:163] LR Model (AbsModelWeight): loadWeightFromURLorLocal(): {{0 map[] map[]}}
I0702 14:02:44.860601       1 model.go:79] Model AbsModelWeight initiated (true)
I0702 14:02:44.860606       1 model.go:87] Model Config NODE_COMPONENTS: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}
I0702 14:02:44.860696       1 lr.go:163] LR Model (AbsComponentModelWeight): loadWeightFromURLorLocal(/var/lib/kepler/data/KerasCompWeightFullPipeline.json): map[dram:{{0.8339537395297559 map[] map[cgroupfs_cpu_usage_us:{6.557768917682226e+06 1.0111459601975505e+14 0.30000764700549093} cgroupfs_memory_usage_bytes:{1.1568395713957231e+07 1.4831755031743084e+14 0.010739137690300415} cgroupfs_system_cpu_usage_us:{320904.6193377788 3.07640381448747e+10 0.2612100015047149} cgroupfs_user_cpu_usage_us:{6.236864298459416e+06 1.0251834339511184e+14 0}]}} pkg:{{24.603798628318298 map[] map[cgroupfs_cpu_usage_us:{6.557768917682226e+06 1.0111459601975505e+14 0} cgroupfs_memory_usage_bytes:{1.1568395713957231e+07 1.4831755031743084e+14 0} cgroupfs_system_cpu_usage_us:{320904.6193377788 3.07640381448747e+10 0} cgroupfs_user_cpu_usage_us:{6.236864298459416e+06 1.0251834339511184e+14 24.50009569917875}]}}]
I0702 14:02:44.860725       1 model.go:79] Model AbsComponentModelWeight initiated (true)
I0702 14:02:44.860731       1 model.go:87] Model Config CONTAINER_TOTAL: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}
I0702 14:02:44.860773       1 lr.go:163] LR Model (DynModelWeight): loadWeightFromURLorLocal(): {{0 map[] map[]}}
I0702 14:02:44.860783       1 model.go:79] Model DynModelWeight initiated (true)
I0702 14:02:44.860790       1 model.go:87] Model Config CONTAINER_COMPONENTS: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json}
I0702 14:03:14.884824       1 lr.go:163] LR Model (DynComponentModelWeight): loadWeightFromURLorLocal(https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json): map[dram:{{0.8339537395297559 map[] map[cgroupfs_cpu_usage_us:{6.557768917682226e+06 1.0111459601975505e+14 0.30000764700549093} cgroupfs_memory_usage_bytes:{1.1568395713957231e+07 1.4831755031743084e+14 0.010739137690300415} cgroupfs_system_cpu_usage_us:{320904.6193377788 3.07640381448747e+10 0.2612100015047149} cgroupfs_user_cpu_usage_us:{6.236864298459416e+06 1.0251834339511184e+14 0}]}} pkg:{{24.603798628318298 map[] map[cgroupfs_cpu_usage_us:{6.557768917682226e+06 1.0111459601975505e+14 0} cgroupfs_memory_usage_bytes:{1.1568395713957231e+07 1.4831755031743084e+14 0} cgroupfs_system_cpu_usage_us:{320904.6193377788 3.07640381448747e+10 0} cgroupfs_user_cpu_usage_us:{6.236864298459416e+06 1.0251834339511184e+14 24.50009569917875}]}}]
I0702 14:03:14.884857       1 model.go:79] Model DynComponentModelWeight initiated (true)
I0702 14:03:14.884863       1 model.go:87] Model Config PROCESS_TOTAL: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}
I0702 14:03:14.884887       1 lr.go:163] LR Model (DynModelWeight): loadWeightFromURLorLocal(): {{0 map[] map[]}}
I0702 14:03:14.884894       1 model.go:79] Model DynModelWeight initiated (true)
I0702 14:03:14.884898       1 model.go:87] Model Config PROCESS_COMPONENTS: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}
```

rootfs, 2023-07-03:
@gpgn the logs look right. I missed your deployment script, it looks you missed `PROMETHEUS_DEPLOY` option in the command `./hack/build-manifest.sh "CI_DEPLOY"`. The `PROMETHEUS_DEPLOY` option will install ServiceMonitor for Prometheus to scrape the kepler endpoint.

If you want to see kepler metrics via `curl`, you can run the following command to check the `kepler_container_joules_total` metric:
```bash
kubectl exec -ti -n kepler daemonset/kepler-exporter -- bash -c "curl localhost:9102/metrics > /tmp/k.log; grep kepler_container_joules /tmp/k.log |sort -k 2 -g"
```

gpgn, 2023-07-03:
Wonderful, thanks a lot!

[image]

closed 2023-07-03 by gpgn

rootfs, 2023-07-03:
cool! @gpgn do you mind writing an instruction or walkthrough for https://github.com/sustainable-computing-io/kepler-doc?

gpgn, 2023-07-04:
Yeah of course! :+1: