sustainable-computing-io/kepler #765: Add node identification for edge computing deployments

aspanner, 2023-07-05:
### What would you like to be added?

Kepler needs a means to uniquely identify power consumption per compute nodes in environments of multiple compute nodes.

### Why is this needed?

An edge scenario has potentially 100s or even thousands of devices of various form factors installed.

A central dashboard or carbon footprint calculation entity needs to be able to correlate power consumption to specific devices. It is hence necessary to provide unique identifies across an edge-device fleet deployment.

For example: To effectively calculate carbon footprint per part produced across a manufacturing plant shop floor it is necessary to do accrual accounting of energy consumption across multiple shopfloor workstations.

rootfs, 2023-07-05:
this could be a good use case for standalone kepler on edge devices. In standalone mode, kepler metrics is not always scraped by prometheus and node info could be missing. A quick fix is to insert node info into kepler metrics.

aspanner, 2023-07-09:
Agree. Where is the best place to insert node info into kepler merics? /cmd/exporter.go?
Option 1:
We randomly generate a node GUID unless there is a config file under /etc/kepler/kepler.config/NODE_ID specifiying a specific node id?
Or perhaps better a CPU/GPU/xPU id instead?

aspanner, 2023-07-28:
@sallyom @husky-parul - keen to get your thoughts.

closed 2023-08-10 by aspanner