sustainable-computing-io/kepler #766: PID 0 does not have cgroup metrics since there is no /proc/0/cgroup

jichenjc, 2023-07-06:
### What happened?

https://github.com/sustainable-computing-io/kepler/pull/715#issuecomment-1623250719

PID 0 does not have cgroup metrics since there is no /proc/0/cgroup 

reported for many pods , seems it's incorrect 

### What did you expect to happen?

no issue above

### How can we reproduce it (as minimally and precisely as possible)?

every deployment

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jichenjc, 2023-07-06:
```
--- a/pkg/collector/container_cgroup_collector.go
+++ b/pkg/collector/container_cgroup_collector.go
@@ -29,7 +29,7 @@ import (
 func (c *Collector) updateCgroupMetrics() {
        for key := range c.ContainersMetrics {
                if c.ContainersMetrics[key].PID == 0 {
-                       klog.V(3).Infof("PID 0 does not have cgroup metrics since there is no /proc/0/cgroup")
+                       klog.V(3).Infof("PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is %s", key)
                        continue
                }
                if c.ContainersMetrics[key].CgroupStatHandler == nil {
diff --git a/pkg/collector/container_hc_collector.go b/pkg/collector/container_hc_collector.go
index 2ed6241..a9e0db3 100644
--- a/pkg/collector/container_hc_collector.go
+++ b/pkg/collector/container_hc_collector.go
@@ -175,6 +175,10 @@ func (c *Collector) updateBPFMetrics() {

                c.createContainersMetricsIfNotExist(containerID, ct.CGroupID, ct.PID, config.EnabledEBPFCgroupID)
                c.ContainersMetrics[containerID].PID = ct.PID
+               klog.V(5).Infof("==========setting containerID %s", containerID)
+               if ct.PID == 0 {
+                       klog.V(5).Infof("-----------get %s PID = 0", containerID)
+               }

```

such code change and wait 10+ min
```
I0706 09:33:20.391662       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 22847c7d65d9e8ea22c3a9ccc2a34c682c50f1e45d0e73fd00a70b42c2247aba
I0706 09:33:20.394510       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 87354914324f51d686261f7b79ec7b93ef700d96c3d734b4c39ef746c9f78a60
I0706 09:33:20.397109       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 609a7dc9681b53fbd07cf9a799ff27cae79226b2e463c62022497c190d221cec
I0706 09:33:20.401783       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 65e7194ea20334fd86fd8eda388d6ce29e1f94430bab94ab5bb592dba92015a9
I0706 09:33:20.402647       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is cd50992e96dd988c2e3a19ffd03ee58368a34d25facd43e2f5c11965b906000e

I0706 09:33:20.415633       1 metric_collector.go:132] energy from pod/container (0 active processes): name: nginx-deployment-7fb96c846b-9cxjg/nginx namespace: default
        cgrouppid: 0 pid: [] comm:  containerid:cd50992e96dd988c2e3a19ffd03ee58368a34d25facd43e2f5c11965b906000e
        Dyn ePkg (mJ): 9512 (1540944) (eCore: 9512 (1540944) eDram: 150 (24300) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        Idle ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        CPUTime:  0 (0)

```

but I never saw `klog.V(5).Infof("==========setting containerID %s", containerID) ` called with 
cd50992e96dd988c2e3a19ffd03ee58368a34d25facd43e2f5c11965b906000e

jichenjc, 2023-07-07:
@rootfs @marceloamaral need your comments.. seems our eBPF collector supposed to collect data at 
pkg/collector/container_hc_collector.go， but some containers is NOT detected at all through eBPF process
per above comment, is it because of no CPU consumption e.g no cpu time ? 

```
I0707 07:04:48.875503       1 metric_collector.go:132] energy from pod/container (0 active processes): name: busybox/busybox namespace: default
        cgrouppid: 0 pid: [] comm:  containerid:532d202f8d1fa755babb02109593fd3d99e69f60e98f3df543643731e5594380
        Dyn ePkg (mJ): 9512 (351944) (eCore: 9512 (351944) eDram: 150 (5550) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 18446744073709541954 (18446744073709541954)
        Idle ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        CPUTime:  0 (0)
        NetTX IRQ: 0 (0)
        NetRX IRQ: 0 (0)
        Block IRQ: 0 (0)
        counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instr:0 (0) cpu_ref_cycles:0 (0)]
        cgroupfs: map[block_devices_used:0 (0) cgroupfs_cpu_usage_us:0 (0) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_kernel_memory_usage_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (0) cgroupfs_system_cpu_usage_us:0 (0) cgroupfs_tcp_memory_usage_bytes:0 (0) cgroupfs_user_cpu_usage_us:0 (0)]
        kubelets: map[container_cpu_usage_seconds_total:0 (0) container_memory_working_set_bytes:0 (196608)]

```

rootfs, 2023-07-07:
@jichenjc if the PID is seen by the collector, there should be CPUTime there. It looks a bug to me the CPUTime is zero

jichenjc, 2023-07-10:
>>>if the PID is seen by the collector, there should be CPUTime there. It looks a bug to me the CPUTime is zero

then this might be our eBPF module issue? 

first, does our eBPF report all container even though there is no cputime at all? if not, then our current logic will not report cgroup data and this makes the memory/cpu allocation info not showing ..

second, as you pointed above, it's a bug in our (kepler) ebpf related code?

rootfs, 2023-07-10:
the ebpf should collect the cpu time for processes that are currently running. The cgroup cpu metrics is read from cgroup fs. Ideally, they should match.

rootfs, 2023-07-10:
@jichenjc please see if #782 fixes the zero `CPUTime` issue

closed 2023-09-15 by stale[bot]