sustainable-computing-io/kepler #769: ebpfBatchGet configuration cause issue

jichenjc, 2023-07-07:
### What happened?

in order to work on https://github.com/sustainable-computing-io/kepler/issues/766
I did following change (to current master)

```
--- a/pkg/collector/container_hc_collector.go
+++ b/pkg/collector/container_hc_collector.go
@@ -36,7 +36,7 @@ import "C"

 var (
        // ebpfBatchGet is true if the kernel supports batch get operation
-       ebpfBatchGet = true
+       ebpfBatchGet = false
        // ebpfBatchGetAndDelete is true if delete all the keys after batch get
        ebpfBatchGetAndDelete = ebpfBatchGet

```

and seems cgroup can't find any data and seeing many error likes following means cgroup seems doesn't work at all

```
I0707 06:49:22.458762       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 53d6512b7b08ae1358ed9b751955bd1635ca4458a2f84c9e90db9ee16e909cf3
I0707 06:49:22.458764       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 14d094eb869fc083503fd195f9604e0963944c8fac1d379cb2d8ec2a33007923
I0707 06:49:22.458767       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 749284ed26a04119bb5f3cf59b0400a007f74c309bacb0fff3f39821a0d10b0f
I0707 06:49:22.458770       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 9c40fcfeb228cb5a30f8ee47114ee1357a6126e094eddf8a45161f8d7d4518b1
I0707 06:49:22.458772       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 6862c0d0005b4ebeb8c82343128d3e4ef43be8867862dc5ee9eb4db9de1d6ff7
I0707 06:49:22.458777       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 74991481dd1c96965750bffd7c4aa1a5d847bf0ffdf0c287b6f4580ebce2c10f
I0707 06:49:22.458780       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is d34a1d0e4f0c2b75808a603f0a13d97cb5bd292a978d15418aca2c03f1741d3b
I0707 06:49:22.458783       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 2aec0f8b0b693dbddae515e393ea7e1a955c1edf5aa512d3ae36e39d5cb0bc9d
I0707 06:49:22.458785       1 container_cgroup_collector.go:32] PID 0 does not have cgroup metrics since there is no /proc/0/cgroup, key is 9137b31323f6d12b26e62904b0b840e6ae3e50bbee0fc7edd8b54c1a8adb8f95

I0707 06:49:22.462330       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462339       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462351       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462374       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462397       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462415       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462425       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462439       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462447       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462456       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462464       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462474       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462490       1 container_power.go:126] the aggregated value has overflowed

```

so @rootfs should we consider remove this option ?

### What did you expect to happen?

nothing wrong report

### How can we reproduce it (as minimally and precisely as possible)?

per above

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jichenjc, 2023-07-07:
um.. ignore this ,this is also reported when the batch operation set to True
```
I0707 06:49:22.462330       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462339       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462351       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462374       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462397       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462415       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462425       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462439       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462447       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462456       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462464       1 container_power.go:126] the aggregated value has overflowed
I0707 06:49:22.462474       1 container_power.go:126] the aggregated value has overflowed
```

rootfs, 2023-07-07:
@sunya-ch can you take a look? thanks

sunya-ch, 2023-07-10:
I think we need to handle both error but maybe separately. 

For, PID=0
We have to confirm the following items.
- [ ] whether the BPF collects any process entry
- [ ] whether the process entry PID is in the same pod PID namespace

For aggregated value has overflowed, it happens when the stat is over MaxUint64 but it the value should be reset to zero and continue. It looks unexpected to me that the log produces that much warning. 

https://github.com/sustainable-computing-io/kepler/blob/d053045c21bc4e9596f0789747e6f988ce09bc04/pkg/collector/metric/types/types.go#L44

jichenjc, 2023-07-10:
ok, let me check how I can handle the 2nd first, then continue to check #1 issue with #766 as well

jichenjc, 2023-07-10:
2nd is caused by the component power is enabled in my env (a RHEL VM) 

```
I0710 09:04:19.232616       1 model.go:87] Model Config CONTAINER_COMPONENTS: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json}
I0710 09:04:19.538137       1 lr.go:163] LR Model (DynComponentModelWeight): loadWeightFromURLorLocal(https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json): map[dram:{{0.8339537395297559 map[] map[cgroupfs_cpu_usage_us:{6.557768917682226e+06 1.0111459601975505e+14 0.30000764700549093} cgroupfs_memory_usage_bytes:{1.1568395713957231e+07 1.4831755031743084e+14 0.010739137690300415} cgroupfs_system_cpu_usage_us:{320904.6193377788 3.07640381448747e+10 0.2612100015047149} cgroupfs_user_cpu_usage_us:{6.236864298459416e+06 1.0251834339511184e+14 0}]}} pkg:{{24.603798628318298 map[] map[cgroupfs_cpu_usage_us:{6.557768917682226e+06 1.0111459601975505e+14 0} cgroupfs_memory_usage_bytes:{1.1568395713957231e+07 1.4831755031743084e+14 0} cgroupfs_system_cpu_usage_us:{320904.6193377788 3.07640381448747e+10 0} cgroupfs_user_cpu_usage_us:{6.236864298459416e+06 1.0251834339511184e+14 24.50009569917875}]}}]
I0710 09:04:19.538234       1 model.go:79] Model DynComponentModelWeight initiated (true)

```

then the component power is 0 but  componentPower.Pkg, componentPower.DRAM is NOT 0
```
I0710 09:04:57.864140       1 container_power.go:107] ========================1====== 0 9512 150
```

marceloamaral, 2023-07-11:
@jichenjc can you check the values when/before it overflow?
Aggregated value and update value?
Maybe the estimator is giving a very big value?

jichenjc, 2023-07-12:
@marceloamaral the reason is likely to be https://github.com/sustainable-computing-io/kepler/pull/779

closed 2023-07-27 by jichenjc