sustainable-computing-io/kepler #771: Security implications of enabling kernel-devel and pods mounting /sys/kernel/debug

BGrasnick, 2023-07-07:
We want to install kepler (https://github.com/sustainable-computing-io/kepler/) in our OpenShift 4.10 cluster.

As a prerequisite for working on OpenShift, we need to enable kernel-devel on master and worker nodes:
https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/cluster-prereqs/51-master-kernel-devel.yaml
https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/cluster-prereqs/51-worker-kernel-devel.yaml

Also, we need to mount the debugFS (/sys/kernel/debug) into the kepler Pods:
https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/exporter/patch/patch-openshift.yaml

For usage on OpenShift, kepler also requires a very powerful SecurityContextConstraints giving itself extensive privileges:
https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/exporter/openshift_scc.yaml

After seeing all this, we grew a little concerned and were wondering if enabling all this could pose a security risk or impact cluster performance.

Could you provide some insight into these aspects and if it would be secure for us to make these changes and use kepler?

rootfs, 2023-07-07:
Thank you @BGrasnick auditing all these issues. We are taking steps to remove these dependency:
- kernel-devel requirement. This requirement comes from bcc to dynamically compile ebpf module. After installing kernel-devel pkgs in Kepler container images, we no longer need it (but the manifest has not updated yet).
- `/sys/kernel/debug`. This is also from bcc library to dynamically resolve kernel data structure. @sunya-ch is adding libbpf option so that no more runtime data structure is needed in this [PR](https://github.com/sustainable-computing-io/kepler/pull/752)
- the powerful SCC indeed needs revisit. I think some of them can be removed (need to test though):
```
allowHostDirVolumePlugin: true
allowHostNetwork: true
allowHostPorts: true
allowHostIPC: true
```

rootfs, 2023-08-02:
cc @vimalk78

PR #926 (2023-11-01): Feat: 771 remove unnecessary security concerns
fixes #771 

- feat: reduce privileges granted through SCC for OpenShift
- feat: set mounted volumes as read only as much as possible
- feat: set pod hostPID to true to avoid collisions/unwanted behavior in host PID namespace
- feat: remove unnecessary mount of /sys/kernel/debug

Same improvements were already merged in the kepler-operator: https://github.com/sustainable-computing-io/kepler-operator/pull/185
And the removal of the /sys/kernel/debug was discussed here: https://github.com/sustainable-computing-io/kepler-model-server/pull/158#discuss [... cut]
closed 2023-10-08 by stale[bot]