sustainable-computing-io/kepler #773: coverage badge gets updated on each unit test

rootfs, 2023-07-07:
### What happened?

coverage badge should only update per PR, but it currently updates every time the unit test is started

https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/unit_test.yml#L43

### What did you expect to happen?

Update coverage per PR, instead of per unit test

### How can we reproduce it (as minimally and precisely as possible)?

the commit history has too many bot update

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

SamYuan1990, 2023-07-09:
@rootfs , but https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/unit_test.yml#L55 it seems even if it's updated, but nothing committed to badge right?

rootfs, 2023-07-10:
@SamYuan1990 see the commit history, there are many bot commits along with a PR

SamYuan1990, 2023-07-10:
> @SamYuan1990 see the commit history, there are many bot commits along with a PR

https://github.com/sustainable-computing-io/kepler/actions/runs/5453435884/jobs/9922274580?pr=752

I randomly pick up one action for unit test, the badge is not been committed.

SamYuan1990, 2023-07-10:
but it's interesting that the bot trigger another action. https://github.com/sustainable-computing-io/kepler/actions/runs/5508865205

PR #780 (2023-07-10): extend ignore path
I hope by this PR fix #773 
as try to add readme file in ignore path to avoid bot update trigger a pipeline job.
closed 2023-07-10 by rootfs