sustainable-computing-io/kepler #777: Kepler installation error in logs and warning while using the command apply

fatma-laribi, 2023-07-10:
Hi,
- After using kubectl apply -f _output/generated-manifest/deployment.yaml in a Kubernetes cluster, I have encountered a warning in the output:
namespace/kepler created
serviceaccount/kepler-sa created
[clusterrole.rbac.authorization.k8s.io/kepler-clusterrole](http://clusterrole.rbac.authorization.k8s.io/kepler-clusterrole) created
[clusterrolebinding.rbac.authorization.k8s.io/kepler-clusterrole-binding](http://clusterrolebinding.rbac.authorization.k8s.io/kepler-clusterrole-binding) created
configmap/kepler-cfm created
service/kepler-exporter created
Warning: would violate PodSecurity "restricted:latest": privileged (container "kepler-exporter" must not set securityContext.privileged=true), allowPrivilegeEscalation != false (container "kepler-exporter" must set securityContext.allowPrivilegeEscalation=false), unrestricted capabilities (container "kepler-exporter" must set securityContext.capabilities.drop=["ALL"]), restricted volume types (volumes "lib-modules", "tracing", "proc" use restricted volume type "hostPath"), runAsNonRoot != true (pod or container "kepler-exporter" must set securityContext.runAsNonRoot=true), seccompProfile (pod or container "kepler-exporter" must set securityContext.seccompProfile.type to "RuntimeDefault" or "Localhost")
daemonset.apps/kepler-exporter created

- After executing kubectl logs daemonset.apps/kepler-exporter here are the errors:
Found 9 pods, using pod/kepler-exporter-45rr5

I0629 15:17:59.325437       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory

I0629 15:17:59.330266       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?

I0629 15:17:59.339086       1 exporter.go:151] Kepler running on version: 364c44f

I0629 15:17:59.339104       1 config.go:212] using gCgroup ID in the BPF program: true

I0629 15:17:59.339139       1 config.go:214] kernel version: 5.15

I0629 15:17:59.339178       1 config.go:174] kernel source dir is set to /usr/share/kepler/kernel_sources

I0629 15:17:59.339224       1 exporter.go:171] EnabledBPFBatchDelete: true

I0629 15:17:59.339346       1 rapl_msr_util.go:136] input/output error

I0629 15:17:59.339373       1 power.go:64] Not able to obtain power, use estimate method

I0629 15:17:59.395021       1 exporter.go:184] Initializing the GPU collector

I0629 15:17:59.395480       1 watcher.go:67] Using in cluster k8s config

modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.15.106-talos

chdir(/lib/modules/5.15.106-talos/build): No such file or directory

I0629 15:17:59.502326       1 bcc_attacher.go:74] failed to attach the bpf program: <nil>

I0629 15:17:59.502346       1 bcc_attacher.go:143] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=24 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, from default kernel source.

I0629 15:17:59.502368       1 bcc_attacher.go:146] trying to load eBPF module with kernel source dir /usr/share/kepler/kernel_sources/4.18.0-477.13.1.el8_8.x86_64

/virtual/main.c:242:1: warning: declaration of 'struct tracepoint__irq__softirq_entry' will not be visible outside of this function [-Wvisibility]

TRACEPOINT_PROBE(irq, softirq_entry)

^

/virtual/include/bcc/helpers.h:1435:46: note: expanded from macro 'TRACEPOINT_PROBE'

int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)

                                             ^

<scratch space>:40:1: note: expanded from here

tracepoint__irq__softirq_entry

^

/virtual/main.c:249:17: error: incomplete definition of type 'struct tracepoint__irq__softirq_entry'

        if (args->vec < 10) {

            ~~~~^

/virtual/main.c:242:1: note: forward declaration of 'struct tracepoint__irq__softirq_entry'

TRACEPOINT_PROBE(irq, softirq_entry)

^

/virtual/include/bcc/helpers.h:1435:46: note: expanded from macro 'TRACEPOINT_PROBE'

int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)

                                             ^

<scratch space>:40:1: note: expanded from here

tracepoint__irq__softirq_entry

^

/virtual/main.c:250:41: error: incomplete definition of type 'struct tracepoint__irq__softirq_entry'

            process_metrics->vec_nr[args->vec] ++;

                                    ~~~~^

/virtual/main.c:242:1: note: forward declaration of 'struct tracepoint__irq__softirq_entry'

TRACEPOINT_PROBE(irq, softirq_entry)

^

/virtual/include/bcc/helpers.h:1435:46: note: expanded from macro 'TRACEPOINT_PROBE'

int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)

                                             ^

<scratch space>:40:1: note: expanded from here

tracepoint__irq__softirq_entry

^

1 warning and 2 errors generated.

I0629 15:18:00.027188       1 bcc_attacher.go:74] failed to attach the bpf program: <nil>

I0629 15:18:00.027214       1 bcc_attacher.go:150] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=24 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, from kernel source "/usr/share/kepler/kernel_sources/4.18.0-477.13.1.el8_8.x86_64"

I0629 15:18:00.027230       1 bcc_attacher.go:146] trying to load eBPF module with kernel source dir /usr/share/kepler/kernel_sources/5.14.0-284.11.1.el9_2.x86_64

/virtual/main.c:242:1: warning: declaration of 'struct tracepoint__irq__softirq_entry' will not be visible outside of this function [-Wvisibility]

TRACEPOINT_PROBE(irq, softirq_entry)

^

/virtual/include/bcc/helpers.h:1435:46: note: expanded from macro 'TRACEPOINT_PROBE'

int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)

                                             ^

<scratch space>:28:1: note: expanded from here

tracepoint__irq__softirq_entry

^

/virtual/main.c:249:17: error: incomplete definition of type 'struct tracepoint__irq__softirq_entry'

        if (args->vec < 10) {

            ~~~~^

/virtual/main.c:242:1: note: forward declaration of 'struct tracepoint__irq__softirq_entry'

TRACEPOINT_PROBE(irq, softirq_entry)

^

/virtual/include/bcc/helpers.h:1435:46: note: expanded from macro 'TRACEPOINT_PROBE'

int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)

                                             ^

<scratch space>:28:1: note: expanded from here

tracepoint__irq__softirq_entry

^

/virtual/main.c:250:41: error: incomplete definition of type 'struct tracepoint__irq__softirq_entry'

            process_metrics->vec_nr[args->vec] ++;

                                    ~~~~^

/virtual/main.c:242:1: note: forward declaration of 'struct tracepoint__irq__softirq_entry'

TRACEPOINT_PROBE(irq, softirq_entry)

^

/virtual/include/bcc/helpers.h:1435:46: note: expanded from macro 'TRACEPOINT_PROBE'

int tracepoint__##category##__##event(struct tracepoint__##category##__##event *args)

                                             ^

<scratch space>:28:1: note: expanded from here

tracepoint__irq__softirq_entry

^

1 warning and 2 errors generated.

I0629 15:18:00.495167       1 bcc_attacher.go:74] failed to attach the bpf program: <nil>

I0629 15:18:00.495191       1 bcc_attacher.go:150] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=24 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, from kernel source "/usr/share/kepler/kernel_sources/5.14.0-284.11.1.el9_2.x86_64"

I0629 15:18:00.495210       1 bcc_attacher.go:158] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=24 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, not able to load eBPF modules

I0629 15:18:00.495230       1 exporter.go:201] failed to start : failed to attach bpf assets: failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=24 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, not able to load eBPF modules

I0629 15:18:00.495380       1 exporter.go:228] Started Kepler in 1.156321857s

rootfs, 2023-07-10:
this looks a privilege issue. We can expand the bcc macro in ebpf to avoid accessing `/sys/kernel` to avoid this situation.

rootfs, 2023-07-10:
@sunya-ch can you port [this](https://github.com/sustainable-computing-io/kepler/blob/main/bpfassets/libbpf/src/kepler.bpf.h#L106) libebpf change? Thank you!

jichenjc, 2023-07-17:
@fatma-laribi could you please help try #781 locally? 
we are not sure the issue can be solved per discussion that by the proposed fix with @sunya-ch @rootfs 
so check the proposed fix in reporting env might be a good thing to try

closed 2023-07-20 by rootfs