sustainable-computing-io/kepler #786: need troubleshooting guideline for future support

sunya-ch, 2023-07-11:
### Which document would you like to be mentioned?

Troubleshooting document for user/contribute to check and add support for uncovered environment.
The troubleshooting should provides the checklist with instructions to find a root cause of issue from the log.
This should be also added to GitHub template when creating kind/bug issue.

### Why is this needed?

The troubleshooting guideline could help maintainers/contributors to determine root cause.
In addition, it can further help on updating the code to support uncovered environment. 

For example, 

When cgroups stat is zero, we can follow the below steps after change log level to maximum.

```mermaid
flowchart TD;
   bpfAttached--no-->kernelDependencyCheck
   bpfAttached--yes-->containerIDResolved
   containerIDResolved--no-->pidNamespaceCheck
   containerIDResolved--yes-->cGroupManagerCreated
   cGroupManagerCreated--no-->cGroupPathCheck
   cGroupManagerCreated--yes--> cGroupStatDeletedOnUpdate
   cGroupStatDeletedOnUpdate--yes-->deletedConditionCheck
```

bpfAttached: check attachment log
kernelDependencyCheck: check permission, check kernel version
containerIDResolved: check "cannot resolve..." and "PID 0..."
pidNamespaceCheck: /proc mount if needed
cGroupManagerCreated: check "could not start cgroup stat handler..."
cGroupPathCheck: check /proc/$(pidof kepler)/cgroup
cGroupStatDeletedOnUpdate: check "Error reading cgroup stats for container"

rootfs, 2023-07-11:
@sunya-ch looks great! can you add to the trouble shooting section [here](https://sustainable-computing.io/usage/trouble_shooting/)?

sunya-ch, 2023-07-12:
@rootfs Oh I didn't notice that we already have that section. I will make an update on that and also to the GitHub template to link to that page. Thanks.

closed 2023-09-18 by stale[bot]