sustainable-computing-io/kepler #787: API server pod watcher missing event, causing missed pod metrics

rootfs, 2023-07-12:
### What happened?

The kube apiserver watcher missed pod status change. As a result, pods are not updated and mistakenly classified as system pods

I added some tracing along [this code block](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/kubernetes/watcher.go#L153-L157)

```go
		for _, condition := range pod.Status.Conditions {
			klog.V(5).Infof("pod update %s %s %s %s", pod.Name, pod.Namespace, condition.Type, condition.Status)
```

This is what is in the log:
```console
$ kubectl logs -n kepler kepler-exporter-z6ptd |grep random |grep pod
I0712 21:20:17.027776       1 watcher.go:161] pod update random-576f489b67-p7kvg cpu-load PodScheduled True
I0712 21:20:17.038032       1 watcher.go:161] pod update random-576f489b67-p7kvg cpu-load PodScheduled True
I0712 21:20:17.049026       1 watcher.go:161] pod update random-576f489b67-p7kvg cpu-load Initialized True
I0712 21:20:17.555706       1 watcher.go:161] pod update random-576f489b67-p7kvg cpu-load Initialized True
I0712 21:20:20.881916       1 watcher.go:161] pod update random-576f489b67-p7kvg cpu-load Initialized True
```

### What did you expect to happen?

The `ContainerReady` status is never received and `condition.Status` is always `True`. In this condition, the pod is never added to metrics cache and gets resolved.

### How can we reproduce it (as minimally and precisely as possible)?

Running Kepler on ROSA seems to produce this consistently.

### Kepler image tag

<details>
latest
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>
AWS, ROSA
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-07-12:
cc @AdrianHammond for reporting this issue

rootfs, 2023-07-12:
@marceloamaral

rootfs, 2023-07-12:
actually, this is not the root cause. closing.

marceloamaral, 2023-07-13:
> Pod random-576f489b67-qq4hk cpu-load is ready with 0 container statuses, 0 init container status, 0 ephemeral statues

Is expected this pod does not have any container running? Or all containers failed to be created?
If no container is running, the pod will never have the  `ContainerReady` status

rootfs, 2023-07-13:
the pod is based on [this test](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/tests/cpu/random.yaml), running a CPU stressing test. After #788, we can see the pod in its own metrics (aka not more as `system_process`)

```console
$ oc exec -ti -n kepler kepler-exporter-8lpdb -- bash -c "curl localhost:9102/metrics |grep kepler_container_joules_total|sort -k 2 -g |tail -10"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  440k    0  440k    0     0  47.8M      0 --:--:-- --:--:-- --:--:-- 47.8M
kepler_container_joules_total{command="ovnkube",container_id="ee0d547a8599aa4ec05f1eddf8a819d2eca94ce12e29d9f1cb4835c7dcb90a72",container_name="ovnkube-node",container_namespace="openshift-ovn-kubernetes",mode="dynamic",pod_name="ovnkube-node-7s4b6"} 969.7180000000001
kepler_container_joules_total{command="thanos",container_id="9052398e2754e7bc410fc8d8cef44ce3ddbff65a74cc8a27e585eb0535c35d2f",container_name="thanos-sidecar",container_namespace="openshift-user-workload-monitoring",mode="dynamic",pod_name="prometheus-user-workload-1"} 969.967
kepler_container_joules_total{command="opm",container_id="f4d9cf8bc0594f1220771b1f1c46f28375351e9f9561fcacb3455c4b8e7c6f98",container_name="registry-server",container_namespace="openshift-observability-operator",mode="dynamic",pod_name="observability-operator-catalog-88vkw"} 970.7349999999999
kepler_container_joules_total{command="registry-s",container_id="29a7077725e772e67c4552ee1a7e980fa82714dff7bfa6b740dc54c3310a43ff",container_name="registry-server",container_namespace="openshift-must-gather-operator",mode="dynamic",pod_name="must-gather-operator-registry-t6gtg"} 971.8729999999999
kepler_container_joules_total{command="registry-s",container_id="0dd690e3c96b4fb7d78e191419e96641f475c7958eea3c17514610e188131a33",container_name="registry-server",container_namespace="openshift-route-monitor-operator",mode="dynamic",pod_name="route-monitor-operator-registry-kv6m6"} 972.4780000000001
kepler_container_joules_total{command="sleep",container_id="e7be7375324a4b2aeb3c77b3bdc2eeed3139407516ea7b3a7ce502a949415c5d",container_name="kube-multus",container_namespace="openshift-multus",mode="dynamic",pod_name="multus-z9rp2"} 972.7570000000001
kepler_container_joules_total{command="kepler",container_id="7c2fb994e9c1671ceeb03ed978402e4002108af94ba41402bb9c0e7222855de9",container_name="kepler-exporter",container_namespace="kepler",mode="dynamic",pod_name="kepler-exporter-8lpdb"} 988.52
kepler_container_joules_total{command="prometheus",container_id="3d2784218396fe5c91c8bb419cc0152ebcbcdc069354de444fb78002f39f0ebf",container_name="prometheus",container_namespace="openshift-user-workload-monitoring",mode="dynamic",pod_name="prometheus-user-workload-1"} 990.1850000000001
kepler_container_joules_total{command="kworker/3:",container_id="system_processes",container_name="system_processes",container_namespace="system",mode="dynamic",pod_name="system_processes"} 1010.1080000000001
kepler_container_joules_total{command="cat",container_id="76f2a9a62e3e616390b5e796a4a1bb01648e69b84810b0cdc4976ca2dfdc048f",container_name="hamster",container_namespace="cpu-load",mode="dynamic",pod_name="random-576f489b67-jmtm5"} 1329.5149999999999
```

PR #788 (2023-07-18): watcher: wait pod status to be container ready to parse container id
fix #787
closed 2023-07-18 by rootfs