sustainable-computing-io/kepler #808: Kepler boot time on BM is much longer than VM.

jiere, 2023-07-21:
### What happened?

Kepler boot time on BM is around 1 min 30 seconds, while on VM is less than 1 second.
After analysis of Kepler verbose log(V=5), found three power model download operations for BM case, which contributes to the cost. 

One is for container component estimator init:
`I0720 02:54:27.856313       1 model.go:87] Model Config CONTAINER_COMPONENTS: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json}`

While we could see [here](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/exporter/exporter.yaml#L32-L35) that we have disabled model download for container component in manifest, why still could see such download..

The other two are [duplicate](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/model.go#L37-L38) process component estimator init:
`I0720 02:54:57.863553       1 model.go:87] Model Config PROCESS_COMPONENTS: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}`

`I0720 02:55:27.870131       1 model.go:87] Model Config PROCESS_COMPONENTS: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}
`

While we dedicatedly set the `InitModelURL` for process component [here](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/process_power.go#L48), which leads to the runtime download.

Are these above behaviors expected then?

### What did you expect to happen?

Use built-in power model by default to save boot time.

### How can we reproduce it (as minimally and precisely as possible)?

`kubectl edit -n kepler daemonset kepler-exporter`

and change `/usr/bin/kepler -v=1` to `/usr/bin/kepler -v=5`. 

Then check the kepler verbose logs:

`kubectl logs -n kepler daemonset/kepler-exporter`

### Kepler image tag

<details>
Latest code
</details>

### Kubernetes version

<details>

```console
$ kubectl version
WARNING: This version information is deprecated and will be replaced with the output from kubectl version --short.  Use --output=yaml|json to get the full version.
Client Version: version.Info{Major:"1", Minor:"26", GitVersion:"v1.26.1", GitCommit:"8f94681cd294aa8cfd3407b8191f6c70214973a4", GitTreeState:"clean", BuildDate:"2023-01-18T15:58:16Z", GoVersion:"go1.19.5", Compiler:"gc", Platform:"linux/amd64"}
Kustomize Version: v4.5.7
Server Version: version.Info{Major:"1", Minor:"27", GitVersion:"v1.27.3", GitCommit:"25b4e43193bcda6c7328a6d147b1fb73a33f1598", GitTreeState:"clean", BuildDate:"2023-06-15T00:36:28Z", GoVersion:"go1.20.5", Compiler:"gc", Platform:"linux/amd64"}
```

</details>

### Cloud provider or bare metal

<details>
KIND
</details>

### OS version

<details>

```console
$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.2 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

$ uname -a
Linux jie-nuc 5.19.0-41-generic #42~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 18 17:40:00 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

sunya-ch, 2023-07-21:
There are three options to load the models with pros/cons.

**Flexibility of model updates for better accuracy:**
Model server >  Remote model file  (URL) > Local model file (in image)

**Overhead and availability**
Local model file > URL > Model server

I agree that we can remove defaultURL in the code out because now we already have the local model file. 
https://github.com/sustainable-computing-io/kepler/blob/8877c42d65dd4125cd40620fad0aada014c1eac8/pkg/model/process_power.go#L39

jiere, 2023-07-21:
Thanks @sunya-ch for your quick comments.
I also noticed such difference:
[Built-in model](https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile#L34): DynComponentModelWeight/**CgroupOnly**/ScikitMixed/ScikitMixed.json

defaultURL in code:  DynComponentModelWeight/**CounterOnly**/ScikitMixed/ScikitMixed.json

Are they the same?

jiere, 2023-07-21:
close with PR merged.

closed 2023-07-21 by jiere