sustainable-computing-io/kepler #811: Expose DynEnergyInOther for containers to Prometheus

tiwatsuka, 2023-07-21:
### What would you like to be added?

Kepler exposes idle power of containers of host components other than CPU and DRAM (`container.IdleEnergyInOther`) to Prometheus, but it doesn't do so dynamic power (`container.DynEnergyInOther`) . It should be exposed as well.

https://github.com/sustainable-computing-io/kepler/blob/f94a6651894f875f556f7494934055dc681ad8ce/pkg/collector/prometheus_collector.go#L765-L770

### Why is this needed?

Both idle and dynamic power attributed to containers are part of power of a container. To visualize container power correctly in dashboard, it is required that both of them are exposed.

marceloamaral, 2023-07-21:
@tiwatsuka can you create a PR for this?

tiwatsuka, 2023-07-27:
closed by #815

closed 2023-07-27 by tiwatsuka