sustainable-computing-io/kepler #812: Usage metric for OTHER in RATIO power model to be configurable

tiwatsuka, 2023-07-21:
### What would you like to be added?

In RATIO power model, Kepler divides power of host components other than CPU and DRAM evenly to all containers. I think it is preferable this behavior to be configurable. 

https://github.com/sustainable-computing-io/kepler/blob/f94a6651894f875f556f7494934055dc681ad8ce/pkg/model/container_power.go#L74

### Why is this needed?

The power of other system components than CPU and DRAM includes like power of fans, loss at PSU, etc. Some of these are indirectly depend on CPU usage of the host. So some cases users would like to divide it by CPU usage or something like that.

---
BTW, it seems that current implementation does not divide dynamic energy of OTHER to containers.  Because `getContainerResUsage()` returns 0 when `usageMetric` is empty, l92 - l95 is skipped.

https://github.com/sustainable-computing-io/kepler/blob/84ce8603c0ce26d33c0e06671e61c912e350ff1d/pkg/model/estimator/local/ratio.go#L90-L96

marceloamaral, 2023-07-21:
https://github.com/sustainable-computing-io/kepler/blob/84ce8603c0ce26d33c0e06671e61c912e350ff1d/pkg/model/estimator/local/ratio.go#L90-L92

We should verify in line 91 if the source is OTHER and allow adding the OTHER energy consumption.
```
if containerResUsage > 0 || component == collector_metric.OTHER || component == collector_metric.UNCORE { 
```

The OTHER power consumption is not necessarily directly related to the CPU, we need to do more investigation on this. Therefore, a safe approach is to evenly divide it.

I created the PR #813 to fix this.

tiwatsuka, 2023-07-27:
I agree with you that it is a safe approach. 
 
My suggestion is making it more flexible. If the usage metric to divide energy was configurable, users could be modify the behavior based on their own observations.

[Here](https://github.com/tiwatsuka/kepler/commit/dce4f6ec920eec11d1eaed04eeba88dccc673bff) is the suggested change. I will make PR if it is acceptable.

rootfs, 2023-07-27:
@tiwatsuka change looks good, PR welcome!

closed 2023-07-29 by rootfs