sustainable-computing-io/kepler #826: integration_test (microshift) needs more logs

rootfs, 2023-07-25:
### What would you like to be added?

I am seeing this test keeps failing yet no useful logs. Logs from one of these runs can be found [here](https://github.com/sustainable-computing-io/kepler/actions/runs/5648369586/job/15304717726#step:8:90)

### Why is this needed?

Need more logs about why this test fails when setting kepler times out on microshift

rootfs, 2023-07-25:
@jiere @vprashar2929 can you take a look?

rootfs, 2023-07-25:
this is seen in two more tests
https://github.com/sustainable-computing-io/kepler/actions/runs/5660558620/job/15336767459?pr=824
https://github.com/sustainable-computing-io/kepler/actions/runs/5660558628/job/15336767034?pr=824

vprashar2929, 2023-07-26:
Yes, I did some digging and seems like kepler-exporter pod gets into the evicted state due to no disk space available on the runner. We have the [step](https://github.com/sustainable-computing-io/kepler/blob/dad4d74397bd22c9e7d87bd7b7cf7317d018efe7/.github/workflows/integration_test.yml#L97) added to clean any dangling images in workflow. I will add a patch to fix this along with some improvement in logging.

vprashar2929, 2023-07-26:
Raised the [PR](https://github.com/sustainable-computing-io/local-dev-cluster/pull/40) for adding support to create volume by ourself before running microshift container. This will not allow disk pressure to be created.

Will raise another PR to improve the logging in CI here.

vprashar2929, 2023-07-27:
Raised [PR](https://github.com/sustainable-computing-io/kepler/pull/835) for improving logging in CI. 
CC: @rootfs

closed 2023-10-02 by stale[bot]