sustainable-computing-io/kepler #828: Kepler uses estimator method for power stats

vimalk78, 2023-07-26:
### What happened?

Installed kepler using latest `kepler-operator`. 
Created `Kepler` spec using 
```
apiVersion: kepler.system.sustainable.computing.io/v1alpha1
kind: Kepler
metadata:
  labels:
    app.kubernetes.io/name: kepler
    app.kubernetes.io/instance: kepler
    app.kubernetes.io/part-of: kepler-operator
    app.kubernetes.io/managed-by: kustomize
    app.kubernetes.io/created-by: kepler-operator
  name: kepler
spec:
  # TODO(user): Add fields here
  exporter:
    image: quay.io/sustainable_computing_io/kepler:latest
    port: 9103
```

Following log line appears in kepler log
```
I0726 06:11:44.552907       1 power.go:64] Not able to obtain power, use estimate method
```
AFAIU estimator method is not to be used in kepler.

Complete Kepler log
```
$ oc -n openshift-kepler-operator logs kepler-exporter-ds-4dkcc -c kepler-exporter 
I0726 06:11:44.541064       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0726 06:11:44.552571       1 exporter.go:155] Kepler running on version: b2ed6ec
I0726 06:11:44.552652       1 config.go:258] using gCgroup ID in the BPF program: true
I0726 06:11:44.552684       1 config.go:260] kernel version: 5.14
I0726 06:11:44.552789       1 config.go:191] kernel source dir is set to /usr/share/kepler/kernel_sources
I0726 06:11:44.552837       1 exporter.go:179] EnabledBPFBatchDelete: true
I0726 06:11:44.552907       1 power.go:64] Not able to obtain power, use estimate method
I0726 06:11:44.552918       1 redfish.go:169] failed to get redfish credential file path
I0726 06:11:44.552923       1 power.go:55] use acpi to obtain power
I0726 06:11:44.553071       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0726 06:11:44.605329       1 exporter.go:198] Initializing the GPU collector
I0726 06:11:50.610069       1 watcher.go:66] Using in cluster k8s config
I0726 06:11:50.710516       1 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
chdir(/lib/modules/5.14.0-284.11.1.el9_2.x86_64/build): No such file or directory
I0726 06:11:50.755873       1 bcc_attacher.go:84] failed to attach the bpf program: <nil>
I0726 06:11:50.755889       1 bcc_attacher.go:158] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=4]: failed to attach the bpf program: <nil>, from default kernel source.
I0726 06:11:50.755899       1 bcc_attacher.go:161] trying to load eBPF module with kernel source dir /usr/share/kepler/kernel_sources/4.18.0-477.13.1.el8_8.x86_64
cannot attach kprobe, probe entry may not exist
I0726 06:11:51.484853       1 bcc_attacher.go:167] Successfully loaded eBPF module with options: [-DMAP_SIZE=10240 -DNUM_CPUS=4] from kernel source "/usr/share/kepler/kernel_sources/4.18.0-477.13.1.el8_8.x86_64"
I0726 06:11:51.484879       1 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=4]
I0726 06:11:51.513341       1 exporter.go:251] Started Kepler in 6.960805584s

```

### What did you expect to happen?

One of the following sources used for power stats
- Sysfs powercap
- MSR
- hwmon

### How can we reproduce it (as minimally and precisely as possible)?

Create an Openshift 4.14 cluster on AWS

Install kepler using kepler operator at commit 74961897db4d3311d74638e6be659121aa5be4d3
using following steps
```
$ make docker-build docker-push IMG=quay.io/vimalkum/kepler-operator:v1alpha1
$ make bundle bundle-build bundle-push IMG=quay.io/vimalkum/kepler-operator:v1alpha1 BUNDLE_IMG=quay.io/vimalkum/kepler-operator-bundle:v1alpha1 
$ operator-sdk run bundle quay.io/vimalkum/kepler-operator-bundle:v1alpha1 --install-mode AllNamespaces --namespace=openshift-operators
$ oc apply -k config/samples/
```

### Kepler image tag

<details>
quay.io/sustainable_computing_io/kepler:latest
</details>

### Kubernetes version

<details>

```console
$ oc version
Client Version: 4.9.10
Server Version: 4.14.0-ec.0
Kubernetes Version: v1.26.2+d2e245f
```

</details>

### Cloud provider or bare metal

<details>
AWS
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
sh-5.1#  cat /etc/os-release
NAME="Red Hat Enterprise Linux"
VERSION="9.1 (Plow)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="9.1"
PLATFORM_ID="platform:el9"
PRETTY_NAME="Red Hat Enterprise Linux 9.1 (Plow)"
ANSI_COLOR="0;31"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:redhat:enterprise_linux:9::baseos"
HOME_URL="https://www.redhat.com/"
DOCUMENTATION_URL="https://access.redhat.com/documentation/red_hat_enterprise_linux/9/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 9"
REDHAT_BUGZILLA_PRODUCT_VERSION=9.1
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.1"

$ uname -a
sh-5.1# uname -a
Linux kepler-exporter-ds-4dkcc 5.14.0-284.11.1.el9_2.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Apr 12 10:45:03 EDT 2023 x86_64 x86_64 x86_64 GNU/Linux

```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

marceloamaral, 2023-07-26:
Thank you for bringing this issue to our attention. We understand the importance of accurately estimating power consumption in AWS clusters which utilizes Virtual Machines (VMs). Unfortunately, VMs on AWS do not expose crucial system power metrics, such as RAPL or ACPI, making it challenging (or impossible) to directly measure power consumption.

To address this limitation, Kepler currently relies on pre-trained power models for estimating power consumption. While this provides useful insights into the container power consumption, we acknowledge that for optimal accuracy, the power models should ideally be trained specifically for each CPU architecture/model.

Currently, our pre-trained power model collection covers a limited set of CPU architecture/models. However, we are committed to expanding this collection to encompass a broader range of CPU architectures/models, thereby enhancing the accuracy of power consumption estimates across AWS VMs.

We appreciate your patience and understanding as we continue to work on improving the power estimation capabilities in AWS clusters. If you have any further questions or suggestions, please don't hesitate to let us know. Your feedback is invaluable to us.

jichenjc, 2023-07-27:
not sure you are looking for something like 
https://github.com/sustainable-computing-io/kepler-model-server/tree/main/tests/test_models ?

e.g https://github.com/sustainable-computing-io/kepler-model-server/blob/main/tests/test_models/AbsComponentModelWeight/Full/KerasCompWeightFullPipeline/KerasCompWeightFullPipeline.json ?

vimalk78, 2023-07-27:
@jichenjc I am looking for some report about on which environments (e.g. any bare metal, AWS, Azure etc) we have tested kepler, and of those environments which could give access to kepler to read CPU power stats.

marceloamaral, 2023-07-27:
Kepler can read Power metrics from any x86 bare metal nodes with RAPL and/or enabled ACPI sensor. However, accessing these metrics requires admin privileges on the node.

Regarding VMs, Kepler relies on pre-trained power models to estimate power consumption, and validation work is currently in progress. Contributions are highly welcomed!

husky-parul, 2023-08-03:
> @jichenjc I am looking for some report about on which environments (e.g. any bare metal, AWS, Azure etc) we have tested kepler, and of those environments which could give access to kepler to read CPU power stats.

@vimalk78 It doesn't makes much difference if it is AWS or Azure. The differentiating criteria is VM/BM or access to power source or not. This page explains https://sustainable-computing.io/design/power_model/  what power model we use in different scenarios.

husky-parul, 2023-08-04:
Can we use Scaphandre to compare the metrics gathered by Kepler? I am not sure if Scaphandre gives container level power metrics if deployed on a test cluster comprising VM based nodes. Wondering if someone investigated this path before I start exploring
@marceloamaral @jichenjc

nikimanoledaki, 2023-08-08:
> Can we use Scaphandre to compare the metrics gathered by Kepler? I am not sure if Scaphandre gives container level power metrics if deployed on a test cluster comprising VM based nodes.

@rossf7 @mrchrisadams do you know if there has been any success in accessing power data (through RAPL or ACPI) on a VM/BM machine in AWS/Azure/GCP with Scaphandre?

marceloamaral, 2023-08-08:
It seems Scaphandre is used for computing process power consumption on bare-metal systems with access to real-time power metrics such as RAPL. It's likely using a ratio power model. For the ratio model, Scaphandre uses the Process Time metric in Linux, measured in Jiffies.

Their official documentation states that work on Kubernetes integration is ongoing:
https://hubblo-org.github.io/scaphandre-documentation/explanations/about-containers.html

The Scaphandre documentation says that aggregating process power to container and pod power consumption is challenging. If someone address this, a comparison between Scaphandre and Kepler on bare-metal nodes becomes a possibility.

It's important to note that Scaphandre does not use a pre-trained power model to estimate power on VMs lacking real-time power metrics like RAPL. Instead, Scaphandre exposes power consumption metrics of a VM from the bare-metal level. This is achieved through hypervisor features. Our future plan for Kepler is to implement a similar approach for scenarios that we might have access to the bare-metal.

rossf7, 2023-08-08:
> do you know if there has been any success in accessing power data (through RAPL or ACPI) on a VM/BM machine in AWS/Azure/GCP with Scaphandre?

@nikimanoledaki I gave this a try with EKS. scaphandre works on a `m5.metal` BM instance but failed in a `m5.xlarge` VM because RAPL is not available. https://github.com/hubblo-org/scaphandre/discussions/277

> Their official documentation states that work on Kubernetes integration is ongoing:

@marceloamaral The docs are out of date here. The k8s integration is merged. Thanks for the heads up I'll update them. 

The integration parses `/proc/PID/cgroup` to get the container ID which it matches against the pods and adds the kubernetes_pod_name and kubernetes_pod_namespace labels to the metrics. If these labels are missing its usually because we can't parse the container ID.

I tested with ubuntu 20.04 with cgroup v1 because this usually works and EKS supports it.

```
curl -s http://localhost:8080/metrics | grep kubernetes_pod
scaph_process_power_consumption_microwatts{kubernetes_pod_namespace="default",kubernetes_node_name="ip-192-168-5-146.us-east-2.compute.internal",cmdline="/usr/local/bin/scaphandreprometheus--containers=true",pid="23463",exe="scaphandre",kubernetes_pod_name="scaphandre-j9xb9",container_id="22b6666521c8843fb12d5ba3ad1e04753c7267133f5a6a3181478f8de6c9938e",container_scheduler="kubernetes"} 18938141
scaph_process_power_consumption_microwatts{pid="10910",exe="coredns",cmdline="/coredns-conf/etc/coredns/Corefile",container_scheduler="kubernetes",container_id="c61de5e360426d242ab8efc3d224c230cf8e98f5e4cd0a4d4bd36f5989ee81c0",kubernetes_pod_name="coredns-8fd4db68f-tl547",kubernetes_pod_namespace="kube-system",kubernetes_node_name="ip-192-168-5-146.us-east-2.compute.internal"} 1052118
scaph_process_power_consumption_microwatts{cmdline="/coredns-conf/etc/coredns/Corefile",kubernetes_node_name="ip-192-168-5-146.us-east-2.compute.internal",exe="coredns",container_scheduler="kubernetes",kubernetes_pod_namespace="kube-system",pid="10909",kubernetes_pod_name="coredns-8fd4db68f-fdxlc",container_id="b9eb7f54db50ee81ff61b8180b2474b35ca812bb9d02ad6b73d77c93e39a5ea5"} 1052118
scaph_process_power_consumption_microwatts{container_scheduler="kubernetes",container_id="99f15d32558e9fb3321cabfe6be2e8247a38bffac0d26a27b97d278d48ea01e2",kubernetes_pod_namespace="kube-system",exe="aws-k8s-agent",kubernetes_pod_name="aws-node-zlndz",kubernetes_node_name="ip-192-168-5-146.us-east-2.compute.internal",pid="10564",cmdline="./aws-k8s-agent|tee-iaws-k8s-agent.log2>&1"} 0
scaph_process_power_consumption_microwatts{exe="aws-vpc-cni",kubernetes_pod_namespace="kube-system",container_scheduler="kubernetes",pid="10547",container_id="99f15d32558e9fb3321cabfe6be2e8247a38bffac0d26a27b97d278d48ea01e2",cmdline="/app/aws-vpc-cni",kubernetes_pod_name="aws-node-zlndz",kubernetes_node_name="ip-192-168-5-146.us-east-2.compute.internal"} 0
scaph_process_power_consumption_microwatts{cmdline="kube-proxy--v=2--config=/var/lib/kube-proxy-config/config--hostname-override=ip-192-168-5-146.us-east-2.compute.internal",pid="10369",kubernetes_pod_namespace="kube-system",kubernetes_pod_name="kube-proxy-zkkrp",kubernetes_node_name="ip-192-168-5-146.us-east-2.compute.internal",exe="kube-proxy",container_id="f68f49de4e7bbe38ee6f604de30f424978a792010f7444c0203f0657d63a8b0e",container_scheduler="kubernetes"} 0

k get po -A
NAMESPACE     NAME                      READY   STATUS    RESTARTS   AGE
default       scaphandre-j9xb9          1/1     Running   0          41m
kube-system   aws-node-zlndz            1/1     Running   0          54m
kube-system   coredns-8fd4db68f-fdxlc   1/1     Running   0          62m
kube-system   coredns-8fd4db68f-tl547   1/1     Running   0          62m
kube-system   kube-proxy-zkkrp          1/1     Running   0          54m
```

> It doesn't makes much difference if it is AWS or Azure. The differentiating criteria is VM/BM or access to power source or not.

@husky-parul I agree I think VM/BM is the distinction that matters and this applies to scaphandre too.

marceloamaral, 2023-08-09:
Thanks for the follow-up, @rossf7.

I've noticed another difference between the scaphandre and kepler power models when running on bare-metal with RAPL. It seems that scaphandre doesn't account for idle power, which refers to the constant power consumption even when there's no significant load being utilized.

The absence of consideration for idle power introduces a limitation – particularly concerning the fair allocation of constant power across all active processes. According to the GHG (GreenHouse Gas) protocol, distributing the constant power should consider in the application's size, while dynamic power allocation should be based on resource utilization.

mrchrisadams, 2023-08-09:
hi @nikimanoledaki, @rossf7 

I hope this isn't too late. The most thorough public investigation about access to underlying machine power usage I know about is the work by Arne and his team at Green Coding Berlin.

It explains how Scaphandre and Kepler measure power differently, and also outlines what access is made available by various cloud providers:

https://www.green-coding.berlin/case-studies/cpu-utilization-usefulness/

That's the latest stuff I know - the article lists a few commands to run to get an understanding of what access is made available by varous cloud providers, as well as outlining why they can either overestimate or underestimate. 

I hope that helps. I'll tag Arne, as I can't remember his username on github, and he might have more to share.

closed 2023-10-11 by vimalk78