sustainable-computing-io/kepler #837: Grafana Dashboard - slow refresh with all namespaces

AdrianHammond, 2023-07-27:
### What happened?

Installed Kepler and Grafana dashboard on a ROSA OCP 4.11.34 cluster (customer test cluster). The dashboard install was using the script in manifests/config/dashboard.

When all namespaces selected the dashboard never refreshes so we get no / incomplete data. If a single namespace is selected e.g. Kepler that data refreshes / displays correctly. 

### What did you expect to happen?

For the dashboard to refresh in reasonable time with all namespaces selected. 

I have tried on demo.redhat.com ROSA cluster and dashboard refresh was acceptable. 

### How can we reproduce it (as minimally and precisely as possible)?

I tried on demo.redhat.com ROSA cluster and could not recreate. Its a solid issue on the customer cluster. 

### Anything else we need to know?

We need not change any manifest setting for dashboard install. Looking at the grafana pods in the kepler namespace they do not look to be consuming significant resources. The dashboard was running on Chrome / Windows. In my tests I am using Chrome/OSX. I will confirm OCP version of my test ROSA cluster, I think it was 4.12. 

### Kepler image tag

:latest - manifest install of kepler used rather than operator as operator install failed. 

### Kubernetes version

1.24

### Cloud provider or bare metal

AWS ROSA

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

AdrianHammond, 2023-07-27:
My testing was on a 4.13.4 ROSA cluster

husky-parul, 2023-07-27:
Do you mean that if you select by namespaces individually you see the power metrics but not if you select all namespaces?
If you are getting metrics for namespaces individually that means kepler is exporting metrics fine and maybe Grafana is getting overwhelmed by the data. 

So just wanted to make sure if you are getting power metrics from Kepler for all namespaces (when selected individually)

Also curious how was Grafana deployed on ROSA? Is it a community operator?

AdrianHammond, 2023-07-27:
Hi Parul, 

Yes, individual namespaces we see the data. When we look at all namespaces we don't see the data. 

We used this [script]{https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/dashboard/deploy-grafana.sh} to install the dashboard.

AdrianHammond, 2023-07-28:
@husky-parul 

As requested we checked that datasource config and it tested okay.

Just to confirm the cluster has 175 namespaces, 773 pods

Thanks Adrian

rootfs, 2023-07-29:
having all namespaces listed doesn't scale well. We can use `topk` function in grafana to list only the top namespaces to avoid rendering complicated dashboard

AdrianHammond, 2023-07-31:
@rootfs @husky-parul  - I am going to suggest that customer tries this [version](https://github.com/AdrianHammond/kepler/blob/topk-dashboard/manifests/config/dashboard/04-grafana-dashboard.yaml) of the standard dashboad which has kepler a default namespace and calcualtes CO2 for top10 namespaces.

rootfs, 2023-07-31:
@AdrianHammond the CO2 dashboard has to ingest the CO2 metrics with kepler. But we can use the similar `topk` function in the kepler dashboard to help reduce the rendering overhead.

AdrianHammond, 2023-08-01:
@husky-parul @rootfs feedback from customer is that demo dashboard now renders now that we have reduced amount of data using topk

rootfs, 2023-08-01:
@AdrianHammond that's great! Do you happen to have a fix? We are happy to merge it any time!

AdrianHammond, 2023-08-01:
@rootfs  - I do have a fix, let me review / tidy-up and I will raise a PR

AdrianHammond, 2023-08-01:
@rootfs  - here is the PR #852

closed 2023-10-07 by stale[bot]