sustainable-computing-io/kepler #843: Ingesting metrics into Prometheus causes very high memory usage on Prometheus

onedr0p, 2023-07-28:
### What would you like to be added?

Hi 👋🏼 

I deployed Kepler with the helm chart [kepler-0.4.5](https://github.com/sustainable-computing-io/kepler-helm-chart/releases/tag/kepler-0.4.5) and noticed that my Prometheus memory usage went from ~3GB to nearly 9GB! After I removed Kepler and reset Prometheus it dropped back down to the 3GB usage.

### Why is this needed?

Well deploying Kepler currently will maybe have some unattended consequences.

rootfs, 2023-07-28:
can you the output of the prometheus tsdb, i.e. via 
```sh
kubectl exec -ti -n monitoring prometheus-k8s-0 -- sh -c "promtool tsdb analyze /prometheus"
```

onedr0p, 2023-07-28:
I'll have to find time to deploy this again, it wasn't hard but I'm going to be on vacation starting tomorrow. My cluster isn't really that large...

```
❯ k get po -A | wc -l
228
```

rootfs, 2023-08-01:
@onedr0p can you also post a list of metrics so we can identify if there are too many info. Thanks

sthaha, 2023-08-02:
@onedr0p what is the scrape interval of the prometheus ? Is there a config you can share?