sustainable-computing-io/kepler #844: kepler (libbpf) misses multiple processes

sunya-ch, 2023-07-28:
### What happened?

Many processes especially the targeting workload is missing from kepler BPF collection.

```
kubectl logs -n kepler kepler-exporter-l8bn7|grep coremark|grep active
I0728 12:06:27.288384       1 metric_collector.go:126] energy from pod/container (1 active processes): name: coremark-cpeh-2340320356-nbjcg/coremark namespace: cpe-operator-system
I0728 12:06:30.054514       1 metric_collector.go:126] energy from pod/container (9 active processes): name: coremark-cpeh-2340320356-nbjcg/coremark namespace: cpe-operator-system
I0728 12:06:33.135661       1 metric_collector.go:126] energy from pod/container (8 active processes): name: coremark-cpeh-2340320356-nbjcg/coremark namespace: cpe-operator-system
I0728 12:06:36.058358       1 metric_collector.go:126] energy from pod/container (8 active processes): name: coremark-cpeh-2340320356-nbjcg/coremark namespace: cpe-operator-system
I0728 12:06:39.063118       1 metric_collector.go:126] energy from pod/container (8 active processes): name: coremark-cpeh-2340320356-nbjcg/coremark namespace: cpe-operator-system
I0728 12:06:42.061351       1 metric_collector.go:126] energy from pod/container (9 active processes): name: coremark-cpeh-2340320356-nbjcg/coremark namespace: cpe-operator-system
I0728 12:06:48.055989       1 metric_collector.go:126] energy from pod/container (0 active processes): name: coremark-cpeh-3118383817-ggvs2/coremark namespace: cpe-operator-system
I0728 12:06:51.064529       1 metric_collector.go:126] energy from pod/container (0 active processes): name: coremark-cpeh-3118383817-ggvs2/coremark namespace: cpe-operator-system
```

### What did you expect to happen?

At least one active process has been detected for each pod.

### How can we reproduce it (as minimally and precisely as possible)?

Run some workload process with high verbosity. 

### Kepler image tag

<details>
https://quay.io/repository/sustainable_computing_io/kepler/manifest/sha256:91f851348b4053ca097f14ad7dac732ae03cb0e9e72bbf821f004a01e2612f3f
</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
Server Version: version.Info{Major:"1", Minor:"25", GitVersion:"v1.25.3", GitCommit:"434bfd82814af038ad94d62ebe59b133fcb50506", GitTreeState:"clean", BuildDate:"2022-10-25T19:35:11Z", GoVersion:"go1.19.2", Compiler:"gc", Platform:"linux/amd64"}
```

</details>

### Cloud provider or bare metal

<details>
kind on bare metal
</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-07-28:
I think those with 0 active processes are pre-populated pods that are queried from kubelet

sunya-ch, 2023-07-31:
I think it did have an activity because coremark with many threads are running but all the lifetime it cannot be detected.
For the cgroup stats, if the activity is not active, pid is not set and stats are not assigned to the container but assigned to the process with pid=0.

jichenjc, 2023-07-31:
might related to https://github.com/sustainable-computing-io/kepler/issues/766

sunya-ch, 2023-07-31:
> might related to #766

Is all active pods such as kepler-exporter has zero activity in your system? 
If all is zero except system-process, it might be a different issue such as different pid (linux) namespace.
If some pods can be detected as above, I think it is the same issue.

rootfs, 2023-07-31:
@sunya-ch this behavior may be correct, as you see in this container (not pod), the container `kube-rbac-proxy` might not be actively running event though `cpe-operator-controller-manager-696fc99fb9-gzxl4` could be busy.

```
(0 active processes): name: cpe-operator-controller-manager-696fc99fb9-gzxl4/kube-rbac-proxy namespace: cpe-operator-system
```

sunya-ch, 2023-08-01:
> @sunya-ch this behavior may be correct, as you see in this container (not pod), the container `kube-rbac-proxy` might not be actively running event though [...]

Yes, the `kube-rbac-proxy` of `cpe-operator` is correct. But, the `coremark` pod which has only one container should not be inactive for all of its lifetime.

jichenjc, 2023-08-02:
>>>If some pods can be detected as above, I think it is the same issue.

they do, I run some nginx and curl it but still not able to see them

rootfs, 2023-08-02:
@sunya-ch @jichenjc this is probably a non-issue, but I just created #854 if you think this needs to be fixed

jichenjc, 2023-08-04:
per https://github.com/sustainable-computing-io/kepler/pull/854#issuecomment-1664880608

marceloamaral, 2023-08-17:
@sunya-ch @rootfs @jichenjc 
The PR #872 will fix this problem.

PR #854 (2023-08-04): collector: print total processes
fix #844 

I tested my local environment, the `node-exporter` container is not running frequently and always gets "0 active process". With this change, this is the new output
```
I0803 14:40:38.844820 4133674 metric_collector.go:126] energy from pod/container (2 of 12 active processes): name: node-exporter-27nsx/node-exporter namespace: monitoring 
I0803 14:40:41.844848 4133674 metric_collector.go:126] energy from pod/container (0 of 12 active processes): name: node-exporter-27nsx/node-exporter namespace: monitoring 
I0803 14:40:44.844257 4133674 metric_collect [... cut]
closed 2023-08-28 by rootfs