sustainable-computing-io/kepler #845: [Tracking] Kepler Adopters

nikimanoledaki, 2023-07-30:
### Which document would you like to be mentioned?

This is a proposal to add a process for tracking Kepler adopters.

Once there is enough interest in this issue, we could move to a formal process for tracking Kepler adopters, as in the steps below inspired by the Flux process for this.

Please share this issue with Kepler users.

### Steps
- [ ] Create `ADOPTERS` page e.g. https://github.com/fluxcd/community/blob/main/ADOPTERS.md
- [ ] Add `Adopters` section to docs website e.g. https://fluxcd.io/adopters/ which reads from [this data](https://github.com/fluxcd/website/blob/main/data/adopters/1-flux-v2.yaml)
- [ ] Add process for users to add their own org as an adopter to the contributing doc e.g. https://fluxcd.io/contributing/flux-adopters/

### Why is this needed?

Knowing who uses Kepler would help drive adoption and get feedback about how Kepler is used.

rootfs, 2023-07-31:
@nikimanoledaki can you help create the `ADOPTERS.md` here and in kepler-doc? Thanks!

Obasoro, 2023-09-01:
@rootfs I would like to try this out personally on my local system. Are there any docs for a guide? Also to contribute to the project.

rootfs, 2023-09-02:
you can find the installation guide here https://sustainable-computing.io/. Please share your experience with us in the doc or the adopters

Obasoro, 2023-09-02:
Thanks @rootfs. I would revert back on this.

Obasoro, 2023-09-04:
Thanks @rootfs.

### Error experienced
when run the code below, the port-forwarding starts
`kubectl -n monitoring port-forward svc/grafana 3000`
Forwarding from 127.0.0.1:3000 -> 3000
Forwarding from [::1]:3000 -> 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
Handling connection for 3000
E0904 05:21:39.151199   57012 portforward.go:370] error creating forwarding stream for port 3000 -> 3000: Timeout occurred
Handling connection for 3000
E0904 05:22:09.422265   57012 portforward.go:370] error creating forwarding stream for port 3000 -> 3000: Timeout occurred
E0904 05:22:09.422660   57012 portforward.go:370] error creating forwarding stream for port 3000 -> 3000: Timeout occurred
E0904 05:22:09.422703   57012 portforward.go:370] error creating forwarding stream for port 3000 -> 3000: Timeout occurred
E0904 05:22:09.818505   57012 portforward.go:347] error creating error stream for port 3000 -> 3000: Timeout occurred
Handling connection for 3000
Handling connection for 3000
E0904 05:23:09.981405   57012 portforward.go:370] error creating forwarding stream for port 3000 -> 3000: Timeout occurred
Handling connection for 3000
E0904 05:23:16.000755   57012 portforward.go:347] error creating error stream for port 3000 -> 3000: Timeout occurred

Obasoro, 2023-09-04:
### Running Kepler on a local kind cluster

I experience this error too after running the code snippet below

`kubectl apply -f _output/generated-manifest/deployment.yaml`

Unable to connect to the server: net/http: request canceled (Client.Timeout exceeded while awaiting headers)

Unable to connect to the server: net/http: TLS handshake timeou

marceloamaral, 2023-09-05:
Is there a list of Kepler Adopters? If Yes, we can start creating this file!

BGrasnick, 2023-09-05:
I would want to tackle this if nobody else already did some work? @nikimanoledaki

husky-parul, 2023-09-05:
I've assigned it to you. Please feel free to pick it up @BGrasnick

Obasoro, 2023-09-05:
I would be glad to also test when done @BGrasnick.

BGrasnick, 2023-09-05:
Feel free to review the two PRs @Obasoro 👍

PR #914 (2023-09-05): Feat: add ADOPTERS.md
Feat: add ADOPTERS.md 
fixes #845 
should not be merged before https://github.com/sustainable-computing-io/kepler-doc/pull/95
commit 4cb1291f (2023-09-05): Feat: add ADOPTERS.md (#914)

fixes #845
should not be merged before https://github.com/sustainable-computing-io/kepler-doc/pull/95
closed 2023-09-05 by rootfs