sustainable-computing-io/kepler #847: kepler rpm starts with 'cannot attach kprobe, probe entry may not exist'

aspanner, 2023-07-31:
### What happened?

Installed latest rpm 0.5.3 on Fedora 38.

Tried 2 different ways with systemctl start kepler and sudo kepler.
Both create the same error message:

❯ sudo dnf localinstall kepler-0.5.3-0.5.3.x86_64.rpm

Last metadata expiration check: 2:55:10 ago on Mon Jul 31 12:14:50 2023.
Dependencies resolved.
===================================================================================================================================================================================================================
 Package                                         Architecture                                    Version                                               Repository                                             Size
===================================================================================================================================================================================================================
Installing:
 kepler                                          x86_64                                          0.5.3-0.5.3                                           @commandline                                          7.5 M

Transaction Summary
===================================================================================================================================================================================================================
Install  1 Package

Total size: 7.5 M
Installed size: 32 M
Is this ok [y/N]: y
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                                                           1/1 
  Installing       : kepler-0.5.3-0.5.3.x86_64                                                                                                                                                                 1/1 
  Running scriptlet: kepler-0.5.3-0.5.3.x86_64                                                                                                                                                                 1/1 
  Verifying        : kepler-0.5.3-0.5.3.x86_64                                                                                                                                                                 1/1 

Installed:
  kepler-0.5.3-0.5.3.x86_64                                                                                                                                                                                        

Complete!
❯ sudo kepler
I0731 15:10:16.761050  323170 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
E0731 15:10:16.762861  323170 utils.go:117] getCPUArch failure: open /var/lib/kepler/data/normalized_cpu_arch.csv: no such file or directory
I0731 15:10:16.769672  323170 exporter.go:155] Kepler running on version: a7a6cb1
I0731 15:10:16.769687  323170 config.go:258] using gCgroup ID in the BPF program: true
I0731 15:10:16.769705  323170 config.go:260] kernel version: 6.3
I0731 15:10:16.769785  323170 exporter.go:179] EnabledBPFBatchDelete: true
I0731 15:10:16.769840  323170 redfish.go:169] failed to get redfish credential file path
I0731 15:10:16.770237  323170 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0731 15:10:16.770398  323170 watcher.go:66] Using in cluster k8s config
I0731 15:10:16.770405  323170 watcher.go:73] failed to get config: unable to load in-cluster configuration, KUBERNETES_SERVICE_HOST and KUBERNETES_SERVICE_PORT must be defined
I0731 15:10:16.770445  323170 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
In file included from /virtual/main.c:17:
In file included from include/linux/sched.h:12:
In file included from arch/x86/include/asm/current.h:9:
In file included from include/linux/cache.h:6:
In file included from arch/x86/include/asm/cache.h:5:
In file included from include/linux/linkage.h:8:
In file included from arch/x86/include/asm/linkage.h:6:
arch/x86/include/asm/ibt.h:77:8: warning: 'nocf_check' attribute ignored; use -fcf-protection to enable the attribute [-Wignored-attributes]
extern __noendbr u64 ibt_save(bool disable);
       ^
arch/x86/include/asm/ibt.h:32:34: note: expanded from macro '__noendbr'
#define __noendbr       __attribute__((nocf_check))
                                       ^
arch/x86/include/asm/ibt.h:78:8: warning: 'nocf_check' attribute ignored; use -fcf-protection to enable the attribute [-Wignored-attributes]
extern __noendbr void ibt_restore(u64 save);
       ^
arch/x86/include/asm/ibt.h:32:34: note: expanded from macro '__noendbr'
#define __noendbr       __attribute__((nocf_check))
                                       ^
2 warnings generated.
cannot attach kprobe, probe entry may not exist
I0731 15:10:17.634387  323170 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=8 -DSET_GROUP_ID]
I0731 15:10:17.634723  323170 exporter.go:251] Started Kepler in 865.067206ms

### What did you expect to happen?

Kepler to start without error message.

### How can we reproduce it (as minimally and precisely as possible)?

Download rpm release-0.5.3 from https://sustainable-computing.io/installation/kepler-rpm/
tar -xfv tarball
rpm localinstall kepler-0.5.3-0.5.3.x86_64.rpm as per documentation at  https://sustainable-computing.io/installation/kepler-rpm/

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

vimalk78, 2023-07-31:
the following command gives list of attached `kprobes` in the kernel
```
sudo cat /sys/kernel/debug/kprobes/list
```

can you verify using this method, and paste the output ?

vimalk78, 2023-07-31:
I installed rpm using the same steps as you mentioned, i get following output
```
$ sudo dnf localinstall ./kepler-0.5.3-0.5.3.x86_64.rpm 
...
...
Installed:
  kepler-0.5.3-0.5.3.x86_64                                                                                                                                                                                                                  

Complete!

$ sudo kepler
I0731 11:06:15.414564  283792 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
E0731 11:06:15.416767  283792 utils.go:117] getCPUArch failure: open /var/lib/kepler/data/normalized_cpu_arch.csv: no such file or directory
I0731 11:06:15.429834  283792 exporter.go:155] Kepler running on version: a7a6cb1
I0731 11:06:15.429854  283792 config.go:258] using gCgroup ID in the BPF program: true
I0731 11:06:15.429975  283792 config.go:260] kernel version: 6.1
I0731 11:06:15.430006  283792 exporter.go:179] EnabledBPFBatchDelete: true
I0731 11:06:15.430110  283792 redfish.go:169] failed to get redfish credential file path
I0731 11:06:15.430705  283792 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0731 11:06:15.431055  283792 watcher.go:66] Using in cluster k8s config
I0731 11:06:15.431063  283792 watcher.go:73] failed to get config: unable to load in-cluster configuration, KUBERNETES_SERVICE_HOST and KUBERNETES_SERVICE_PORT must be defined
I0731 11:06:15.431170  283792 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
cannot attach kprobe, probe entry may not exist
I0731 11:06:16.391791  283792 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=8]
I0731 11:06:16.391856  283792 exporter.go:224] failed to start : failed to get response: failed to read from "/var/run/secrets/kubernetes.io/serviceaccount/token": open /var/run/secrets/kubernetes.io/serviceaccount/token: no such file or directory
I0731 11:06:16.391953  283792 exporter.go:251] Started Kepler in 962.145961ms

```
it says `cannot attach kprobe, probe entry may not exist`  but there is an entry in kprobe list
```
$ sudo cat /sys/kernel/debug/kprobes/list 
ffffffffa412ab20  k  finish_task_switch.isra.0+0x0    [FTRACE]

```

jichenjc, 2023-07-31:
I am guessing it's because following (but not sure) 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/bpfassets/attacher/bcc_attacher.go#L101

so it cause this report
https://github.com/iovisor/bcc/blob/master/src/cc/libbpf.c#L1266

maybe print some log then check what's the output?

aspanner, 2023-07-31:
I thought, that it might be bc my /proc/sys/kernel/perf_event_paranoid only allows user space measurements as it was set to 2. But I changed the setting to -1 now and it still behaves the same:

❯ cat /proc/sys/kernel/perf_event_paranoid
-1
❯ sudo kepler
I0731 17:28:20.922271    7126 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
E0731 17:28:20.923402    7126 utils.go:117] getCPUArch failure: open /var/lib/kepler/data/normalized_cpu_arch.csv: no such file or directory
I0731 17:28:20.929297    7126 exporter.go:155] Kepler running on version: a7a6cb1
I0731 17:28:20.929304    7126 config.go:258] using gCgroup ID in the BPF program: true
I0731 17:28:20.929317    7126 config.go:260] kernel version: 6.3
I0731 17:28:20.929397    7126 exporter.go:179] EnabledBPFBatchDelete: true
I0731 17:28:20.929431    7126 redfish.go:169] failed to get redfish credential file path
I0731 17:28:20.929798    7126 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0731 17:28:20.929927    7126 watcher.go:66] Using in cluster k8s config
I0731 17:28:20.929932    7126 watcher.go:73] failed to get config: unable to load in-cluster configuration, KUBERNETES_SERVICE_HOST and KUBERNETES_SERVICE_PORT must be defined
I0731 17:28:20.929972    7126 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
In file included from /virtual/main.c:17:
In file included from include/linux/sched.h:12:
In file included from arch/x86/include/asm/current.h:9:
In file included from include/linux/cache.h:6:
In file included from arch/x86/include/asm/cache.h:5:
In file included from include/linux/linkage.h:8:
In file included from arch/x86/include/asm/linkage.h:6:
arch/x86/include/asm/ibt.h:77:8: warning: 'nocf_check' attribute ignored; use -fcf-protection to enable the attribute [-Wignored-attributes]
extern __noendbr u64 ibt_save(bool disable);
       ^
arch/x86/include/asm/ibt.h:32:34: note: expanded from macro '__noendbr'
#define __noendbr       __attribute__((nocf_check))
                                       ^
arch/x86/include/asm/ibt.h:78:8: warning: 'nocf_check' attribute ignored; use -fcf-protection to enable the attribute [-Wignored-attributes]
extern __noendbr void ibt_restore(u64 save);
       ^
arch/x86/include/asm/ibt.h:32:34: note: expanded from macro '__noendbr'
#define __noendbr       __attribute__((nocf_check))
                                       ^
2 warnings generated.
cannot attach kprobe, probe entry may not exist

aspanner, 2023-07-31:
> the following command gives list of attached `kprobes` in the kernel
> 
> ```
> sudo cat /sys/kernel/debug/kprobes/list
> ```
> 
> can you verify using this method, and paste the output ?

I've got the same output as you:
[root@y1 kprobes]# cat list 
ffffffff83151f70  k  finish_task_switch.isra.0+0x0    [FTRACE]

vimalk78, 2023-07-31:
your are on a higher kernel version though, `6.3` , i am on `6.1` . Perhaps there is some difference there.

vimalk78, 2023-07-31:
i think the issue solved by PR #848 , as this a temporary error. and kprobe is successfully attached.

jichenjc, 2023-07-31:
that's not fixed , it's just provide more info about the error itself to avoid potential confusion :)

vimalk78, 2023-07-31:
@jichenjc you mean attaching to `finish_task_switch.isra.0` can also fail?

jichenjc, 2023-07-31:
I mean the error will be printed anyhow, but additional logs from this PR will give us more hint on what's really doing..

vimalk78, 2023-07-31:
can we use this? 
```
$ grep finish_task_switch /proc/kallsyms 
0000000000000000 t finish_task_switch.isra.0
```

the file `/proc/kallsym` contains the exact symbol name. this way we can attach kprobe to the exact function without trying and failing.

from `man proc` :
```
       /proc/kallsyms (since Linux 2.5.71)
              This  holds  the  kernel  exported  symbol  definitions used by the modules(X) tools to dynamically link and bind loadable modules.  In Linux 2.5.47 and earlier, a similar file with slightly different syntax was named
              ksyms.

```

rootfs, 2023-07-31:
both kprobe function signatures are tried. The first error message means the `finish_task_switch` function is not found, then the `.isara.0` is tried. And that one gets to the log message
```
I0731 15:10:17.634387 323170 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=8 -DSET_GROUP_ID]
```

rootfs, 2023-07-31:
closed by #848

closed 2023-07-31 by rootfs