sustainable-computing-io/kepler #857: Verify energy accuracy

tobby-yuan, 2023-08-07:
### Which document would you like to be mentioned?

Hello everyone, I want to ask two questions.
1. Which model will be use for energy estimation if I only deploy [kepler using helm chart](https://github.com/sustainable-computing-io/kepler-helm-chart)?
2.  Is there any theorem or paper supporting this model can be used for energy estimation?

### Why is this needed?

I want to know the accuracy of energy estimation.

jichenjc, 2023-08-16:
@rootfs @marceloamaral I believe you have a set of papers about the model , maybe you can share or put into README (seems not yet?)

tobby-yuan, 2023-09-03:
@rootfs @marceloamaral @jichenjc Would you already put a set of papers about the model into README ?

closed 2023-11-09 by stale[bot]