sustainable-computing-io/kepler #860: Kepler cgroup metrics are zeroes

tobby-yuan, 2023-08-11:
### What happened?

Hi everyone,
This is my VM Environment
- kepler chart version: 0.5.0
- Ubuntu: 22.04.3 LTS
- Linux Kernel: 5.15.0-78-generic x86_64
- kubelet: 1.21.0
- cgroup: v2

But my some metrics(e.g. kepler_container_cgroupfs_cpu_usage_us_total, kepler_container_core_joules_total and so on) is alway 0. The following is my kepler log

```
I0810 13:36:11.444138       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0810 13:36:11.456218       1 exporter.go:155] Kepler running on version: a7a6cb1
I0810 13:36:11.456299       1 config.go:258] using gCgroup ID in the BPF program: true
I0810 13:36:11.456394       1 config.go:260] kernel version: 5.15
I0810 13:36:11.456441       1 exporter.go:179] EnabledBPFBatchDelete: true
I0810 13:36:11.456485       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0810 13:36:11.456556       1 power.go:64] Not able to obtain power, use estimate method
I0810 13:36:11.456585       1 redfish.go:169] failed to get redfish credential file path
I0810 13:36:11.456601       1 power.go:55] use acpi to obtain power
I0810 13:36:11.456790       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0810 13:36:11.467127       1 exporter.go:198] Initializing the GPU collector
I0810 13:36:17.469943       1 watcher.go:66] Using in cluster k8s config
I0810 13:36:17.571852       1 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
cannot attach kprobe, probe entry may not exist
I0810 13:36:18.414147       1 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]
I0810 13:36:18.440766       1 exporter.go:251] Started Kepler in 6.984594733s
```
It would affect estimate energy consumed(kepler_container_joules_total), right??

### What did you expect to happen?

 .

### How can we reproduce it (as minimally and precisely as possible)?

.

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jokuniew, 2023-08-11:
I have the same issue with **bare-metal** setup
ubuntu 22.04.1 LTS
Linux 5.15.0-43-generic
kind v0.20.0 go1.20.4 linux/amd64
kubectl v1.27.4
cgroup: v2
CRI:
RuntimeName:  containerd
RuntimeVersion:  1.6.22
all paths (/proc, /usr/src) mounted as in guide https://sustainable-computing.io/installation/local-cluster/

```
I0811 15:22:08.623322       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0811 15:22:08.665991       1 exporter.go:155] Kepler running on version: a7a6cb1
I0811 15:22:08.666020       1 config.go:258] using gCgroup ID in the BPF program: true
I0811 15:22:08.666055       1 config.go:260] kernel version: 5.15
I0811 15:22:08.666083       1 exporter.go:179] EnabledBPFBatchDelete: true
I0811 15:22:08.666175       1 power.go:53] use sysfs to obtain power
I0811 15:22:08.666195       1 redfish.go:169] failed to get redfish credential file path
I0811 15:22:08.666210       1 power.go:55] use acpi to obtain power
I0811 15:22:08.676428       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0811 15:22:08.793791       1 exporter.go:198] Initializing the GPU collector
I0811 15:22:14.799490       1 watcher.go:66] Using in cluster k8s config
I0811 15:22:14.901204       1 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
cannot attach kprobe, probe entry may not exist
I0811 15:22:15.761644       1 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=64 -DSET_GROUP_ID]
I0811 15:22:15.777384       1 exporter.go:251] Started Kepler in 7.111410022s
```

The log that concerns me the most is **`cannot attach kprobe, probe entry may not exist`**
I didn't have this error on ubuntu 20.04 with kernel 5.4.

I've seen in issues that @rootfs recommends to check `grep finish_task_switch /proc/kallsyms`
and with kernel 5.14 i have values 
```
ffffffffa96f38b0 t finish_task_switch.isra.0
ffffffffaa2ab721 t finish_task_switch.isra.0.cold
ffffffffc0d750d0 t bpf_prog_a4e3c0d94fadbd36_kprobe__finish_task_switch [bpf]
```
kepler didnt log the error (as I believe, it should) https://github.com/sustainable-computing-io/kepler/blob/main/pkg/bpfassets/attacher/bcc_attacher.go#L94

and finally.. attached or doesnt attached to the finish_task_switch?

tobby-yuan, 2023-08-11:
Hi @jokuniew, this is my log of checking  `grep finish_task_switch /proc/kallsyms`
```
ffffffff8aafbc40 t finish_task_switch.isra.0
ffffffff8b70bdb5 t finish_task_switch.isra.0.cold
ffffffffc0e987f0 t bpf_prog_a4e3c0d94fadbd36_kprobe__finish_task_switch [bpf]
```
This log is same as yours.
So, how can I solve this promblem or why did this problem happen?

rootfs, 2023-08-14:
can you try the libbpf image? Please change the kepler image from `latest` to `latest-libbpf`. You can do it by 

`kubectl edit -n kepler daemonset kepler-exporter`

jichenjc, 2023-08-15:
>>>Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=64 -DSET_GROUP_ID]

we saw this usually means the attach works? otherwise if the attach failed then we will not reach here ..

>>>But my some metrics(e.g. kepler_container_cgroupfs_cpu_usage_us_total, kepler_container_core_joules_total and so on) is alway 0

if you set kepler to --v 5 it might print some info which should be helpful to know why 0 is reported , whether it's work as design or something really wrong

tobby-yuan, 2023-08-15:
@jichenjc Do you mean that I should change loglevel of kepler to 5?

jichenjc, 2023-08-15:
yes, change to 5 it will print some logs and should have more info about the every 3 seconds report of the metrics including cgroup and eBPF , it might give you additional info on why it's 0

jokuniew, 2023-08-16:
> can you try the libbpf image? Please change the kepler image from `latest` to `latest-libbpf`. You can do it by
> 
> `kubectl edit -n kepler daemonset kepler-exporter`

@rootfs I've changed the image
ends up with panic
```
 kubectl logs -n kepler               kepler-nhl4b  
I0816 06:19:11.896894       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0816 06:19:11.930157       1 exporter.go:156] Kepler running on version: c2f4277
I0816 06:19:11.930175       1 config.go:263] using gCgroup ID in the BPF program: true
I0816 06:19:11.930205       1 config.go:265] kernel version: 5.15
I0816 06:19:11.930263       1 config.go:290] The Idle power will be exposed. Are you running on Baremetal or using single VM per node?
I0816 06:19:11.930269       1 exporter.go:181] EnabledBPFBatchDelete: true
I0816 06:19:11.930309       1 power.go:54] use sysfs to obtain power
I0816 06:19:11.930316       1 redfish.go:169] failed to get redfish credential file path
I0816 06:19:11.930321       1 power.go:56] use acpi to obtain power
I0816 06:19:11.937727       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0816 06:19:12.034630       1 container_energy.go:109] Using the Ratio/AbsPower Power Model to estimate Container Platform Power
I0816 06:19:12.034662       1 container_energy.go:118] Using the Ratio/AbsPower Power Model to estimate Container Component Power
I0816 06:19:12.034680       1 process_power.go:108] Using the Ratio/AbsPower Power Model to estimate Process Platform Power
I0816 06:19:12.034700       1 process_power.go:117] Using the Ratio/AbsPower Power Model to estimate Process Component Power
I0816 06:19:12.034999       1 node_platform_energy.go:52] Using the LinearRegressor/AbsModelWeight Power Model to estimate Node Platform Power
I0816 06:19:12.035093       1 exporter.go:204] Initializing the GPU collector
I0816 06:19:18.040743       1 watcher.go:66] Using in cluster k8s config
I0816 06:19:18.141471       1 bpf_perf.go:123] LibbpfBuilt: true, BccBuilt: false
libbpf: loading /var/lib/kepler/bpfassets/amd64_kepler.bpf.o
libbpf: elf: section(3) tracepoint/sched/sched_switch, size 2376, link 0, flags 6, type=1
libbpf: sec 'tracepoint/sched/sched_switch': found program 'kepler_trace' at insn offset 0 (0 bytes), code size 297 insns (2376 bytes)
libbpf: elf: section(4) .reltracepoint/sched/sched_switch, size 352, link 26, flags 40, type=9
libbpf: elf: section(5) tracepoint/irq/softirq_entry, size 144, link 0, flags 6, type=1
libbpf: sec 'tracepoint/irq/softirq_entry': found program 'kepler_irq_trace' at insn offset 0 (0 bytes), code size 18 insns (144 bytes)
libbpf: elf: section(6) .reltracepoint/irq/softirq_entry, size 16, link 26, flags 40, type=9
libbpf: elf: section(7) .maps, size 352, link 0, flags 3, type=1
libbpf: elf: section(8) license, size 4, link 0, flags 3, type=1
libbpf: license of /var/lib/kepler/bpfassets/amd64_kepler.bpf.o is GPL
libbpf: elf: section(17) .BTF, size 5838, link 0, flags 0, type=1
libbpf: elf: section(19) .BTF.ext, size 2072, link 0, flags 0, type=1
libbpf: elf: section(26) .symtab, size 984, link 1, flags 0, type=2
libbpf: looking for externs among 41 symbols...
libbpf: collected 0 externs total
libbpf: map 'processes': at sec_idx 7, offset 0.
libbpf: map 'processes': found type = 1.
libbpf: map 'processes': found key [6], sz = 4.
libbpf: map 'processes': found value [10], sz = 88.
libbpf: map 'processes': found max_entries = 10240.
libbpf: map 'pid_time': at sec_idx 7, offset 32.
libbpf: map 'pid_time': found type = 1.
libbpf: map 'pid_time': found key [6], sz = 4.
libbpf: map 'pid_time': found value [12], sz = 8.
libbpf: map 'pid_time': found max_entries = 10240.
libbpf: map 'cpu_cycles_hc_reader': at sec_idx 7, offset 64.
libbpf: map 'cpu_cycles_hc_reader': found type = 4.
libbpf: map 'cpu_cycles_hc_reader': found key [2], sz = 4.
libbpf: map 'cpu_cycles_hc_reader': found value [6], sz = 4.
libbpf: map 'cpu_cycles_hc_reader': found max_entries = 128.
libbpf: map 'cpu_cycles': at sec_idx 7, offset 96.
libbpf: map 'cpu_cycles': found type = 2.
libbpf: map 'cpu_cycles': found key [6], sz = 4.
libbpf: map 'cpu_cycles': found value [12], sz = 8.
libbpf: map 'cpu_cycles': found max_entries = 128.
libbpf: map 'cpu_ref_cycles_hc_reader': at sec_idx 7, offset 128.
libbpf: map 'cpu_ref_cycles_hc_reader': found type = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found key [2], sz = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found value [6], sz = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found max_entries = 128.
libbpf: map 'cpu_ref_cycles': at sec_idx 7, offset 160.
libbpf: map 'cpu_ref_cycles': found type = 2.
libbpf: map 'cpu_ref_cycles': found key [6], sz = 4.
libbpf: map 'cpu_ref_cycles': found value [12], sz = 8.
[... 78 lines cut ...]
libbpf: map 'cpu_ref_cycles_hc_reader': created successfully, fd=14
libbpf: map 'cpu_ref_cycles': created successfully, fd=15
libbpf: map 'cpu_instr_hc_reader': created successfully, fd=16
libbpf: map 'cpu_instr': created successfully, fd=17
libbpf: map 'cache_miss_hc_reader': created successfully, fd=18
libbpf: map 'cache_miss': created successfully, fd=19
libbpf: map 'cpu_freq_array': created successfully, fd=20
I0816 06:19:18.242250       1 libbpf_attacher.go:153] Successfully load eBPF module from libbpf object
I0816 06:19:18.265889       1 exporter.go:257] Started Kepler in 6.335731883s
[... cut ...]
        /opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/collector/metric_collector.go:120 +0x10b
github.com/sustainable-computing-io/kepler/pkg/manager.(*CollectorManager).Start.func1()
        /opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/manager/manager.go:72 +0x7b
created by github.com/sustainable-computing-io/kepler/pkg/manager.(*CollectorManager).Start
        /opt/app-root/src/github.com/sustainable-computing-io/kepler/pkg/manager/manager.go:64 +0x6a
```

jokuniew, 2023-08-16:
I need to add, that previously I've deployed kepler via **helm chart**. All reported errors are from helm deployment

When I deployed kepler with **manifests** ( make & apply) I've got right log message and metrics are valid

```
I0816 09:24:28.557873       1 watcher.go:66] Using in cluster k8s config
I0816 09:24:28.658788       1 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
cannot attach kprobe, probe entry may not exist
I0816 09:24:29.503997       1 bcc_attacher.go:94] attaching kprobe to finish_task_switch failed, trying finish_task_switch.isra.0 instead
I0816 09:24:29.522823       1 bcc_attacher.go:183] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=64 -DSET_GROUP_ID]
I0816 09:24:29.551285       1 exporter.go:257] Started Kepler in 7.069365222s
```

tobby-yuan, 2023-08-18:
> This is my log when change log level to 5

```
$ kubectl logs -n kepler        kepler-exporter-rjmb2 kepler-exporter
I0818 07:18:32.850191       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0818 07:18:32.855273       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0818 07:18:32.865892       1 exporter.go:158] Kepler running on version: b57ffa3
I0818 07:18:32.865922       1 config.go:267] using gCgroup ID in the BPF program: true
I0818 07:18:32.865980       1 config.go:269] kernel version: 5.15
I0818 07:18:32.867090       1 exporter.go:184] EnabledBPFBatchDelete: true
I0818 07:18:32.867110       1 config.go:140] ENABLE_EBPF_CGROUPID: true
I0818 07:18:32.867113       1 config.go:141] ENABLE_GPU: true
I0818 07:18:32.867159       1 config.go:142] ENABLE_QAT: false
I0818 07:18:32.867162       1 config.go:143] ENABLE_PROCESS_METRICS: false
I0818 07:18:32.867165       1 config.go:144] EXPOSE_HW_COUNTER_METRICS: false
I0818 07:18:32.867180       1 config.go:145] EXPOSE_CGROUP_METRICS: true
I0818 07:18:32.867183       1 config.go:146] EXPOSE_KUBELET_METRICS: true
I0818 07:18:32.867185       1 config.go:147] EXPOSE_IRQ_COUNTER_METRICS: true
I0818 07:18:32.867188       1 config.go:148] EXPOSE_ESTIMATED_IDLE_POWER_METRICS: false. This only impacts when the power is estimated using pre-prained models. Estimated idle power is meaningful only when Kepler is running on bare-metal or with a single virtual machine (VM) on the node.
I0818 07:18:32.867321       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0818 07:18:32.867436       1 power.go:71] Unable to obtain power, use estimate method
I0818 07:18:32.867557       1 redfish.go:169] failed to get redfish credential file path
I0818 07:18:32.867761       1 power.go:56] use acpi to obtain power
I0818 07:18:32.869199       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0818 07:18:32.869256       1 bpf_perf.go:81] hardeware counter metrics config false
I0818 07:18:32.869294       1 bpf_perf.go:83] hardeware counter metrics not enabled
I0818 07:18:32.869339       1 bpf_perf.go:99] irq counter metrics config true
I0818 07:18:32.892403       1 utils.go:56] Available ebpf metrics: [bpf_cpu_time_us bpf_net_tx_irq bpf_net_rx_irq bpf_block_irq]
I0818 07:18:32.892440       1 utils.go:57] Available counter metrics: []
I0818 07:18:32.892443       1 utils.go:58] Available cgroup metrics from cgroup: [cgroupfs_memory_usage_bytes cgroupfs_kernel_memory_usage_bytes cgroupfs_tcp_memory_usage_bytes cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us cgroupfs_ioread_bytes cgroupfs_iowrite_bytes block_devices_used]
I0818 07:18:32.892466       1 utils.go:59] Available cgroup metrics from kubelet: [kubelet_cpu_usage kubelet_memory_bytes]
I0818 07:18:32.892550       1 model.go:174] Model Config CONTAINER_TOTAL: {ModelType:Ratio ModelOutputType:AbsModelWeight TrainerName: EnergySource:acpi SelectFilter: InitModelURL: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0818 07:18:32.892581       1 model.go:94] Using Power Model Ratio
I0818 07:18:32.892591       1 container_energy.go:109] Using the Ratio/AbsModelWeight Power Model to estimate Container Platform Power
I0818 07:18:32.892597       1 model.go:174] Model Config CONTAINER_COMPONENTS: {ModelType:EstimatorSidecar ModelOutputType:AbsModelWeight TrainerName: EnergySource:rapl SelectFilter: InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentPower/CgroupOnly/ScikitMixed.zip IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0818 07:18:32.894319       1 estimate.go:140] estimator unmarshal error: json: cannot unmarshal array into Go struct field ComponentPowerResponse.powers of type map[string][]float64 ({"powers": [], "msg": "fail to handle request: __init__() got an unexpected keyword argument 'source'"})
I0818 07:18:32.894378       1 container_energy.go:120] Failed to create EstimatorSidecar/AbsModelWeight Power Model to estimate Container Component Power: json: cannot unmarshal array into Go struct field ComponentPowerResponse.powers of type map[string][]float64
I0818 07:18:32.894401       1 model.go:174] Model Config PROCESS_TOTAL: {ModelType:Ratio ModelOutputType:AbsModelWeight TrainerName: EnergySource:acpi SelectFilter: InitModelURL: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0818 07:18:32.894420       1 model.go:94] Using Power Model Ratio
I0818 07:18:32.894429       1 process_power.go:108] Using the Ratio/AbsModelWeight Power Model to estimate Process Platform Power
I0818 07:18:32.894436       1 model.go:174] Model Config PROCESS_COMPONENTS: {ModelType:Ratio ModelOutputType:AbsModelWeight TrainerName: EnergySource:rapl SelectFilter: InitModelURL: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0818 07:18:32.894447       1 model.go:94] Using Power Model Ratio
[... cut ...]
Energy stat: map[0:166626 (166626)] (0)Energy stat: map[0:499878 (499878)] (0)Energy stat: map[0:0 (0)] (0)Energy stat: map[0:5004 (5004)] (0)Energy stat: map[0:55542 (222168)] (0)Energy stat: map[0:55542 (555420)] (0)Energy stat: map[0:0 (0)] (0)Energy stat: map[0:1668 (6672)] (0)Energy stat: map[0:166626 (388794)] (0)Energy stat: map[0:499878 (1055298)] (0)Energy stat: map[0:0 (0)] (0)Energy stat: map[0:5004 (11676)] (0)Energy stat: map[0:55542 (444336)] (0)Energy stat: map[0:55542 (1110840)] (0)Energy stat: map[0:0 (0)] (0)Energy stat: map[0:1668 (13344)] (0)
```

tobby-yuan, 2023-08-18:
> yes, change to 5 it will print some logs and should have more info about the every 3 seconds report of the metrics including cgroup and eBPF , it might give you additional info on why it's 0

```

root@understudent:~# kubectl logs -n kepler        kepler-exporter-rjmb2 kepler-exporter
I0818 07:18:32.850191       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0818 07:18:32.855273       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0818 07:18:32.865892       1 exporter.go:158] Kepler running on version: b57ffa3
I0818 07:18:32.865922       1 config.go:267] using gCgroup ID in the BPF program: true
I0818 07:18:32.865980       1 config.go:269] kernel version: 5.15
I0818 07:18:32.867090       1 exporter.go:184] EnabledBPFBatchDelete: true
I0818 07:18:32.867110       1 config.go:140] ENABLE_EBPF_CGROUPID: true
I0818 07:18:32.867113       1 config.go:141] ENABLE_GPU: true
I0818 07:18:32.867159       1 config.go:142] ENABLE_QAT: false
I0818 07:18:32.867162       1 config.go:143] ENABLE_PROCESS_METRICS: false
I0818 07:18:32.867165       1 config.go:144] EXPOSE_HW_COUNTER_METRICS: false
I0818 07:18:32.867180       1 config.go:145] EXPOSE_CGROUP_METRICS: true
I0818 07:18:32.867183       1 config.go:146] EXPOSE_KUBELET_METRICS: true
I0818 07:18:32.867185       1 config.go:147] EXPOSE_IRQ_COUNTER_METRICS: true
I0818 07:18:32.867188       1 config.go:148] EXPOSE_ESTIMATED_IDLE_POWER_METRICS: false. This only impacts when the power is estimated using pre-prained models. Estimated idle power is meaningful only when Kepler is running on bare-metal or with a single virtual machine (VM) on the node.
I0818 07:18:32.867321       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0818 07:18:32.867436       1 power.go:71] Unable to obtain power, use estimate method
I0818 07:18:32.867557       1 redfish.go:169] failed to get redfish credential file path
I0818 07:18:32.867761       1 power.go:56] use acpi to obtain power
I0818 07:18:32.869199       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0818 07:18:32.869256       1 bpf_perf.go:81] hardeware counter metrics config false
I0818 07:18:32.869294       1 bpf_perf.go:83] hardeware counter metrics not enabled
I0818 07:18:32.869339       1 bpf_perf.go:99] irq counter metrics config true
I0818 07:18:32.892403       1 utils.go:56] Available ebpf metrics: [bpf_cpu_time_us bpf_net_tx_irq bpf_net_rx_irq bpf_block_irq]
I0818 07:18:32.892440       1 utils.go:57] Available counter metrics: []
I0818 07:18:32.892443       1 utils.go:58] Available cgroup metrics from cgroup: [cgroupfs_memory_usage_bytes cgroupfs_kernel_memory_usage_bytes cgroupfs_tcp_memory_usage_bytes cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us cgroupfs_ioread_bytes cgroupfs_iowrite_bytes block_devices_used]
I0818 07:18:32.892466       1 utils.go:59] Available cgroup metrics from kubelet: [kubelet_cpu_usage kubelet_memory_bytes]
I0818 07:18:32.892550       1 model.go:174] Model Config CONTAINER_TOTAL: {ModelType:Ratio ModelOutputType:AbsModelWeight TrainerName: EnergySource:acpi SelectFilter: InitModelURL: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0818 07:18:32.892581       1 model.go:94] Using Power Model Ratio
I0818 07:18:32.892591       1 container_energy.go:109] Using the Ratio/AbsModelWeight Power Model to estimate Container Platform Power
I0818 07:18:32.892597       1 model.go:174] Model Config CONTAINER_COMPONENTS: {ModelType:EstimatorSidecar ModelOutputType:AbsModelWeight TrainerName: EnergySource:rapl SelectFilter: InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentPower/CgroupOnly/ScikitMixed.zip IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0818 07:18:32.894319       1 estimate.go:140] estimator unmarshal error: json: cannot unmarshal array into Go struct field ComponentPowerResponse.powers of type map[string][]float64 ({"powers": [], "msg": "fail to handle request: __init__() got an unexpected keyword argument 'source'"})
I0818 07:18:32.894378       1 container_energy.go:120] Failed to create EstimatorSidecar/AbsModelWeight Power Model to estimate Container Component Power: json: cannot unmarshal array into Go struct field ComponentPowerResponse.powers of type map[string][]float64
I0818 07:18:32.894401       1 model.go:174] Model Config PROCESS_TOTAL: {ModelType:Ratio ModelOutputType:AbsModelWeight TrainerName: EnergySource:acpi SelectFilter: InitModelURL: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0818 07:18:32.894420       1 model.go:94] Using Power Model Ratio
I0818 07:18:32.894429       1 process_power.go:108] Using the Ratio/AbsModelWeight Power Model to estimate Process Platform Power
I0818 07:18:32.894436       1 model.go:174] Model Config PROCESS_COMPONENTS: {ModelType:Ratio ModelOutputType:AbsModelWeight TrainerName: EnergySource:rapl SelectFilter: InitModelURL: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
[... cut ...]
Energy stat: map[0:166626 (166626)] (0)Energy stat: map[0:499878 (499878)] (0)Energy stat: map[0:0 (0)] (0)Energy stat: map[0:5004 (5004)] (0)Energy stat: map[0:55542 (222168)] (0)Energy stat: map[0:55542 (555420)] (0)Energy stat: map[0:0 (0)] (0)Energy stat: map[0:1668 (6672)] (0)Energy stat: map[0:166626 (388794)] (0)Energy stat: map[0:499878 (1055298)] (0)Energy stat: map[0:0 (0)] (0)Energy stat: map[0:5004 (11676)] (0)Energy stat: map[0:55542 (444336)] (0)Energy stat: map[0:55542 (1110840)] (0)Energy stat: map[0:0 (0)] (0)Energy stat: map[0:1668 (13344)] (0)
```

rootfs, 2023-08-18:
@tobby-yuan the models are relocated, please do a fresh install with the latest manifests and kepler image.

tobby-yuan, 2023-08-19:
@rootfs  Hi, I do a fresh install. I use `git clone https://github.com/sustainable-computing-io/kepler.git` to git latest branch and install it. However, it appear image pull back off error

```
$kubectl describe pods -n kepler kepler-exporter-vvwng kepler-exporter
Name:         kepler-exporter-vvwng
Namespace:    kepler
Priority:     0
Node:         understudent/10.0.10.202
Start Time:   Sat, 19 Aug 2023 04:34:39 +0000
Labels:       app.kubernetes.io/component=exporter
              app.kubernetes.io/name=kepler-exporter
              controller-revision-hash=6cddd6d58c
              pod-template-generation=2
              sustainable-computing.io/app=kepler
Annotations:  <none>
Status:       Pending
IP:           10.244.0.57
IPs:
  IP:           10.244.0.57
Controlled By:  DaemonSet/kepler-exporter
Containers:
  kepler-exporter:
    Container ID:  docker://09923571a47e0877c5f76ad308e8427d85fd250debbf7944389c7a621d376489
    Image:         quay.io/sustainable_computing_io/kepler:latest-libbpf
    Image ID:      docker-pullable://quay.io/sustainable_computing_io/kepler@sha256:31e8e82dea60d0170d1f526da605eefe80a7e7a3e543c20e142154bef16368d0
    Port:          9102/TCP
    Host Port:     0/TCP
    Command:
      /bin/sh
      -c
    Args:
      until [ -e /tmp/estimator.sock ]; do sleep 1; done && /usr/bin/kepler -v=1 -kernel-source-dir=/usr/share/kepler/kernel_sources -redfish-cred-file-path=/etc/redfish/redfish.csv
    State:          Running
      Started:      Sat, 19 Aug 2023 04:34:42 +0000
    Ready:          True
    Restart Count:  0
    Requests:
      cpu:     100m
      memory:  400Mi
    Liveness:  http-get http://:9102/healthz delay=10s timeout=10s period=60s #success=1 #failure=5
    Environment:
      NODE_IP:     (v1:status.hostIP)
      NODE_NAME:   (v1:spec.nodeName)
    Mounts:
      /etc/kepler/kepler.config from cfm (ro)
      /etc/redfish from redfish (ro)
      /lib/modules from lib-modules (rw)
      /proc from proc (rw)
      /sys from tracing (rw)
      /tmp from tmp (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-sk86l (ro)
  estimator:
    Container ID:
    Image:         kepler_model_server
    Image ID:
    Port:          <none>
    Host Port:     <none>
    Command:
      python3.8
    Args:
      -u
      src/estimate/estimator.py
    State:          Waiting
      Reason:       ErrImagePull
    Ready:          False
    Restart Count:  0
    Environment:    <none>
    Mounts:
      /etc/kepler/kepler.config from cfm (ro)
      /tmp from tmp (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-sk86l (ro)
Conditions:
  Type              Status
  Initialized       True
  Ready             False
  ContainersReady   False
  PodScheduled      True
Volumes:
  tmp:
    Type:       EmptyDir (a temporary directory that shares a pod's lifetime)
    Medium:
    SizeLimit:  <unset>
  lib-modules:
    Type:          HostPath (bare host directory volume)
    Path:          /lib/modules
    HostPathType:  Directory
  tracing:
    Type:          HostPath (bare host directory volume)
    Path:          /sys
    HostPathType:  Directory
  proc:
    Type:          HostPath (bare host directory volume)
    Path:          /proc
    HostPathType:  Directory
  cfm:
    Type:      ConfigMap (a volume populated by a ConfigMap)
    Name:      kepler-cfm
    Optional:  false
  redfish:
    Type:        Secret (a volume populated by a Secret)
    SecretName:  redfish-4kh9d7bc7m
    Optional:    false
  kube-api-access-sk86l:
    Type:                    Projected (a volume that contains injected data from multiple sources)
    TokenExpirationSeconds:  3607
    ConfigMapName:           kube-root-ca.crt
    ConfigMapOptional:       <nil>
    DownwardAPI:             true
QoS Class:                   Burstable
Node-Selectors:              <none>
Tolerations:                 node-role.kubernetes.io/master:NoSchedule
                             node.kubernetes.io/disk-pressure:NoSchedule op=Exists
                             node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists
                             node.kubernetes.io/pid-pressure:NoSchedule op=Exists
                             node.kubernetes.io/unreachable:NoExecute op=Exists
                             node.kubernetes.io/unschedulable:NoSchedule op=Exists
Events:
  Type     Reason     Age                From               Message
  ----     ------     ----               ----               -------
  Normal   Scheduled  46s                default-scheduler  Successfully assigned kepler/kepler-exporter-vvwng to understudent
  Normal   Pulling    46s                kubelet            Pulling image "quay.io/sustainable_computing_io/kepler:latest-libbpf"
  Normal   Pulled     43s                kubelet            Successfully pulled image "quay.io/sustainable_computing_io/kepler:latest-libbpf" in 2.477170075s
  Normal   Created    43s                kubelet            Created container kepler-exporter
  Normal   Started    43s                kubelet            Started container kepler-exporter
  Normal   Pulling    28s (x2 over 43s)  kubelet            Pulling image "kepler_model_server"
  Warning  Failed     26s (x2 over 41s)  kubelet            Failed to pull image "kepler_model_server": rpc error: code = Unknown desc = Error response from daemon: pull access denied for kepler_model_server, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
  Warning  Failed     26s (x2 over 41s)  kubelet            Error: ErrImagePull
  Normal   BackOff    10s (x2 over 40s)  kubelet            Back-off pulling image "kepler_model_server"
  Warning  Failed     10s (x2 over 40s)  kubelet            Error: ImagePullBackOff
Error from server (NotFound): pods "kepler-exporter" not found
```

tobby-yuan, 2023-08-19:
Hi @rootfs, I see the document of [kepler](https://sustainable-computing.io/). The deployment scenario is deleted so the sidecar scenario is not avaliable now?

Moreover, I do a fresh install through git the latest branch of kepler `git clone https://github.com/sustainable-computing-io/kepler.git`.
Then, I make kepler using `make build-manifest OPTS="MODEL_SERVER_DEPLOY"`.  I find the pod of kepler model server have the problem of ImagePullBackOff. 
The following is the description of kepler model server
```
kubectl describe pods -n kepler kepler-model-server-6df95dbd98-c7bf2
Name:         kepler-model-server-6df95dbd98-c7bf2
Namespace:    kepler
Priority:     0
Node:         understudent/10.0.10.202
Start Time:   Sat, 19 Aug 2023 05:03:44 +0000
Labels:       app.kubernetes.io/component=model-server
              app.kubernetes.io/name=kepler-model-server
              pod-template-hash=6df95dbd98
              sustainable-computing.io/app=kepler
Annotations:  <none>
Status:       Pending
IP:           10.244.0.60
IPs:
  IP:           10.244.0.60
Controlled By:  ReplicaSet/kepler-model-server-6df95dbd98
Containers:
  server-api:
    Container ID:
    Image:         kepler_model_server
    Image ID:
    Port:          8100/TCP
    Host Port:     0/TCP
    Command:
      python3.8
    Args:
      -u
      src/server/model_server.py
    State:          Waiting
      Reason:       ImagePullBackOff
    Ready:          False
    Restart Count:  0
    Environment:    <none>
    Mounts:
      /data from model-data (rw)
      /etc/kepler/kepler.config from cfm (ro)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-g8n4g (ro)
Conditions:
  Type              Status
  Initialized       True
  Ready             False
  ContainersReady   False
  PodScheduled      True
Volumes:
  cfm:
    Type:      ConfigMap (a volume populated by a ConfigMap)
    Name:      kepler-model-server-cfm
    Optional:  false
  model-data:
    Type:       EmptyDir (a temporary directory that shares a pod's lifetime)
    Medium:
    SizeLimit:  <unset>
  kube-api-access-g8n4g:
    Type:                    Projected (a volume that contains injected data from multiple sources)
    TokenExpirationSeconds:  3607
    ConfigMapName:           kube-root-ca.crt
    ConfigMapOptional:       <nil>
    DownwardAPI:             true
QoS Class:                   BestEffort
Node-Selectors:              <none>
Tolerations:                 node.kubernetes.io/not-ready:NoExecute op=Exists for 300s
                             node.kubernetes.io/unreachable:NoExecute op=Exists for 300s
Events:
  Type     Reason     Age                From               Message
[... 27 lines cut ...]
I0819 05:03:50.787850       1 process_power.go:117] Using the Ratio/AbsModelWeight Power Model to estimate Process Component Power
I0819 05:03:50.803348       1 node_platform_energy.go:52] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power
I0819 05:03:50.809533       1 node_component_energy.go:56] Using the LinearRegressor/AbsPower Power Model to estimate Node Component Power
I0819 05:03:50.809552       1 exporter.go:207] Initializing the GPU collector
I0819 05:03:56.810069       1 watcher.go:66] Using in cluster k8s config
I0819 05:03:56.911214       1 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.15.0-78-generic
chdir(/lib/modules/5.15.0-78-generic/build): No such file or directory
I0819 05:03:56.916961       1 bcc_attacher.go:80] failed to attach the bpf program: <nil>
I0819 05:03:56.916991       1 bcc_attacher.go:155] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, from default kernel source.
I0819 05:03:56.917017       1 bcc_attacher.go:158] trying to load eBPF module with kernel source dir /usr/share/kepler/kernel_sources/4.18.0-477.13.1.el8_8.x86_64
cannot attach kprobe, probe entry may not exist
I0819 05:03:57.747601       1 bcc_attacher.go:94] attaching kprobe to finish_task_switch failed, trying finish_task_switch.isra.0 instead
I0819 05:03:57.764705       1 bcc_attacher.go:164] Successfully loaded eBPF module with options: [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID] from kernel source "/usr/share/kepler/kernel_sources/4.18.0-477.13.1.el8_8.x86_64"
I0819 05:03:57.764746       1 bcc_attacher.go:183] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]
I0819 05:03:57.781869       1 node_platform_energy.go:81] Failed to get node platform power model Weight for model type AbsPower is not valid: &map[core:{{0 map[] map[kubelet_cpu_usage:{7.027426160337554 103.57941658210045 0} kubelet_memory_bytes:{417264.87763713085 2.0114948671646638e+12 0}]}} dram:{{2.6353736295072387 map[] map[kubelet_cpu_usage:{7.027426160337554 103.57941658210045 1.33012561695696} kubelet_memory_bytes:{417264.87763713085 2.0114948671646638e+12 0.16372489537216126}]}} package:{{140.8809244744518 map[] map[kubelet_cpu_usage:{7.027426160337554 103.57941658210045 124.69008698216625} kubelet_memory_bytes:{417264.87763713085 2.0114948671646638e+12 -2.582378633216328}]}} uncore:{{0 map[] map[kubelet_cpu_usage:{7.027426160337554 103.57941658210045 0} kubelet_memory_bytes:{417264.87763713085 2.0114948671646638e+12 0}]}}]
[... cut ...]
I0819 05:04:03.793515       1 node_platform_energy.go:81] Failed to get node platform power model Weight for model type AbsPower is not valid: &map[core:{{0 map[] map[kubelet_cpu_usage:{7.027426160337554 103.57941658210045 0} kubelet_memory_bytes:{417264.87763713085 2.0114948671646638e+12 0}]}} dram:{{2.6353736295072387 map[] map[kubelet_cpu_usage:{7.027426160337554 103.57941658210045 1.33012561695696} kubelet_memory_bytes:{417264.87763713085 2.0114948671646638e+12 0.16372489537216126}]}} package:{{140.
```

Lai-Kenny, 2023-08-19:
Hi, everyone. I found `make build-manifest OPTS="ESTIMATOR_SIDECAR_DEPLOY"` and `make build-manifest OPTS="MODEL_SERVER_DEPLOY"`. They seem to have switched to model server, right?

sunya-ch, 2023-09-26:
@Lai-Kenny I think it might be related to the issue fixed by https://github.com/sustainable-computing-io/kepler/pull/891. 
Could you refetch the upstream repo and try again?
If it is still there, could you share the result of 

```
# for model server case
kubectl get deploy -n kepler -oyaml
# for sidecar case
kubectl get ds -n kepler -oyaml
```

kgamanji, 2023-10-16:
This might be helpful to someone who is testing Kepler locally on their kid cluster:

I have set up Kepler on a kind cluster on my machine.  Initially, all the metrics were set to zero. Used the troubleshooting guide and checked I was using cgroup v2, all good on this side. Fixed it by upgrading the Kepler image to pull `latest-libbpf` tag.

sunya-ch, 2023-12-19:
Shall we close the issue?

closed 2024-02-24 by stale[bot]