sustainable-computing-io/kepler #861: kepler rpm start issues

aspanner, 2023-08-12:
### What happened?

rpm eleast 0.5.3 installed
starting kepler on the CLI or via systemctl start kepler show the same error message:

kepler.service - Kepler
     Loaded: loaded (/usr/lib/systemd/system/kepler.service; disabled; preset: disabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Fri 2023-08-11 13:14:18 AEST; 1 day 1h ago
   Main PID: 254003 (kepler)
      Tasks: 12 (limit: 38092)
     Memory: 72.7M
        CPU: 20.917s
     CGroup: /system.slice/kepler.service
             └─254003 kepler

Aug 11 13:14:18 y1 kepler[254003]: arch/x86/include/asm/ibt.h:78:8: warning: 'nocf_check' attribute ignored; use -fcf-protection to enable the attribute [-Wignored-attributes]
Aug 11 13:14:18 y1 kepler[254003]: extern __noendbr void ibt_restore(u64 save);
Aug 11 13:14:18 y1 kepler[254003]:        ^
Aug 11 13:14:18 y1 kepler[254003]: arch/x86/include/asm/ibt.h:32:34: note: expanded from macro '__noendbr'
Aug 11 13:14:18 y1 kepler[254003]: #define __noendbr       __attribute__((nocf_check))
Aug 11 13:14:18 y1 kepler[254003]:                                        ^
Aug 11 13:14:19 y1 kepler[254003]: 2 warnings generated.
Aug 11 13:14:19 y1 kepler[254003]: cannot attach kprobe, probe entry may not exist
Aug 11 13:14:19 y1 kepler[254003]: I0811 13:14:19.320129  254003 bcc_attacher.go:186] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=8 -DSET_GROUP_ID]
Aug 11 13:14:19 y1 kepler[254003]: I0811 13:14:19.320393  254003 exporter.go:251] Started Kepler in 846.78589ms

### What did you expect to happen?

I expected to not see an issues attacking a kernel probe, as I am expecting this is fundamental to get power metrics.

### How can we reproduce it (as minimally and precisely as possible)?

Follow the kepler rpm installation instructions, use release 0.5.3 and start kepler either via cli or systemctl start kepler

### Anything else we need to know?

localhost:8888/metrics is showing metrics.

# HELP go_gc_duration_seconds A summary of the pause duration of garbage collection cycles.
# TYPE go_gc_duration_seconds summary
go_gc_duration_seconds{quantile="0"} 1.676e-05
go_gc_duration_seconds{quantile="0.25"} 2.9869e-05
go_gc_duration_seconds{quantile="0.5"} 4.697e-05
go_gc_duration_seconds{quantile="0.75"} 6.5199e-05
go_gc_duration_seconds{quantile="1"} 0.001010315
go_gc_duration_seconds_sum 0.045946091
go_gc_duration_seconds_count 930
# HELP go_goroutines Number of goroutines that currently exist.
# TYPE go_goroutines gauge
go_goroutines 139
# HELP go_info Information about the Go environment.
# TYPE go_info gauge
go_info{version="go1.18.1"} 1
# HELP go_memstats_alloc_bytes Number of bytes allocated and still in use.
# TYPE go_memstats_alloc_bytes gauge
go_memstats_alloc_bytes 5.898624e+06
# HELP go_memstats_alloc_bytes_total Total number of bytes allocated, even if freed.
# TYPE go_memstats_alloc_bytes_total counter
go_memstats_alloc_bytes_total 2.086199544e+09
# HELP go_memstats_buck_hash_sys_bytes Number of bytes used by the profiling bucket hash table.
# TYPE go_memstats_buck_hash_sys_bytes gauge
go_memstats_buck_hash_sys_bytes 1.528293e+06
# HELP go_memstats_frees_total Total number of frees.
# TYPE go_memstats_frees_total counter
go_memstats_frees_total 2.3230602e+07
# HELP go_memstats_gc_sys_bytes Number of bytes used for garbage collection system metadata.
# TYPE go_memstats_gc_sys_bytes gauge
go_memstats_gc_sys_bytes 5.3818e+06
[... 124 lines cut ...]

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here
no command arguments used
</details>

### Container runtime (CRI) and version (if applicable)

<details>
n/a
</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>
n/a
</details>

rootfs, 2023-08-14:
@aspanner isn't it the same as #847

jichenjc, 2023-08-15:
yes, `rpm eleast 0.5.3 installed` doesn't have #847 log
I think this warning can be ignored, any functional impact you saw?

closed 2023-10-21 by stale[bot]