sustainable-computing-io/kepler #864: Cann't see metric from port 9102

tobby-yuan, 2023-08-14:
### What happened?

```
$ kubectl logs -n kepler kepler-exporter-jsl4b kepler-exporter
I0814 09:07:22.514442       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0814 09:07:22.557394       1 exporter.go:156] Kepler running on version: 315b519
I0814 09:07:22.557644       1 config.go:263] using gCgroup ID in the BPF program: true
I0814 09:07:22.557830       1 config.go:265] kernel version: 5.15
I0814 09:07:22.557994       1 exporter.go:181] EnabledBPFBatchDelete: true
I0814 09:07:22.558099       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0814 09:07:22.558405       1 power.go:71] Unable to obtain power, use estimate method
I0814 09:07:22.558544       1 redfish.go:169] failed to get redfish credential file path
I0814 09:07:22.558611       1 power.go:56] use acpi to obtain power
I0814 09:07:22.558993       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0814 09:07:22.594274       1 container_energy.go:109] Using the Ratio/AbsPower Power Model to estimate Container Platform Power
I0814 09:07:23.075697       1 container_energy.go:120] Failed to create EstimatorSidecar/DynComponentPower Power Model to estimate Container Component Power: json: cannot unmarshal array into Go struct field ComponentPowerResponse.powers of type map[string][]float64
I0814 09:07:23.075748       1 process_power.go:108] Using the Ratio/AbsPower Power Model to estimate Process Platform Power
I0814 09:07:23.075765       1 process_power.go:117] Using the Ratio/AbsPower Power Model to estimate Process Component Power
I0814 09:07:23.076352       1 node_platform_energy.go:52] Using the LinearRegressor/AbsModelWeight Power Model to estimate Node Platform Power
I0814 09:07:23.076502       1 node_component_energy.go:56] Using the LinearRegressor/AbsComponentModelWeight Power Model to estimate Node Component Power
I0814 09:07:23.076512       1 exporter.go:204] Initializing the GPU collector
I0814 09:07:29.078878       1 watcher.go:66] Using in cluster k8s config
I0814 09:07:29.179632       1 bpf_perf.go:123] LibbpfBuilt: false, BccBuilt: true
modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.15.0-78-generic
chdir(/lib/modules/5.15.0-78-generic/build): No such file or directory
I0814 09:07:29.190394       1 bcc_attacher.go:80] failed to attach the bpf program: <nil>
I0814 09:07:29.190477       1 bcc_attacher.go:155] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, from default kernel source.
I0814 09:07:29.190569       1 bcc_attacher.go:170] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, not able to load eBPF modules
I0814 09:07:29.190608       1 exporter.go:230] failed to start : failed to attach bpf assets: failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, not able to load eBPF modules
I0814 09:07:29.190837       1 exporter.go:257] Started Kepler in 6.633837734s
$ kubectl logs -n kepler kepler-exporter-jsl4b estimator
2023-08-14 09:07:20.240218: W tensorflow/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory
2023-08-14 09:07:20.240274: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
```

### What did you expect to happen?

`http://localhost:9102/metrics` need to show metrics

### How can we reproduce it (as minimally and precisely as possible)?

This is my VM Environment
- Ubuntu: 22.04.3 LTS
- Linux Kernel: 5.15.0-78-generic x86_64
- kubelet: 1.21.0
- cgroup: v2

This is my insatllation step:
`git clone https://github.com/sustainable-computing-io/kepler.git`

`make build-manifest OPTS="ESTIMATOR_SIDECAR_DEPLOY"`

`kubectl apply -f _output/generated-manifest/deployment.yaml`

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

rootfs, 2023-08-14:
ebpf not started, that looks the root cause
```
I0814 09:07:29.190477       1 bcc_attacher.go:155] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, from default kernel source.
I0814 09:07:29.190569       1 bcc_attacher.go:170] failed to attach perf module with options [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, not able to load eBPF modules
```

@tobby-yuan can you try the libbpf image? Please change the kepler image from `latest` to `latest-libbpf`. You can do it by `kubectl edit -n kepler daemonset kepler-exporter`

tobby-yuan, 2023-08-14:
@rootfs Thanks for your reply. I change kepler image from `latest` to `latest-libbpf`.
This is log of kepler, is it work?
```
$ kubectl logs -n kepler        kepler-exporter-bxh48  kepler-exporter
I0814 13:59:55.671043       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0814 13:59:55.682049       1 exporter.go:156] Kepler running on version: 301f96b
I0814 13:59:55.682068       1 config.go:263] using gCgroup ID in the BPF program: true
I0814 13:59:55.682100       1 config.go:265] kernel version: 5.15
I0814 13:59:55.682128       1 exporter.go:181] EnabledBPFBatchDelete: true
I0814 13:59:55.682143       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0814 13:59:55.682177       1 power.go:71] Unable to obtain power, use estimate method
I0814 13:59:55.682185       1 redfish.go:169] failed to get redfish credential file path
I0814 13:59:55.682189       1 power.go:56] use acpi to obtain power
I0814 13:59:55.682331       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0814 13:59:55.688077       1 container_energy.go:109] Using the Ratio/AbsPower Power Model to estimate Container Platform Power
I0814 13:59:56.091302       1 container_energy.go:118] Using the EstimatorSidecar/DynComponentPower Power Model to estimate Container Component Power
I0814 13:59:56.091380       1 process_power.go:108] Using the Ratio/AbsPower Power Model to estimate Process Platform Power
I0814 13:59:56.091397       1 process_power.go:117] Using the Ratio/AbsPower Power Model to estimate Process Component Power
I0814 13:59:56.095246       1 node_platform_energy.go:52] Using the LinearRegressor/AbsModelWeight Power Model to estimate Node Platform Power
I0814 13:59:56.095578       1 node_component_energy.go:56] Using the LinearRegressor/AbsComponentModelWeight Power Model to estimate Node Component Power
I0814 13:59:56.095639       1 exporter.go:204] Initializing the GPU collector
I0814 14:00:02.101263       1 watcher.go:66] Using in cluster k8s config
I0814 14:00:02.203023       1 bpf_perf.go:123] LibbpfBuilt: true, BccBuilt: false
libbpf: loading /var/lib/kepler/bpfassets/amd64_kepler.bpf.o
libbpf: elf: section(3) tracepoint/sched/sched_switch, size 2352, link 0, flags 6, type=1
libbpf: sec 'tracepoint/sched/sched_switch': found program 'kepler_trace' at insn offset 0 (0 bytes), code size 294 insns (2352 bytes)
libbpf: elf: section(4) .reltracepoint/sched/sched_switch, size 352, link 26, flags 40, type=9
libbpf: elf: section(5) tracepoint/irq/softirq_entry, size 144, link 0, flags 6, type=1
libbpf: sec 'tracepoint/irq/softirq_entry': found program 'kepler_irq_trace' at insn offset 0 (0 bytes), code size 18 insns (144 bytes)
libbpf: elf: section(6) .reltracepoint/irq/softirq_entry, size 16, link 26, flags 40, type=9
libbpf: elf: section(7) .maps, size 352, link 0, flags 3, type=1
libbpf: elf: section(8) license, size 4, link 0, flags 3, type=1
libbpf: license of /var/lib/kepler/bpfassets/amd64_kepler.bpf.o is GPL
libbpf: elf: section(17) .BTF, size 5820, link 0, flags 0, type=1
libbpf: elf: section(19) .BTF.ext, size 2056, link 0, flags 0, type=1
libbpf: elf: section(26) .symtab, size 984, link 1, flags 0, type=2
libbpf: looking for externs among 41 symbols...
libbpf: collected 0 externs total
libbpf: map 'processes': at sec_idx 7, offset 0.
libbpf: map 'processes': found type = 1.
libbpf: map 'processes': found key [6], sz = 4.
libbpf: map 'processes': found value [10], sz = 88.
libbpf: map 'processes': found max_entries = 10240.
libbpf: map 'pid_time': at sec_idx 7, offset 32.
libbpf: map 'pid_time': found type = 1.
libbpf: map 'pid_time': found key [6], sz = 4.
libbpf: map 'pid_time': found value [12], sz = 8.
libbpf: map 'pid_time': found max_entries = 10240.
libbpf: map 'cpu_cycles_hc_reader': at sec_idx 7, offset 64.
libbpf: map 'cpu_cycles_hc_reader': found type = 4.
libbpf: map 'cpu_cycles_hc_reader': found key [2], sz = 4.
libbpf: map 'cpu_cycles_hc_reader': found value [6], sz = 4.
libbpf: map 'cpu_cycles_hc_reader': found max_entries = 128.
libbpf: map 'cpu_cycles': at sec_idx 7, offset 96.
libbpf: map 'cpu_cycles': found type = 2.
libbpf: map 'cpu_cycles': found key [6], sz = 4.
libbpf: map 'cpu_cycles': found value [12], sz = 8.
libbpf: map 'cpu_cycles': found max_entries = 128.
libbpf: map 'cpu_ref_cycles_hc_reader': at sec_idx 7, offset 128.
libbpf: map 'cpu_ref_cycles_hc_reader': found type = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found key [2], sz = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found value [6], sz = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found max_entries = 128.
libbpf: map 'cpu_ref_cycles': at sec_idx 7, offset 160.
libbpf: map 'cpu_ref_cycles': found type = 2.
libbpf: map 'cpu_ref_cycles': found key [6], sz = 4.
libbpf: map 'cpu_ref_cycles': found value [12], sz = 8.
libbpf: map 'cpu_ref_cycles': found max_entries = 128.
libbpf: map 'cpu_instr_hc_reader': at sec_idx 7, offset 192.
libbpf: map 'cpu_instr_hc_reader': found type = 4.
[... 69 lines cut ...]
libbpf: sec '.reltracepoint/irq/softirq_entry': relo #0: insn #5 against 'processes'
libbpf: prog 'kepler_irq_trace': found map 0 (processes, sec 7, off 0) for insn #5
libbpf: map 'processes': created successfully, fd=10
libbpf: map 'pid_time': created successfully, fd=11
libbpf: map 'cpu_cycles_hc_reader': created successfully, fd=12
[... cut ...]
I0814 14:00:02.207110       1 libbpf_attacher.go:153] Successfully load eBPF module from libbpf object
I0814 14:00:02.219268       1 exporter.go:257] Started Kepler in 6.537229901s
```

Moreover, I can get metrics inside kepler pod.
`kubectl exec -ti -n kepler kepler-exporter -- bash  -c "curl localhost:9102/metrics"| grep kepler_container_core_joules_total `

However, I can't get metrics ouside kepler pod.
```
$ curl http://localhost:9102//metrics
curl: (7) Failed to connect to localhost port 9102 after 0 ms: Connection refused
```

rootfs, 2023-08-14:
@tobby-yuan cool! that's a good sign of progress in your environment. Depending on your network configuration, the kepler service endpoint may not be accessible on your local host, `kubectl exec -ti -n kepler kepler-exporter -- bash  -c "curl localhost:9102/metrics"` is the best way to get a quick output for sanity test.

closed 2023-08-14 by rootfs

tobby-yuan, 2023-08-14:
@rootfs I need to get metrics ouside kepler pod. Would you have any suggestion? Many thanks!

rootfs, 2023-08-14:
@tobby-yuan you need prometheus to scrape the metrics, then you can query prometheus to get the metrics