sustainable-computing-io/kepler #866: An error occurred while executing make build_containerized

Yanbo0101, 2023-08-15:
### What happened?

Hi, when I git clone the latest keper and execute make build_containerized, I encounter this error.

`make build_containerized

GO111MODULE=off go get -u github.com/go-bindata/go-bindata/...
./hack/bindata.sh
go mod tidy -v
go mod vendor
./automation/presubmit-tests/gofmt.sh
Checking go format
echo BIN_TIMESTAMP==2023-08-15T13:10:49Z
BIN_TIMESTAMP==2023-08-15T13:10:49Z
/usr/bin/docker build -t yanbo0101/kepler:v0.5.3-35-g1b3b5be-linux-amd64 \
        -f /home/yanbo/Project/kepler/build/Dockerfile \
        --network host \
        --build-arg SOURCE_GIT_TAG=v0.5.3-35-g1b3b5be \
        --build-arg BIN_TIMESTAMP=2023-08-15T13:10:49Z \
        --platform="linux/amd64" \
        .
[+] Building 2.8s (17/26)
 => [internal] load .dockerignore                                                                                                                                          0.0s
 => => transferring context: 2B                                                                                                                                            0.0s
 => [internal] load build definition from Dockerfile                                                                                                                       0.0s
 => => transferring dockerfile: 1.82kB                                                                                                                                     0.0s
 => [internal] load metadata for quay.io/sustainable_computing_io/kepler_kernel_source_images:ubi9                                                                         1.3s
 => [internal] load metadata for quay.io/sustainable_computing_io/kepler_builder:ubi-9-bcc-0.26-go1.18                                                                     1.3s
 => [internal] load metadata for quay.io/sustainable_computing_io/kepler_kernel_source_images:ubi8                                                                         1.3s
 => [internal] load metadata for quay.io/sustainable_computing_io/kepler_base:ubi-9-bcc-0.26                                                                               1.3s
 => FROM quay.io/sustainable_computing_io/kepler_kernel_source_images:ubi9@sha256:1029f8eeb3cf59486d819116a5883567ed366376e839f51d1796d480a2562eab                         0.0s
 => [builder 1/6] FROM quay.io/sustainable_computing_io/kepler_builder:ubi-9-bcc-0.26-go1.18@sha256:824d2613bee945fa5f4facdc2abd749831cdeca89c613947840496036910b04f       0.0s
 => ERROR https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/AbsComponentModelWeight/Full/KerasCompWeightFullPipeline/  1.3s
 => FROM quay.io/sustainable_computing_io/kepler_kernel_source_images:ubi8@sha256:58068024facac79ce49d0ed4029d667a194c601fedabbdf711a5b4acbeacbfbf                         0.0s
 => [internal] load build context                                                                                                                                          0.4s
 => => transferring context: 67.46MB                                                                                                                                       0.4s
 => ERROR https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixe  1.3s
 => [stage-1 1/9] FROM quay.io/sustainable_computing_io/kepler_base:ubi-9-bcc-0.26@sha256:dc90a1fe63bed297f763d51dc4f08f5ae0a8d5d2c40275ef794ffd80b5454f30                 0.0s
 => CACHED [builder 2/6] WORKDIR /opt/app-root/src/github.com/sustainable-computing-io/kepler                                                                              0.0s
 => [builder 3/6] COPY . .                                                                                                                                                 0.7s
 => [builder 4/6] RUN mkdir -p data                                                                                                                                        0.1s
 => CANCELED [builder 5/6] RUN make build SOURCE_GIT_TAG=v0.5.3-35-g1b3b5be BIN_TIMESTAMP=2023-08-15T13:10:49Z                                                             0.1s
------
 > https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/AbsComponentModelWeight/Full/KerasCompWeightFullPipeline/KerasCompWeightFullPipeline.json:
------
------
 > https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json:
------
ERROR: failed to solve: failed to load cache key: invalid response status 404
`
There are no relevant files in the sustainable-computing-io/kepler-model-server/main/tests/, but they are still added to Kepler's dockerfile
`
ADD https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json /var/lib/kepler/data/ScikitMixed.json

ADD https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/AbsComponentModelWeight/Full/KerasCompWeightFullPipeline/KerasCompWeightFullPipeline.json /var/lib/kepler/data/KerasCompWeightFullPipeline.json
`

### What did you expect to happen?

.

### How can we reproduce it (as minimally and precisely as possible)?

.

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>
.

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

marceloamaral, 2023-08-15:
@sunya-ch were these files removed? ^^

sunya-ch, 2023-08-15:
Yes, deprecated now.

Need PR to change to deprecated branch. 

```
ADD https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/deprecated/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json /var/lib/kepler/data/ScikitMixed.json

ADD https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/deprecated/tests/test_models/AbsComponentModelWeight/Full/KerasCompWeightFullPipeline/KerasCompWeightFullPipeline.json /var/lib/kepler/data/KerasCompWeightFullPipeline.json
```

rootfs, 2023-08-15:
I created a branch that keeps the tests models
https://github.com/sustainable-computing-io/kepler-model-server/tree/test_models

PR #868 (2023-08-15): image: retrieve pretrained models from test_models branch
fix #866 #867
closed 2023-08-15 by husky-parul