sustainable-computing-io/kepler #888: Failed to pull image "kepler_model_server"

tobby-yuan, 2023-08-25:
### What happened?

Hi everone, I install kepler using Manifests and option `ESTIMATOR_SIDECAR_DEPLOY`(command: `make build-manifest OPTS="ESTIMATOR_SIDECAR_DEPLOY"`).
Then, I deploy kepler(command: `kubectl apply -f _output/generated-manifest/deployment.yaml`).
However, I encountered and following error
```
# kubectl get pods -A
NAMESPACE     NAME                                                         READY   STATUS         RESTARTS   AGE
kepler        kepler-exporter-9s4bk                                        1/2     ErrImagePull   0          18s
```

The following is description of this pod:
```
# kubectl describe pods -n kepler        kepler-exporter-9s4bk
Name:         kepler-exporter-9s4bk
Namespace:    kepler
Priority:     0
Node:         understudent/10.0.10.202
Start Time:   Fri, 25 Aug 2023 04:37:40 +0000
Labels:       app.kubernetes.io/component=exporter
              app.kubernetes.io/name=kepler-exporter
              controller-revision-hash=d46fbb7bc
              pod-template-generation=1
              sustainable-computing.io/app=kepler
Annotations:  <none>
Status:       Pending
IP:           10.244.0.71
IPs:
  IP:           10.244.0.71
Controlled By:  DaemonSet/kepler-exporter
Containers:
  kepler-exporter:
    Container ID:  docker://bc025ac92c05336c6f1b041071da78e70d7883a63cc26ba56ec86466188aa15a
    Image:         quay.io/sustainable_computing_io/kepler:latest
    Image ID:      docker-pullable://quay.io/sustainable_computing_io/kepler@sha256:e80cfa22bf41280d0663ac490a28624d4436ae4e56609a0ed27d1bcd82a52b75
    Port:          9102/TCP
    Host Port:     0/TCP
    Command:
      /bin/sh
      -c
    Args:
      until [ -e /tmp/estimator.sock ]; do sleep 1; done && /usr/bin/kepler -v=1 -kernel-source-dir=/usr/share/kepler/kernel_sources -redfish-cred-file-path=/etc/redfish/redfish.csv
    State:          Running
      Started:      Fri, 25 Aug 2023 04:37:44 +0000
    Ready:          True
    Restart Count:  0
    Requests:
      cpu:     100m
      memory:  400Mi
    Liveness:  http-get http://:9102/healthz delay=10s timeout=10s period=60s #success=1 #failure=5
    Environment:
      NODE_IP:     (v1:status.hostIP)
      NODE_NAME:   (v1:spec.nodeName)
    Mounts:
      /etc/kepler/kepler.config from cfm (ro)
      /etc/redfish from redfish (ro)
      /lib/modules from lib-modules (rw)
      /proc from proc (rw)
      /sys from tracing (rw)
      /tmp from tmp (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-5n76j (ro)
  estimator:
    Container ID:
    Image:         kepler_model_server
    Image ID:
    Port:          <none>
    Host Port:     <none>
    Command:
      python3.8
    Args:
      -u
      src/estimate/estimator.py
    State:          Waiting
      Reason:       ImagePullBackOff
    Ready:          False
    Restart Count:  0
    Environment:    <none>
    Mounts:
      /etc/kepler/kepler.config from cfm (ro)
      /tmp from tmp (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from kube-api-access-5n76j (ro)
Conditions:
  Type              Status
  Initialized       True
  Ready             False
  ContainersReady   False
  PodScheduled      True
Volumes:
  tmp:
    Type:       EmptyDir (a temporary directory that shares a pod's lifetime)
    Medium:
    SizeLimit:  <unset>
  lib-modules:
    Type:          HostPath (bare host directory volume)
    Path:          /lib/modules
    HostPathType:  Directory
  tracing:
    Type:          HostPath (bare host directory volume)
    Path:          /sys
    HostPathType:  Directory
  proc:
    Type:          HostPath (bare host directory volume)
    Path:          /proc
    HostPathType:  Directory
  cfm:
    Type:      ConfigMap (a volume populated by a ConfigMap)
    Name:      kepler-cfm
    Optional:  false
  redfish:
    Type:        Secret (a volume populated by a Secret)
    SecretName:  redfish-4kh9d7bc7m
    Optional:    false
  kube-api-access-5n76j:
    Type:                    Projected (a volume that contains injected data from multiple sources)
    TokenExpirationSeconds:  3607
    ConfigMapName:           kube-root-ca.crt
    ConfigMapOptional:       <nil>
    DownwardAPI:             true
QoS Class:                   Burstable
Node-Selectors:              <none>
Tolerations:                 node-role.kubernetes.io/master:NoSchedule
                             node.kubernetes.io/disk-pressure:NoSchedule op=Exists
                             node.kubernetes.io/memory-pressure:NoSchedule op=Exists
                             node.kubernetes.io/not-ready:NoExecute op=Exists
                             node.kubernetes.io/pid-pressure:NoSchedule op=Exists
                             node.kubernetes.io/unreachable:NoExecute op=Exists
                             node.kubernetes.io/unschedulable:NoSchedule op=Exists
Events:
  Type     Reason     Age                From               Message
  ----     ------     ----               ----               -------
  Normal   Scheduled  32s                default-scheduler  Successfully assigned kepler/kepler-exporter-9s4bk to understudent
  Normal   Pulling    32s                kubelet            Pulling image "quay.io/sustainable_computing_io/kepler:latest"
  Normal   Pulled     30s                kubelet            Successfully pulled image "quay.io/sustainable_computing_io/kepler:latest" in 2.560478285s
  Normal   Created    30s                kubelet            Created container kepler-exporter
  Normal   Started    29s                kubelet            Started container kepler-exporter
  Normal   BackOff    26s                kubelet            Back-off pulling image "kepler_model_server"
  Warning  Failed     26s                kubelet            Error: ImagePullBackOff
  Normal   Pulling    11s (x2 over 29s)  kubelet            Pulling image "kepler_model_server"
  Warning  Failed     9s (x2 over 27s)   kubelet            Failed to pull image "kepler_model_server": rpc error: code = Unknown desc = Error response from daemon: pull access denied for kepler_model_server, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
  Warning  Failed     9s (x2 over 27s)   kubelet            Error: ErrImagePull
```

### What did you expect to happen?

"kepler_model_server" should be pull.

### How can we reproduce it (as minimally and precisely as possible)?

Install kepler using Manifests and option `ESTIMATOR_SIDECAR_DEPLOY`

### Kepler image tag

<details>

</details>

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

<details>

</details>

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

sunya-ch, 2023-08-28:
Currently, the flow is building only version tag (v0.6) for now but I will also add latest tag to the flow.

tobby-yuan, 2023-08-28:
@sunya-ch  Why the pod will pull image `kepler_model_server` not `quay.io/sustainable_computing_io/kepler_model_server`?

sunya-ch, 2023-08-28:
Oh.. I see. It is a different issue. We do miss the `kustomize  edit set image` command for the kepler-model-server in the Makefile. 

On the kepler-model-server repo, we set image in the Makefile here:  https://github.com/sustainable-computing-io/kepler-model-server/blob/0bcf63a9625ff71dea293774bd761727ae859d24/Makefile#L73.

sunya-ch, 2023-08-28:
Please confirm these two PRs.

https://github.com/sustainable-computing-io/kepler-model-server/pull/129
https://github.com/sustainable-computing-io/kepler/pull/891

tobby-yuan, 2023-08-28:
@sunya-ch What day does this issue will be fixed?

Can I change the image from `kepler_model_server` to `quay.io/sustainable_computing_io/kepler_model_server` directly using `kubectl edit -n kepler daemonset kepler-exporter` if I wand to fix the bug?

sunya-ch, 2023-08-28:
> @sunya-ch What day does this issue will be fixed?
> 
> Can I change the image from `kepler_model_server` to `quay.io/sustainable_computing_io/kepler_model_server` directly using `kubectl edit -n kepler daemonset kepler-exporter` if I wand to fix the bug?

Yes, you can directly change the image to `quay.io/sustainable_computing_io/kepler_model_server:v0.6` in the daemonset.

Lai-Kenny, 2023-08-28:
Hi,  @sunya-ch . I found make build-manifest `OPTS="ESTIMATOR_SIDECAR_DEPLOY"` and make build-manifest `OPTS="MODEL_SERVER_DEPLOY"`. They seem to have switched to model server, right?

tobby-yuan, 2023-08-28:
@sunya-ch I changed the model server image to the `quay.io/sustainable_computing_io/kepler_model_server:v0.6` and exporter image to `quay.io/sustainable_computing_io/kepler:latest-libbpf` in the daemonset.

These are logs of kepler exporter.
```
kubectl logs -n kepler kepler-exporter-f7fmw kepler-exporter
I0828 11:44:43.463430       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0828 11:44:43.468290       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0828 11:44:43.477644       1 exporter.go:158] Kepler running on version: fad28b7
I0828 11:44:43.477694       1 config.go:270] using gCgroup ID in the BPF program: true
I0828 11:44:43.477748       1 config.go:272] kernel version: 5.15
I0828 11:44:43.477810       1 exporter.go:170] LibbpfBuilt: true, BccBuilt: false
I0828 11:44:43.477846       1 exporter.go:189] EnabledBPFBatchDelete: true
I0828 11:44:43.477890       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0828 11:44:43.477995       1 power.go:71] Unable to obtain power, use estimate method
I0828 11:44:43.478043       1 redfish.go:173] failed to initialize node credential: no supported node credential implementation
I0828 11:44:43.478050       1 power.go:56] use acpi to obtain power
I0828 11:44:43.478243       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0828 11:44:43.493568       1 container_energy.go:109] Using the Ratio/DynPower Power Model to estimate Container Platform Power
I0828 11:44:43.493589       1 container_energy.go:118] Using the Ratio/DynPower Power Model to estimate Container Component Power
I0828 11:44:43.493617       1 process_power.go:108] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I0828 11:44:43.493627       1 process_power.go:117] Using the Ratio/DynPower Power Model to estimate Process Component Power
I0828 11:44:43.493803       1 node_platform_energy.go:53] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power
I0828 11:44:44.709229       1 node_component_energy.go:54] Using the EstimatorSidecar/AbsPower Power Model to estimate Node Component Power
I0828 11:44:44.709405       1 exporter.go:212] Initializing the GPU collector
I0828 11:44:50.714948       1 watcher.go:66] Using in cluster k8s config
libbpf: loading /var/lib/kepler/bpfassets/amd64_kepler.bpf.o
libbpf: elf: section(3) tracepoint/sched/sched_switch, size 2344, link 0, flags 6, type=1
libbpf: sec 'tracepoint/sched/sched_switch': found program 'kepler_trace' at insn offset 0 (0 bytes), code size 293 insns (2344 bytes)
libbpf: elf: section(4) .reltracepoint/sched/sched_switch, size 352, link 29, flags 40, type=9
libbpf: elf: section(5) tracepoint/irq/softirq_entry, size 144, link 0, flags 6, type=1
libbpf: sec 'tracepoint/irq/softirq_entry': found program 'kepler_irq_trace' at insn offset 0 (0 bytes), code size 18 insns (144 bytes)
libbpf: elf: section(6) .reltracepoint/irq/softirq_entry, size 16, link 29, flags 40, type=9
libbpf: elf: section(7) .maps, size 352, link 0, flags 3, type=1
libbpf: elf: section(8) license, size 4, link 0, flags 3, type=1
libbpf: license of /var/lib/kepler/bpfassets/amd64_kepler.bpf.o is GPL
libbpf: elf: section(19) .BTF, size 5759, link 0, flags 0, type=1
libbpf: elf: section(21) .BTF.ext, size 2120, link 0, flags 0, type=1
libbpf: elf: section(29) .symtab, size 1056, link 1, flags 0, type=2
libbpf: looking for externs among 44 symbols...
libbpf: collected 0 externs total
libbpf: map 'processes': at sec_idx 7, offset 0.
libbpf: map 'processes': found type = 1.
libbpf: map 'processes': found key [6], sz = 4.
libbpf: map 'processes': found value [10], sz = 88.
libbpf: map 'processes': found max_entries = 32768.
libbpf: map 'pid_time': at sec_idx 7, offset 32.
libbpf: map 'pid_time': found type = 1.
libbpf: map 'pid_time': found key [6], sz = 4.
libbpf: map 'pid_time': found value [12], sz = 8.
libbpf: map 'pid_time': found max_entries = 32768.
libbpf: map 'cpu_cycles_hc_reader': at sec_idx 7, offset 64.
libbpf: map 'cpu_cycles_hc_reader': found type = 4.
libbpf: map 'cpu_cycles_hc_reader': found key [2], sz = 4.
libbpf: map 'cpu_cycles_hc_reader': found value [6], sz = 4.
libbpf: map 'cpu_cycles_hc_reader': found max_entries = 128.
libbpf: map 'cpu_cycles': at sec_idx 7, offset 96.
libbpf: map 'cpu_cycles': found type = 2.
libbpf: map 'cpu_cycles': found key [6], sz = 4.
libbpf: map 'cpu_cycles': found value [12], sz = 8.
libbpf: map 'cpu_cycles': found max_entries = 128.
libbpf: map 'cpu_ref_cycles_hc_reader': at sec_idx 7, offset 128.
libbpf: map 'cpu_ref_cycles_hc_reader': found type = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found key [2], sz = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found value [6], sz = 4.
libbpf: map 'cpu_ref_cycles_hc_reader': found max_entries = 128.
libbpf: map 'cpu_ref_cycles': at sec_idx 7, offset 160.
libbpf: map 'cpu_ref_cycles': found type = 2.
libbpf: map 'cpu_ref_cycles': found key [6], sz = 4.
libbpf: map 'cpu_ref_cycles': found value [12], sz = 8.
libbpf: map 'cpu_ref_cycles': found max_entries = 128.
[... 65 lines cut ...]
libbpf: prog 'kepler_trace': found map 0 (processes, sec 7, off 0) for insn #241
libbpf: sec '.reltracepoint/sched/sched_switch': relo #20: insn #261 against 'processes'
libbpf: prog 'kepler_trace': found map 0 (processes, sec 7, off 0) for insn #261
[... cut ...]
libbpf: map 'cache_miss': created successfully, fd=19
libbpf: map 'cpu_freq_array': created successfully, fd=20
I0828 11:44:50.823749       1 libbpf_attacher.go:143] failed to get perf event cpu_instructions_hc_reader: failed to find BPF map cpu_instructions_hc_reader: no such file or directory
I0828 11:44:50.824146       1 libbpf_attacher.go:157] Successfully load eBPF module from libbpf object
I0828 11:44:50.857140       1 exporter.go:276] Started Kepler in 7.379527116s
```

Is this right? kpler works normally or not?

rootfs, 2023-08-28:
this looks right, kepler is started successfully.

closed 2023-08-28 by rootfs