sustainable-computing-io/kepler #897: Kepler reports huge energy for some pods in K8s cluster on cloud VMs

jwjwjw3, 2023-08-29:
### What happened?

Kepler has very huge energy estimations for some pods that runs in less than an hour (at Peta Joules scale, or 10^15 J) inside a 1-master several-worker K8s cluster on IBM Cloud VMs without GPU.

Here are the output metric results of "kepler_container_joules_total" from a kepler pod:
```
# HELP kepler_container_joules_total Aggregated RAPL Package + Uncore + DRAM + GPU + other host components (platform - package - dram) in joules
# TYPE kepler_container_joules_total counter
kepler_container_joules_total{command="",container_id="325652d9b4cf1a02c37334423932361aa5b9974b2670b5f1e3eb8651d2febe0a",container_name="install-cni",container_namespace="kube-flannel",mode="dynamic",pod_name="kube-flannel-ds-j94ms"} 9.223372037599701e+14
kepler_container_joules_total{command="",container_id="5ad81f5e6bb2283d1274daf53f6f0a21ab8e7e9438d9454ff2eb2316c69e0318",container_name="install-cni-plugin",container_namespace="kube-flannel",mode="dynamic",pod_name="kube-flannel-ds-j94ms"} 9.223372037599701e+14
kepler_container_joules_total{command="",container_id="ebab60e1126208931dfe80d40bcc8396c83e93db3b55c7d8510b6b37844cd315",container_name="init-config-reloader",container_namespace="monitoring",mode="dynamic",pod_name="prometheus-k8s-0"} 9.223372037597599e+14
kepler_container_joules_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",mode="dynamic",pod_name="system_processes"} 9.223372037599701e+14
kepler_container_joules_total{command="adapter",container_id="04f21f73f80980e6a1b88c81c19bfe9d0857e35ab8702a1faa189269c0a48147",container_name="prometheus-adapter",container_namespace="monitoring",mode="dynamic",pod_name="prometheus-adapter-757f9b4cf9-c9b44"} 9.223372037599701e+14
kepler_container_joules_total{command="alertmanag",container_id="166f9252acb14a6642bc9b0ccd527a687ef271edc69ac2e53ac3928a51578809",container_name="alertmanager",container_namespace="monitoring",mode="dynamic",pod_name="alertmanager-main-0"} 9.223372037599701e+14
kepler_container_joules_total{command="darknet",container_id="0552b5305b39b0b79bfcb2ce2648ea25ab5df4408685048bb9f8e57a41640dd2",container_name="yolo",container_namespace="default",mode="dynamic",pod_name="yolo3-q5l2x"} 9.223372036930931e+14
kepler_container_joules_total{command="flanneld",container_id="41691adab518f27196bcf7874e47e10c2958f06cf4af7e9f63f5f4e7223703bd",container_name="kube-flannel",container_namespace="kube-flannel",mode="dynamic",pod_name="kube-flannel-ds-j94ms"} 9.223372037599701e+14
kepler_container_joules_total{command="grafana",container_id="8fecd84b0a991b9be7fba4072e3369f3b141e243f28e644393037940e5285c4e",container_name="grafana",container_namespace="monitoring",mode="dynamic",pod_name="grafana-6849bbf859-lp6zd"} 9.223372037599701e+14
kepler_container_joules_total{command="kepler",container_id="8cb78c9dfb7dbbb8280b364491eb8463b9c54fb3948b78a5e6ad1a089ead2a2d",container_name="kepler-exporter",container_namespace="kepler",mode="dynamic",pod_name="kepler-exporter-kwtq9"} 9.223372037599701e+14
kepler_container_joules_total{command="kube-proxy",container_id="a107d34704860f6c9285ed070b57c90f908f8180fd8c2c2d38950d4bd7f5163f",container_name="kube-proxy",container_namespace="kube-system",mode="dynamic",pod_name="kube-proxy-f4w6l"} 9.223372037599701e+14
kepler_container_joules_total{command="kube-rbac-",container_id="2340d0b7fd868ca3f75c54ca808d98deaf30fb505bc719b6c2da8d24af2ad6dd",container_name="kube-rbac-proxy",container_namespace="monitoring",mode="dynamic",pod_name="node-exporter-wtfjx"} 9.223372037599701e+14
kepler_container_joules_total{command="node_expor",container_id="b12cf24237e961451db66f174589b531b9c9566e191f4cba5394dc5b563b79e3",container_name="node-exporter",container_namespace="monitoring",mode="dynamic",pod_name="node-exporter-wtfjx"} 9.223372037599701e+14
kepler_container_joules_total{command="prometheus",container_id="186e9a4421aeb580458408d2ede2d1408d4c89baf0c8e3b40ab0560c8e1b70ea",container_name="config-reloader",container_namespace="monitoring",mode="dynamic",pod_name="alertmanager-main-0"} 9.223372037599701e+14
kepler_container_joules_total{command="prometheus",container_id="6fbc0fa14852d32454a3ab5ec9206b466836dc5c2a43609fd494fe9032860cf9",container_name="prometheus",container_namespace="monitoring",mode="dynamic",pod_name="prometheus-k8s-0"} 9.223372037597599e+14
kepler_container_joules_total{command="prometheus",container_id="ff2bc5984e5374e82b852cbcb5793bd4c708d0cc9143f3ad3b4b9e8bd9c0b252",container_name="config-reloader",container_namespace="monitoring",mode="dynamic",pod_name="prometheus-k8s-0"} 9.223372037597599e+14
```
Here are the output metric results of "kepler_container_package_joules_total":

```
# HELP kepler_container_package_joules_total Aggregated RAPL value in package (socket) in joules
# TYPE kepler_container_package_joules_total counter
kepler_container_package_joules_total{command="",container_id="325652d9b4cf1a02c37334423932361aa5b9974b2670b5f1e3eb8651d2febe0a",container_name="install-cni",container_namespace="kube-flannel",mode="dynamic",pod_name="kube-flannel-ds-j94ms"} 9.223372037371426e+14
kepler_container_package_joules_total{command="",container_id="5ad81f5e6bb2283d1274daf53f6f0a21ab8e7e9438d9454ff2eb2316c69e0318",container_name="install-cni-plugin",container_namespace="kube-flannel",mode="dynamic",pod_name="kube-flannel-ds-j94ms"} 9.223372037371426e+14
kepler_container_package_joules_total{command="",container_id="ebab60e1126208931dfe80d40bcc8396c83e93db3b55c7d8510b6b37844cd315",container_name="init-config-reloader",container_namespace="monitoring",mode="dynamic",pod_name="prometheus-k8s-0"} 9.223372037369908e+14
kepler_container_package_joules_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",mode="dynamic",pod_name="system_processes"} 9.223372037371426e+14
kepler_container_package_joules_total{command="adapter",container_id="04f21f73f80980e6a1b88c81c19bfe9d0857e35ab8702a1faa189269c0a48147",container_name="prometheus-adapter",container_namespace="monitoring",mode="dynamic",pod_name="prometheus-adapter-757f9b4cf9-c9b44"} 9.223372037371426e+14
kepler_container_package_joules_total{command="alertmanag",container_id="166f9252acb14a6642bc9b0ccd527a687ef271edc69ac2e53ac3928a51578809",container_name="alertmanager",container_namespace="monitoring",mode="dynamic",pod_name="alertmanager-main-0"} 9.223372037371426e+14
kepler_container_package_joules_total{command="darknet",container_id="0552b5305b39b0b79bfcb2ce2648ea25ab5df4408685048bb9f8e57a41640dd2",container_name="yolo",container_namespace="default",mode="dynamic",pod_name="yolo3-q5l2x"} 9.223372036896671e+14
kepler_container_package_joules_total{command="flanneld",container_id="41691adab518f27196bcf7874e47e10c2958f06cf4af7e9f63f5f4e7223703bd",container_name="kube-flannel",container_namespace="kube-flannel",mode="dynamic",pod_name="kube-flannel-ds-j94ms"} 9.223372037371426e+14
kepler_container_package_joules_total{command="grafana",container_id="8fecd84b0a991b9be7fba4072e3369f3b141e243f28e644393037940e5285c4e",container_name="grafana",container_namespace="monitoring",mode="dynamic",pod_name="grafana-6849bbf859-lp6zd"} 9.223372037371426e+14
kepler_container_package_joules_total{command="kepler",container_id="8cb78c9dfb7dbbb8280b364491eb8463b9c54fb3948b78a5e6ad1a089ead2a2d",container_name="kepler-exporter",container_namespace="kepler",mode="dynamic",pod_name="kepler-exporter-kwtq9"} 9.223372037371426e+14
kepler_container_package_joules_total{command="kube-proxy",container_id="a107d34704860f6c9285ed070b57c90f908f8180fd8c2c2d38950d4bd7f5163f",container_name="kube-proxy",container_namespace="kube-system",mode="dynamic",pod_name="kube-proxy-f4w6l"} 9.223372037371426e+14
kepler_container_package_joules_total{command="kube-rbac-",container_id="2340d0b7fd868ca3f75c54ca808d98deaf30fb505bc719b6c2da8d24af2ad6dd",container_name="kube-rbac-proxy",container_namespace="monitoring",mode="dynamic",pod_name="node-exporter-wtfjx"} 9.223372037371426e+14
kepler_container_package_joules_total{command="node_expor",container_id="b12cf24237e961451db66f174589b531b9c9566e191f4cba5394dc5b563b79e3",container_name="node-exporter",container_namespace="monitoring",mode="dynamic",pod_name="node-exporter-wtfjx"} 9.223372037371426e+14
[... cut ...]
2. containerd version < 1.7.2 (I'm not 100% sure about containerd version, when I tested Kepler 0.5.1, 0.5.3, 0.5.4 with containerd as container runtime last week, I didn't pin containerd and runc version, just let apt decide which version to install. Today I tried containerd version 1.7.2, and around 5 YOLO pods completed in 30mins, the Peta Joule issue didn't appear in this 30min run. I will continue testing containerd 1.7.2 together with some previous containerd versions, and post an update if Peta Joule issue is seen) 
</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>
Pod network plugin is Flannel v0.22.1

Kube-Prometheus version is 0.12

</details>

rootfs, 2023-08-29:
@jwjwjw3 thanks for reporting. 
- Can you get the kepler logs `kubectl logs -n kepler daemonset/kepler-exporter`? 
- Can you get the `kepler_container_cgroupfs_cpu_usage_us_total` metrics too?

rootfs, 2023-08-29:
@sunya-ch @marceloamaral any thoughts?

jwjwjw3, 2023-08-29:
@rootfs I just confirmed that the Peta Joule issue still exists in K8s 1.25.11, Kepler 0.5.3, with container runtime: containerd 1.7.2 (installed by ```apt install 1.7.2-0ubuntu1~22.04.1```), runc 1.1.7 (installed by ```apt install 1.1.7-0ubuntu1~22.04.1```) when running the aforementioned YOLO containers. The K8s cluster has 1 master and 4 workers. Here I provide Kepler logs, Kepler pod metrics, and some Prometheus WebUI snapshots on Kepler pod metrics in this run:

Kepler logs from a kepler pod on a node with many pods having Peta Joules of energy:
```console
$ kubectl logs -n kepler daemonset/kepler-exporter
Found 4 pods, using pod/kepler-exporter-5f8pc
I0828 22:29:40.475407       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0828 22:29:40.479699       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0828 22:29:40.489821       1 exporter.go:158] Kepler running on version: 4b8a281
I0828 22:29:40.489840       1 config.go:270] using gCgroup ID in the BPF program: true
I0828 22:29:40.489909       1 config.go:272] kernel version: 5.15
I0828 22:29:40.489948       1 exporter.go:170] LibbpfBuilt: false, BccBuilt: true
I0828 22:29:40.490008       1 config.go:203] kernel source dir is set to /usr/share/kepler/kernel_sources
I0828 22:29:40.490088       1 exporter.go:189] EnabledBPFBatchDelete: true
I0828 22:29:40.490119       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0828 22:29:40.490149       1 power.go:71] Unable to obtain power, use estimate method
I0828 22:29:40.490204       1 redfish.go:173] failed to initialize node credential: no supported node credential implementation
I0828 22:29:40.490214       1 power.go:56] use acpi to obtain power
I0828 22:29:40.490411       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0828 22:29:40.513566       1 container_energy.go:109] Using the Ratio/DynPower Power Model to estimate Container Platform Power
I0828 22:29:40.513585       1 container_energy.go:118] Using the Ratio/DynPower Power Model to estimate Container Component Power
I0828 22:29:40.513594       1 process_power.go:108] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I0828 22:29:40.513601       1 process_power.go:117] Using the Ratio/DynPower Power Model to estimate Process Component Power
I0828 22:29:40.513759       1 node_platform_energy.go:53] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power
I0828 22:29:40.513829       1 node_component_energy.go:54] Using the LinearRegressor/AbsPower Power Model to estimate Node Component Power
I0828 22:29:40.513847       1 exporter.go:212] Initializing the GPU collector
I0828 22:29:46.519275       1 watcher.go:66] Using in cluster k8s config
cannot attach kprobe, probe entry may not exist
I0828 22:29:48.489797       1 bcc_attacher.go:94] attaching kprobe to finish_task_switch failed, trying finish_task_switch.isra.0 instead
perf_event_open: No such file or directory
I0828 22:29:48.507702       1 bcc_attacher.go:120] failed to attach perf event cpu_cycles_hc_reader: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I0828 22:29:48.507769       1 bcc_attacher.go:120] failed to attach perf event cpu_ref_cycles_hc_reader: failed to open bpf perf event: no such file or directory
I0828 22:29:48.507811       1 bcc_attacher.go:120] failed to attach perf event cpu_instructions_hc_reader: passed table has wrong size: key_size=0, leaf_size=0
perf_event_open: No such file or directory
I0828 22:29:48.507853       1 bcc_attacher.go:120] failed to attach perf event cache_miss_hc_reader: failed to open bpf perf event: no such file or directory
I0828 22:29:48.507866       1 bcc_attacher.go:183] Successfully load eBPF module from bcc with option: [-DMAP_SIZE=10240 -DNUM_CPUS=4 -DSET_GROUP_ID]
I0828 22:29:48.518177       1 exporter.go:276] Started Kepler in 8.028379633s
```
Here are three screen snapshots of three Kepler metrics from Prometheus WebUI on the node where pod ```kepler-exporter-5f8pc``` runs: 
[image]
[image]

[image]

I also attach a file of all kepler metrics from this kepler-exporter-5f8pc pod: 
[metrics_kepler-exporter-5f8pc.txt](https://github.com/sustainable-computing-io/kepler/files/12459929/metrics_kepler-exporter-5f8pc.txt)

In this file, the ```kepler_container_cgroupfs_cpu_usage_us_total``` are:
```
# HELP kepler_container_cgroupfs_cpu_usage_us_total Aggregated cpu usage obtained from cGroups
# TYPE kepler_container_cgroupfs_cpu_usage_us_total counter
kepler_container_cgroupfs_cpu_usage_us_total{command="",container_id="2626cc5d6a3dd846dd4cad38113e7520ded93730b38f2cb8c82c8acd5e4f1002",container_name="install-cni",container_namespace="kube-flannel",pod_name="kube-flannel-ds-vfprk"} 0
kepler_container_cgroupfs_cpu_usage_us_total{command="",container_id="b61af7de88d5650fc8c91fe98d0a9f61bb07fbd4572964de3e2d5204749dec92",container_name="install-cni-plugin",container_namespace="kube-flannel",pod_name="kube-flannel-ds-vfprk"} 0
kepler_container_cgroupfs_cpu_usage_us_total{command="",container_id="de628e68b632c564351eb8623b3e960b71aa268ef51a31b11d058e90c8e6055a",container_name="init-config-reloader",container_namespace="monitoring",pod_name="prometheus-k8s-0"} 0
[... cut ...]
kepler_container_cgroupfs_cpu_usage_us_total{command="prometheus",container_id="49df3448d192b77ba95912d4435565edd90237071224534a555a50dd4900ea78",container_name="config-reloader",container_namespace="monitoring",pod_name="prometheus-k8s-0"} 7.21321e+06
kepler_container_cgroupfs_cpu_usage_us_total{command="prometheus",container_id="b974c96a7b263d9e583c62bf2086f3d695f9df67d5b3e7fed260d0361f35ce49",container_name="prometheus",container_namespace="monitoring",pod_name="prometheus-k8s-0"} 7.21321e+06
```

(In this run, yolowithsleep and yolo K8s jobs are using the same YOLO image, the containers in these two kinds of jobs are the same)

sunya-ch, 2023-08-29:
We are investigating the bug on training pipeline and still working on end-to-end (between kepler and estimator) model validation flow. Need more time to identify the issue. The current one is more like POC to show how the flow works. Any contribution of identification and fix is always welcome. This is scheduled to be stable in v0.6 release.

Also, could you please share kubelet metrics value? With the current built model, we use power based on kubelet metrics because there was a bug on kepler to collect data from other sources before which fixed by the PR https://github.com/sustainable-computing-io/kepler/pull/872.

rootfs, 2023-08-29:
I think there might be a calculation issue. See the two metrics below, the `yolo` metrics look reasonable while the `alertmanag` metrics is too high
```
kepler_container_core_joules_total{command="alertmanag",container_id="838402c9f69bbd2cab6df2052aff389023a40fa20f335dd7108db072ca5d6e3c",container_name="alertmanager",container_namespace="monitoring",mode="dynamic",pod_name="alertmanager-main-0"} 8.198552922686818e+15
kepler_container_core_joules_total{command="darknet",container_id="1f91cdac940dfbed113b407a9b4e9e5c31b5ceaab555f8797f5da6a3d9f83594",container_name="yolo",container_namespace="default",mode="dynamic",pod_name="yolo15-wqrct"} 41910.987
```

```
kepler_container_joules_total{command="alertmanag",container_id="838402c9f69bbd2cab6df2052aff389023a40fa20f335dd7108db072ca5d6e3c",container_name="alertmanager",container_namespace="monitoring",mode="dynamic",pod_name="alertmanager-main-0"} 8.198552922006828e+15
kepler_container_joules_total{command="darknet",container_id="1f91cdac940dfbed113b407a9b4e9e5c31b5ceaab555f8797f5da6a3d9f83594",container_name="yolo",container_namespace="default",mode="dynamic",pod_name="yolo15-wqrct"} 14439.228000000001
```

rootfs, 2023-08-29:
@marceloamaral can you take a look? thanks

jwjwjw3, 2023-08-29:
> 
> Also, could you please share kubelet metrics value? With the current built model, we use power based on kubelet metrics because there was a bug on kepler to collect data from other sources before which fixed by the PR #872.

Here are the ```kepler_container_kubelet_cpu_usage_total``` and ```kepler_container_kubelet_memory_bytes_total``` metrics from the same kepler-exporter-5f8pc pod's metric results file [metrics_kepler-exporter-5f8pc.txt](https://github.com/sustainable-computing-io/kepler/files/12469469/metrics_kepler-exporter-5f8pc.txt):

```
# HELP kepler_container_kubelet_cpu_usage_total Aggregated cpu usage obtained from kubelet
# TYPE kepler_container_kubelet_cpu_usage_total counter
kepler_container_kubelet_cpu_usage_total{command="",container_id="2626cc5d6a3dd846dd4cad38113e7520ded93730b38f2cb8c82c8acd5e4f1002",container_name="install-cni",container_namespace="kube-flannel",pod_name="kube-flannel-ds-vfprk"} 0
kepler_container_kubelet_cpu_usage_total{command="",container_id="b61af7de88d5650fc8c91fe98d0a9f61bb07fbd4572964de3e2d5204749dec92",container_name="install-cni-plugin",container_namespace="kube-flannel",pod_name="kube-flannel-ds-vfprk"} 0
kepler_container_kubelet_cpu_usage_total{command="",container_id="de628e68b632c564351eb8623b3e960b71aa268ef51a31b11d058e90c8e6055a",container_name="init-config-reloader",container_namespace="monitoring",pod_name="prometheus-k8s-0"} 0
kepler_container_kubelet_cpu_usage_total{command="",container_id="system_processes",container_name="system_processes",container_namespace="system",pod_name="system_processes"} 0
kepler_container_kubelet_cpu_usage_total{command="adapter",container_id="b2ff22e91648fa669fb1d3c3edfb77973a6ae59dd172cde78d3c3458c6250e6a",container_name="prometheus-adapter",container_namespace="monitoring",pod_name="prometheus-adapter-757f9b4cf9-rrj9q"} 91
kepler_container_kubelet_cpu_usage_total{command="alertmanag",container_id="838402c9f69bbd2cab6df2052aff389023a40fa20f335dd7108db072ca5d6e3c",container_name="alertmanager",container_namespace="monitoring",pod_name="alertmanager-main-0"} 57
kepler_container_kubelet_cpu_usage_total{command="darknet",container_id="1f91cdac940dfbed113b407a9b4e9e5c31b5ceaab555f8797f5da6a3d9f83594",container_name="yolo",container_namespace="default",pod_name="yolo15-wqrct"} 715
kepler_container_kubelet_cpu_usage_total{command="darknet",container_id="a16b6566b69b041e6668d4731a1c3a35894ab6b4ea5f65a9719f79c15e58dee9",container_name="yolowithsleep",container_namespace="default",pod_name="yolowithsleep41-4d5vc"} 988
kepler_container_kubelet_cpu_usage_total{command="darknet",container_id="e558972edd385a8499485eb75037c22d81c2c05533d62b3cc7f25c3fa27c93db",container_name="yolo",container_namespace="default",pod_name="yolo36-z529t"} 1904
kepler_container_kubelet_cpu_usage_total{command="flanneld",container_id="f745393bc39ec8f42c9c5ca47ab9bdf5cc9e17a9e08ae81a943c27924b3e41fb",container_name="kube-flannel",container_namespace="kube-flannel",pod_name="kube-flannel-ds-vfprk"} 119
kepler_container_kubelet_cpu_usage_total{command="grafana",container_id="4116090336e342511081982a5af074afcded1b7773bcb0ebb6c3ec907ada094e",container_name="grafana",container_namespace="monitoring",pod_name="grafana-6849bbf859-f2mcq"} 138
kepler_container_kubelet_cpu_usage_total{command="kepler",container_id="1b5a27926f68d8d577262fca4c3befd2d08312aed8e3a95e084b392745af7dbe",container_name="kepler-exporter",container_namespace="kepler",pod_name="kepler-exporter-5f8pc"} 130
kepler_container_kubelet_cpu_usage_total{command="kube-proxy",container_id="f6c0d8d04c88622794bb6f35d50c963c10232f4c242e3e6e0eab4062fb1325ce",container_name="kube-proxy",container_namespace="kube-system",pod_name="kube-proxy-6fp2t"} 7
kepler_container_kubelet_cpu_usage_total{command="kube-rbac-",container_id="f5d76b2f7276e1379d80267cc8e60eb9df6a8d334636d06ffc5d0c15153be541",container_name="kube-rbac-proxy",container_namespace="monitoring",pod_name="node-exporter-vz54f"} 5
kepler_container_kubelet_cpu_usage_total{command="node_expor",container_id="09aabdad5687f3716f1d8fd5727c0808afe66cf18069739245e0f4512a760a30",container_name="node-exporter",container_namespace="monitoring",pod_name="node-exporter-vz54f"} 42
kepler_container_kubelet_cpu_usage_total{command="prometheus",container_id="1d8ca3f2356c1b553c4ebdeaba077c0dc238c5354d12f83e3bdafbb7f3187241",container_name="config-reloader",container_namespace="monitoring",pod_name="alertmanager-main-0"} 3
kepler_container_kubelet_cpu_usage_total{command="prometheus",container_id="49df3448d192b77ba95912d4435565edd90237071224534a555a50dd4900ea78",container_name="config-reloader",container_namespace="monitoring",pod_name="prometheus-k8s-0"} 5
kepler_container_kubelet_cpu_usage_total{command="prometheus",container_id="b974c96a7b263d9e583c62bf2086f3d695f9df67d5b3e7fed260d0361f35ce49",container_name="prometheus",container_namespace="monitoring",pod_name="prometheus-k8s-0"} 539
# HELP kepler_container_kubelet_memory_bytes_total Aggregated memory bytes obtained from kubelet
# TYPE kepler_container_kubelet_memory_bytes_total counter
kepler_container_kubelet_memory_bytes_total{command="",container_id="2626cc5d6a3dd846dd4cad38113e7520ded93730b38f2cb8c82c8acd5e4f1002",container_name="install-cni",container_namespace="kube-flannel",pod_name="kube-flannel-ds-vfprk"} 0
[... cut ...] pod_name="prometheus-k8s-0"} 4.25070592e+08
```

And I also attach the two screen snapshots of the kubelet CPU and memory metrics in the run I posted yesterday: 

[image]

In the Kubelet CPU metric plot above, the many lines increasing from 0 to ~2K are YOLO pods, the slowly increasing lines at bottom are Prometheus and Kube-system pods. 

[image]

In this Kubelet memory metric plot above, the oscillating lines (from 0 to 900MB) are YOLO pods, the other lines are Prometheus and Kube-system pods.

rootfs, 2023-08-31:
@jwjwjw3 can you try the latest kepler image `quay.io/repository/sustainable_computing_io/kepler:latest`

jwjwjw3, 2023-08-31:
> @jwjwjw3 can you try the latest kepler image `quay.io/sustainable_computing_io/kepler:latest`

@rootfs last night I tried the latest Kepler, deployed by manifests. Unfortunately the Peta Joule issue still exists. 

I git cloned the latest version of Kepler from GitHub, used the latest commit as:
```console
$ git status 
HEAD detached at 5488b12
nothing to commit, working tree clean
```
I also made sure there is not ```_output``` folder, then I run ```make build-manifest OPTS="PROMETHEUS_DEPLOY"``` and a new ```_output``` folder is generated, I can find ```_output/_generated_manifest/deployment.yml``` under the Kepler folder. I used this to deploy Kepler.

I created a new set of IBM Cloud Ubuntu 22.04 VMs, this time 1 master with 6 workers. I still used K8s 1.25.11 with container runtime containerd 1.7.2, runc 1.1.7, flannel v0.22.1, and Kube-Prometheus 0.12. The workloads I tested includes YOLO (DockerHub:  jvpoidaq/cosco_defog_yolo:latest), PocketSphinx (DockerHub:  jvpoidaq/cosco_defog_pocketsphinx:latest, and Aeneas (DockerHub:  jvpoidaq/cosco_defog_aeneas:latest), still the same container images as listed before. I see two nodes (worker node 05 and 06) where some YOLO pods still have Peta Joules of energy while the other 4 nodes seem normal. I list Prometheus WebUI's result of worker node 05 here and post some more details I found:

[image]

From the figure above we see the run takes 8-9 hours, multiple abnormal spikes are seen on this worker node 05. 

[image]

[image]

In the figures above, I zoomed in to the first spike that happened around 1:30am. The spike on ```kepler_node_core_joules_total``` comes very fast and also goes away very fast. At 1:30:22am the ```kepler_node_core_joules_total``` for worker 05 is 4339377.18 Joules (still normal), then at 1:30:25am it goes to >18^15 Joules; at 1:30:28am it maintains the same huge value, and then 3s later it drops to 0 Joules. 

[image]

Almost at the same time around 1:30:25am, the three YOLO containers suddenly consumes a huge amount of energy, the top three lines are three YOLO containers inside three YOLO pods (yolo16, yolo17, and yolo18), the 4th line around 4.00PJ is a system process in the figure above.

 
[image]

I think the Kubelet CPU usage in the above plot seems normal, the reason why we see some line segments and some dots in between line segments are: YOLO containers uses a script named ```execute_yolo.sh``` to call ```darknet``` binary for computing results on multiple RGB images in a loop, one image at a time; when ```darknet``` runs on an image we get a line segment, and when one image finishes the command of this container changes from ```darknet``` back to ```execute_yolo.sh``` and we see a dot. 

[image]

The cgroupfs plot above shows almost no CPU usage for the pods on worker node 05, the top curve is flat around 1:30:25am. And also I provide two more tables that indicate there is almost no change in cgroupfs cpu usage metrics for a whole 23s interval from 1:30:08am to 1:30:31am:

[image]

[image]

sunya-ch, 2023-09-01:
We have no `core`, `uncore` model because it is not available in the machine we collected.  
Could you also show `container_package` and `node_package` power?
node_package power should show the result from the model and container_package power will show the result after applying the ratio. 
If node_package_power is wrong, it is bug at the beginning of estimation.
If container_package_power is wrong, it is bug when we apply the ratio.
If both are fine, it is bug when computing the core power.

jwjwjw3, 2023-09-01:
> We have no `core`, `uncore` model because it is not available in the machine we collected. Could you also show `container_package` and `node_package` power after applying the latest image? node_package power should show the result from the model and container_package power will show the result after applying the ratio. If node_package_power is wrong, it is bug at the beginning of estimation. If container_package_power is wrong, it is bug when we apply the ratio. If both are fine, it is bug when computing the core power.

Sure, here are the two metric plots from Prometheus:

[image]

[image]

The two snapshots are from the same run as my last post (latest Kepler at Git commit 5488b12), and I'm still using results from worker node 05 at around 1:30:25am. Both metrics have Peta Joules...

rootfs, 2023-09-01:
@jwjwjw3 thanks for the analysis, can you dump the kepler metrics from the nodes that have high joules and normal joules so we can see what the delta look like?

jwjwjw3, 2023-09-02:
> * dump all the kepler metrics from the nodes (i.e. nodes 4 or 5) that have high joules and normal (any of the other 4 nodes) joules so we can see what the delta look like?

Sure, actually this time I want to provide the Prometheus database that contains all Kepler metrics for that run of 8-9 hours. The google drive link to the Prometheus database is: https://drive.google.com/drive/folders/1gzcv77N7DRk9Bs1d-QEK0nFmU579Mdom?usp=sharing

Here is the way to quickly launch Prometheus WebUI and see the Kepler metrics on all the nodes when some nodes suddenly consumes Peta Joules of energy, I take ```kepler_node_package_joules_total``` as example:

1. Go to Prometheus official website at https://prometheus.io/download/ and download Prometheus executable file. 
2. Run this command in a terminal: ```prometheus --storage.tsdb.path /path/to/Prometheus_storage_db_2023-08-30_jwjwjw3 --web.listen-address localhost:9090``` to start a Prometheus server at http://localhost:9090 
3. Use any web browser to access the Prometheus WebUI page at http://localhost:9090, now you can just type the Kepler metric ```kepler_node_package_joules_total``` in the Prometheus WebUI, unclick "use local time" at top left (so that we always use UTC time), select "Table" instead of "Graph", click and select time "2023-08-31 6:30:22" (which is 2023-08-31 1:30:22 in US Central time as my previous posts). When selecting time, you should see the WebUI like this:

[image]

Note that you can directly click "increment second" top arrow or "decrement second" bottom arrow in the "Pick second" section on the floating time window, as I marked with red circles. As you click the arrows you should simultaneously see the table behind the floating time window change, without closing or leaving the floating time window. Click the "increment second" once so that we move to UTC time "2023-08-31 6:30:23", you should immediately notice that worker node 05's value suddenly becomes huge, this is the spike I mentioned in my last post. 

> * get the kepler logs from these two nodes

I didn't save Kepler logs for this run... I will try another run soon, and save Kepler metrics together with Kepler logs.

rootfs, 2023-09-02:
thanks @jwjwjw3

sunya-ch, 2023-09-04:
Thank you @jwjwjw3
Now, we are able to confirm that the problem is at the estimation by model. 
I hope this is a final request to you, could you apply irate() to the plot on kubelet? 
Because the result is an aggregated value, I would like to see the rate value which uses for estimation in long run including at the time of peak during `1:30:20` - `1:30:25`. I guess the rate of kubelet must be very high at that point.

jwjwjw3, 2023-09-04:
> Thank you @jwjwjw3 Now, we are able to confirm that the problem is at the estimation by model. I hope this is a final request to you, could you apply irate() to the plot on kubelet? Because the [...]

Sure, here I attached the ```irate()``` results from Prometheus plots using the Prometheus database at https://drive.google.com/drive/folders/1gzcv77N7DRk9Bs1d-QEK0nFmU579Mdom?usp=sharing as I used in my previous post, for consistency I used UTC time for all plots below:

[image]

Above is an overall plot on metric ```kepler_container_kubelet_cpu_usage_total```, the increasing rates from user pods (YOLO, PocketSphinx, and Aeneas containers) are fluctuating. Some peaks with value >5 are from YOLO pods, some other peaks are from other pods like PocketSphinx pods.

[image]

The zoom-in plot above on worker node 05 at around ```6:30:25 UTC time``` (1:30:25 US Central time) shows a clear peak of yolo16 pod. 

[image]

The overall plot above on ```kepler_container_kubelet_memory_bytes_total``` shows the memory usage increase rates of user pods are also fluctuating. Almost all peaks above 200m are from yolo pods.

[image]

Above is the zoom-in ```irate()``` plot of memory usage. Yolo pods tends to fluctuate a lot. At around ```6:30:30 UTC time``` (1:30:30 US Central time) I see two peaks, first peak at value 270M is from yolo17 pod as in the plot, and the immediate following second peak at around 280M is from yolo18 pod. Also, at ```6:30:25 UTC time``` (1:30:25 US Central time) there are two smaller peaks at ~100m at almost the same time, these two smaller peaks are from yolo16 and yolo17.

[image]

Here is the same zoom-in ```irate()``` plot of memory usage plot with time resolution increased from 3s to 1s.

Additional note: when I built the yolo container and run it on a local machine within a docker environment, I found that when yolo container starts to work on a new RGB image (from command ```execute_yolo.sh``` to ```darknet```), its memory usage quickly increases from <100MB to ~1GB within 1 second, then have small flucuations around 1GB for a short amount of time; after this RGB image finishes the memory usage quickly drops to <100MB within 1 second. My other two pods (PocketSphinx, Aeneas) memory usages do not change this fast. Is it possible that the quick and large memory fluctuation in my YOLO pods caused the Peta Joule issue?

sunya-ch, 2023-09-05:
> > Thank you @jwjwjw3 Now, we are able to confirm that the problem is at the estimation by model. I hope this is a final request to you, could you apply irate() to the plot on kubelet? Because the result is an aggregated value, I would like to see the rate value which uses for estimation in long run including at the time of peak during `1:30:20` - `1:30:25`. I guess the rate of kubelet must be very high at that point.
> 
> Sure, here I attached the `irate()` results from Prometheus plots using the Prometheus database at https://drive.google.com/drive/folders/1gzcv77N7DRk9Bs1d-QEK0nFmU579Mdom?usp=sharing as I used in my previous post, for consistency I used UTC time for all plots below:
> 
> [image]
> 
> Above is an overall plot on metric `kepler_container_kubelet_cpu_usage_total`, the increasing rates from user pods (YOLO, PocketSphinx, and Aeneas containers) are fluctuating. Some peaks with value >5 are from YOLO pods, some other peaks are from other pods like PocketSphinx pods.
> 
> [image]
> 
> The zoom-in plot above on worker node 05 at around `6:30:25 UTC time` (1:30:25 US Central time) shows a clear peak of yolo16 pod.
> 
> [image]
> 
> The overall plot above on `kepler_container_kubelet_memory_bytes_total` shows the memory usage increase rates of user pods are also fluctuating. Almost all peaks above 200m are from yolo pods.
> 
> [image]
> 
> Above is the zoom-in `irate()` plot of memory usage. Yolo pods tends to fluctuate a lot. At around `6:30:30 UTC time` (1:30:30 US Central time) I see two peaks, first peak at value 270M is from yolo17 pod as in the plot, and the immediate following second peak at around 280M is from yolo18 pod. Also, at `6:30:25 UTC time` (1:30:25 US Central time) there are two smaller peaks at ~100m at almost the same time, these two smaller peaks are from yolo16 and yolo17.
> 
> [image]
> 
> Here is the same zoom-in `irate()` plot of memory usage plot with time resolution increased from 3s to 1s.
> 
> Additional note: when I built the yolo container and run it on a local machine within a docker environment, I found that when yolo container starts to work on a new RGB image (from command `execute_yolo.sh` to `darknet`), its memory usage quickly increases from <100MB to ~1GB within 1 second, then have small flucuations around 1GB for a short amount of time; after this RGB image finishes the memory usage quickly drops to <100MB within 1 second. My other two pods (PocketSphinx, Aeneas) memory usages do not change this fast. I'm wondering that are the quick and large memory fluctuations in my YOLO pods related with the Peta Joule issue?

Thank you so much. I think it has a high potential that the Peta Joule comes from the large memory fluctuation. memory_bytes seems to be not a good feature in the model. We should consider removing it or find a better way to transform this metric before training. 

Currently, I change the default model to use only CPU time by https://github.com/sustainable-computing-io/kepler/pull/903. You may try re-running the yolo and see whether the peak is still there.

marceloamaral, 2023-09-05:
If the model update doesn't resolve this problem, it could be related to the issue described in https://github.com/sustainable-computing-io/kepler-model-server/pull/146.

closed 2023-09-06 by jwjwjw3