sustainable-computing-io/kepler #898: Error about Unknown node feature: metrcis adding 0 value

Lai-Kenny, 2023-08-29:
### What happened?

I make build-manifest OPTS="ESTIMATOR_SIDECAR_DEPLOY"
And change the model server image to the `quay.io/sustainable_computing_io/kepler_model_server:v0.6` and exporter image to `quay.io/sustainable_computing_io/kepler:latest-libbpf` in the daemonset.
Finally, change log level to v5.
The following is my kepler exporter log
```
I0829 03:43:02.332357       1 gpu.go:46] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0829 03:43:02.338686       1 qat.go:35] Failed to init qat-telemtry err: could not get qat status exit status 127
I0829 03:43:02.347937       1 exporter.go:158] Kepler running on version: 4b8a281
I0829 03:43:02.348008       1 config.go:270] using gCgroup ID in the BPF program: true
I0829 03:43:02.348079       1 config.go:272] kernel version: 5.15
I0829 03:43:02.348169       1 exporter.go:170] LibbpfBuilt: true, BccBuilt: false
I0829 03:43:02.348201       1 exporter.go:189] EnabledBPFBatchDelete: true
I0829 03:43:02.348260       1 config.go:143] ENABLE_EBPF_CGROUPID: true
I0829 03:43:02.348277       1 config.go:144] ENABLE_GPU: true
I0829 03:43:02.348287       1 config.go:145] ENABLE_QAT: false
I0829 03:43:02.348306       1 config.go:146] ENABLE_PROCESS_METRICS: false
I0829 03:43:02.348341       1 config.go:147] EXPOSE_HW_COUNTER_METRICS: false
I0829 03:43:02.348357       1 config.go:148] EXPOSE_CGROUP_METRICS: true
I0829 03:43:02.348365       1 config.go:149] EXPOSE_KUBELET_METRICS: true
I0829 03:43:02.348374       1 config.go:150] EXPOSE_IRQ_COUNTER_METRICS: true
I0829 03:43:02.348417       1 config.go:151] EXPOSE_ESTIMATED_IDLE_POWER_METRICS: false. This only impacts when the power is estimated using pre-prained models. Estimated idle power is meaningful only when Kepler is running on bare-metal or with a single virtual machine (VM) on the node.
I0829 03:43:02.348489       1 rapl_msr_util.go:129] failed to open path /dev/cpu/0/msr: no such file or directory
I0829 03:43:02.348569       1 power.go:71] Unable to obtain power, use estimate method
I0829 03:43:02.348640       1 csv_cred.go:68] failed to read csv file: node name understudent not found in file /etc/redfish/redfish.csv
I0829 03:43:02.348699       1 redfish.go:173] failed to initialize node credential: no supported node credential implementation
I0829 03:43:02.348743       1 power.go:56] use acpi to obtain power
I0829 03:43:02.348950       1 acpi.go:67] Could not find any ACPI power meter path. Is it a VM?
I0829 03:43:02.349009       1 bpf_perf.go:82] hardeware counter metrics config false
I0829 03:43:02.349023       1 bpf_perf.go:84] hardeware counter metrics not enabled
I0829 03:43:02.349054       1 bpf_perf.go:100] irq counter metrics config true
I0829 03:43:02.376817       1 utils.go:56] Available ebpf metrics: [bpf_cpu_time_us bpf_net_tx_irq bpf_net_rx_irq bpf_block_irq]
I0829 03:43:02.376855       1 utils.go:57] Available counter metrics: []
I0829 03:43:02.376858       1 utils.go:58] Available cgroup metrics from cgroup: [cgroupfs_memory_usage_bytes cgroupfs_kernel_memory_usage_bytes cgroupfs_tcp_memory_usage_bytes cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us cgroupfs_ioread_bytes cgroupfs_iowrite_bytes block_device                                                s_used]
I0829 03:43:02.376878       1 utils.go:59] Available cgroup metrics from kubelet: [kubelet_cpu_usage kubelet_memory_bytes]
I0829 03:43:02.376986       1 model.go:176] Model Config CONTAINER_TOTAL: {ModelType:Ratio ModelOutputType:DynPower TrainerName: EnergySource:acpi SelectFilter: InitModelURL: InitModelFilepath: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0829 03:43:02.377018       1 model.go:94] Using Power Model Ratio
I0829 03:43:02.377039       1 container_energy.go:109] Using the Ratio/DynPower Power Model to estimate Container Platform Power
I0829 03:43:02.377050       1 model.go:176] Model Config CONTAINER_COMPONENTS: {ModelType:Ratio ModelOutputType:DynPower TrainerName: EnergySource:rapl SelectFilter: InitModelURL: InitModelFilepath: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataF                                                eatureValues:[]}
I0829 03:43:02.377059       1 model.go:94] Using Power Model Ratio
I0829 03:43:02.377064       1 container_energy.go:118] Using the Ratio/DynPower Power Model to estimate Container Component Power
I0829 03:43:02.377073       1 model.go:176] Model Config PROCESS_TOTAL: {ModelType:Ratio ModelOutputType:DynPower TrainerName: EnergySource:acpi SelectFilter: InitModelURL: InitModelFilepath: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0829 03:43:02.377082       1 model.go:94] Using Power Model Ratio
I0829 03:43:02.377087       1 process_power.go:108] Using the Ratio/DynPower Power Model to estimate Process Platform Power
I0829 03:43:02.377094       1 model.go:176] Model Config PROCESS_COMPONENTS: {ModelType:Ratio ModelOutputType:DynPower TrainerName: EnergySource:rapl SelectFilter: InitModelURL: InitModelFilepath: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFea                                                tureValues:[]}
I0829 03:43:02.377100       1 model.go:94] Using Power Model Ratio
I0829 03:43:02.377105       1 process_power.go:117] Using the Ratio/DynPower Power Model to estimate Process Component Power
I0829 03:43:02.377115       1 model.go:176] Model Config NODE_TOTAL: {ModelType:LinearRegressor ModelOutputType:AbsPower TrainerName: EnergySource:acpi SelectFilter: InitModelURL: InitModelFilepath: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataF                                                eatureValues:[]}
I0829 03:43:02.377346       1 lr.go:188] LR Model (AbsPower): loadWeightFromURLorLocal(): &map[platform:{{4067.51980453514 map[] map[kubelet_cpu_usage:{8.225074882327771 61.882611238231696 128.77011809213587} kubelet_memory_bytes:{9.011536046098989e+06 4.377864223762704e+14 11.4818854701994}]}}] (error: <nil>)
I0829 03:43:02.377386       1 model.go:120] Using Power Model AbsPower
I0829 03:43:02.377396       1 node_platform_energy.go:53] Using the LinearRegressor/AbsPower Power Model to estimate Node Platform Power
I0829 03:43:02.377405       1 model.go:176] Model Config NODE_COMPONENTS: {ModelType:EstimatorSidecar ModelOutputType:AbsPower TrainerName: EnergySource:rapl SelectFilter: InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-db/main/models/Linux-4.15.0-213-generic-x86_64_v0.6/rapl/AbsPower/KubeletOnly/GradientBoostingRegressorTrainer_1.zip InitModelFilepath: IsNodePowerModel:false ContainerFeatureNames:[] NodeFeatureNames:[] SystemMetaDataFeatureNames:[] SystemMetaDataFeatureValues:[]}
I0829 03:43:03.904356       1 model.go:144] Using Power Model AbsPower
I0829 03:43:03.904382       1 node_component_energy.go:54] Using the EstimatorSidecar/AbsPower Power Model to estimate Node Component Power
I0829 03:43:03.904390       1 exporter.go:212] Initializing the GPU collector
I0829 03:43:09.905557       1 watcher.go:66] Using in cluster k8s config
I0829 03:43:09.906214       1 reflector.go:221] Starting reflector <unspecified> (0s) from github.com/sustainable-computing-io/kepler/pkg/kubernetes/watcher.go:122
I0829 03:43:09.906251       1 reflector.go:257] Listing and watching <unspecified> from github.com/sustainable-computing-io/kepler/pkg/kubernetes/watcher.go :122
I0829 03:43:10.006102       1 shared_informer.go:285] caches populated
libbpf: loading /var/lib/kepler/bpfassets/amd64_kepler.bpf.o
libbpf: elf: section(3) tracepoint/sched/sched_switch, size 2344, link 0, flags 6, type=1
libbpf: sec 'tracepoint/sched/sched_switch': found program 'kepler_trace' at insn offset 0 (0 bytes), code size 293 insns (2344 bytes)
libbpf: elf: section(4) .reltracepoint/sched/sched_switch, size 352, link 29, flags 40, type=9
libbpf: elf: section(5) tracepoint/irq/softirq_entry, size 144, link 0, flags 6, type=1
libbpf: sec 'tracepoint/irq/softirq_entry': found program 'kepler_irq_trace' at insn offset 0 (0 bytes), code size 18 insns (144 bytes)
[... cut ...]
libbpf: license of /var/lib/kepler/bpfassets/amd64_kepler.bpf.o is GPL
[... 549 lines cut ...]

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

marceloamaral, 2023-08-29:
@Lai-Kenny Is this error occurring later in the log, or does it only appear in the initial moments when Kepler is initiating?

The initialization of the node resource utilization map happens after the container metrics have been updated within the AddNodeResUsageFromContainerResUsage function. During the initial few seconds, the container metrics might not have been collected yet, but they should become available in few more seconds. Could you please verify this?

By the way, we could prevent all these error messages by including a return statement while the node metrics map is still empty...

Lai-Kenny, 2023-08-29:
Hi, @marceloamaral . I found the following error continues to occur.
```
I0829 13:03:16.074472       1 container_metric.go:189] cannot extract:
I0829 13:03:16.074474       1 container_metric.go:189] cannot extract:
I0829 13:03:16.074476       1 container_metric.go:189] cannot extract: gpu_sm_util
I0829 13:03:16.074479       1 container_metric.go:189] cannot extract: cpu_instructions
I0829 13:03:16.074481       1 container_metric.go:189] cannot extract: cpu_instructions
I0829 13:03:16.074484       1 container_metric.go:189] cannot extract: cpu_instructions
I0829 13:03:16.074490       1 container_metric.go:189] cannot extract: cache_miss
```

marceloamaral, 2023-08-30:
@Lai-Kenny This is because you are running on a VM, which does not have hardware counter metrics (`cpu_instructions`, `cache_miss`) and does not seems to have GPU metrics (`gpu_sm_util`). But as expected, the other metrics were added to the node map and we do not see the warnings anymore.

Lai-Kenny, 2023-08-30:
@marceloamaral ok, I got it. But I have a question about it. That will affect energy estimate?

vimalk78, 2023-08-30:
@Lai-Kenny this issue has been fixed in PR  https://github.com/sustainable-computing-io/kepler/pull/896 , you can try updating kepler code to lates or use latest image `quay.io/sustainable_computing_io/kepler:latest`

jichenjc, 2023-08-31:
>>>this issue has been fixed in PR https://github.com/sustainable-computing-io/kepler/pull/896 , you can try updating kepler code to lates or use latest image

I doubt it could per https://github.com/sustainable-computing-io/kepler/issues/898#issuecomment-1697412939
as this seems a VM env and #896 seems for other purpose..

closed 2023-11-06 by stale[bot]