sustainable-computing-io/kepler #901: Kubelet and cGroups exported metrics have something wrong

sunya-ch, 2023-08-31:
### What happened?

I've recently finished integrating a tool to visualize data from the model-server to validate the power model on Kepler and found that there are some issues from the metric data exported from the Kepler as shown in example plot here: https://github.com/sustainable-computing-io/kepler-model-server/issues/132#issuecomment-1698826365. 

Hardware counter has already been fixed but cGroups metric is still unable to use. 
Also, Kubelet metrics go wrong at some point of merge.

Kubelet is used to work before but now it seems wrongly export. 

This is total power collected for Coremark. 
[image]

These are metrics we got for cGroup and Kubelet:
[image]
[image]

### What did you expect to happen?

We expect the same CPU (time) usage to have the same trend as power like the other metric sources such as BPF and Counter.

[image]
[image]
Note that CPU instruction is not fixed yet when collecting the data. It should be fixed after 57a9823ab5430c03742f29d8fc8c577d26c423f4.

---

Kubelet metrics are used to be as expected before fixing the other metric sources in https://github.com/sustainable-computing-io/kepler/pull/872 but not sure what is the root cause of change. 
[image].

### How can we reproduce it (as minimally and precisely as possible)?

1. Run Kepler on BM 
2. Collect data with coremark on CPE https://github.com/sustainable-computing-io/kepler-model-server/blob/055eaf0342b0fd8595d10b7b9f99b403cbcb6ecb/model_training/script.sh#L159
3. Visualize with `plot` command  https://github.com/sustainable-computing-io/kepler-model-server/blob/main/cmd/README.md

### Kepler image tag

9b378ef7e33777898297ec9305cdf9174df7de28

### Kubernetes version

<details>

```console
$ kubectl version
# paste output here
```

</details>

### Cloud provider or bare metal

Bare metal

### OS version

<details>

```console
# On Linux:
$ cat /etc/os-release
# paste output here
$ uname -a
# paste output here

# On Windows:
C:\> wmic os get Caption, Version, BuildNumber, OSArchitecture
# paste output here
```

</details>

### Install tools

<details>

</details>

### Kepler deployment config

<details>

For on kubernetes:
```console
$ KEPLER_NAMESPACE=kepler

# provide kepler configmap
$ kubectl get configmap kepler-cfm -n ${KEPLER_NAMESPACE} 
# paste output here

# provide kepler deployment description
$ kubectl describe deployment kepler-exporter -n ${KEPLER_NAMESPACE} 
```

For standalone:
# put your Kepler command argument here

</details>

### Container runtime (CRI) and version (if applicable)

<details>

</details>

### Related plugins (CNI, CSI, ...) and versions (if applicable)

<details>

</details>

jichenjc, 2023-08-31:
to confirm my understanding , cgroup/kubelet are something we collect from system directly 
so you are considering we are using wrong source or some calculation issue? Thanks

marceloamaral, 2023-08-31:
@sunya-ch the Kubelet metrics looks a little bit similar, except that the newer version (after PR 872) has bigger spikes.
Which could be expected since the PR 872 `fix get containerID intermittently returning system_processes instead of the real containerID`. 

What about the cGroup metrics before the PR 872? Is it the same?

sunya-ch, 2023-08-31:
> @sunya-ch the Kubelet metrics looks a little bit similar, except that the newer version (after PR 872) has bigger spikes. Which could be expected since the PR 872 `fix get containerID intermittently [...]

Before 872, Kepler cannot properly get cgroup metric.

sunya-ch, 2023-08-31:
> to confirm my understanding , cgroup/kubelet are something we collect from system directly
> so you are considering we are using wrong source or some calculation issue? Thanks

It should collect from system directly. What can go wrong should not be calculation but the logic to put the read value to the container map.

rootfs, 2023-09-21:
also refer to #943 

If bpf features are sufficient, we can evaluate default to bpf models.

rootfs, 2023-10-05:
Since kubelet metrics can be used when ebpf counters are missing (see #959), we will keep it in those situations.

closed 2023-12-11 by stale[bot]